Koenkk/zigbee2mqtt #14828: OWON THS317-ET Temperature Sensor throws error when adding device

GerhardLang, 2022-11-05:
### What happened?

Hello.

I did buy the supported OWON temp sensor with external probe.
When adding, it works, but shows the following error and no temperature is available from device.

Any ideas?

Error 2022-11-05 16:33:13Failed to configure 'Haus Vorratsraum TK Thermometer', attempt 2 (TypeError: Cannot read properties of undefined (reading 'bind') at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:24) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/owon.js:185:29) at Configure.configure (/app/lib/extension/configure.ts:115:37) at EventEmitter.<anonymous> (/app/lib/extension/configure.ts:80:62) at EventEmitter.emit (node:events:525:35) at EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:52:22) at Controller.<anonymous> (/app/lib/zigbee.ts:66:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14) at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:557:14))
Error 2022-11-05 16:33:14Failed to configure 'Haus Vorratsraum TK Thermometer', attempt 3 (TypeError: Cannot read properties of undefined (reading 'bind') at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:24) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/owon.js:185:29) at Configure.configure (/app/lib/extension/configure.ts:115:37) at EventEmitter.<anonymous> (/app/lib/extension/configure.ts:80:62) at EventEmitter.emit (node:events:525:35) at EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:52:22) at Controller.<anonymous> (/app/lib/zigbee.ts:66:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14) at Controller.onDeviceAnnounce (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:442:14))

### What did you expect to happen?

expected the supported device to work :-)

### How to reproduce it (minimal and precise)

when adding the device the error occurs.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0

danieledwardgeorgehitchcock, 2022-11-05:
I successfully submitted a PR today Koenkk/zigbee-herdsman-converters#4890 which should fix this issue.

Please switch to the Dev branch for now

Koenkk, 2022-11-06:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/14809

closed 2022-11-06 by Koenkk