Koenkk/zigbee2mqtt #14829: [Feature request]: stopping write broken data

VladiGorb, 2022-11-05:
### Is your feature request related to a problem? Please describe

When devices marked as "not available" "offline", data of its last parameters continue writes to HA. I was disappointed to see my flower, wich was owflul, bcs flowersensor say to me 3days "dumidity 60%", nor device is offline. Yes, I guess I need a watchdog to any sensors, that need to control timeout of sensors, but if Z2M have availability options and desides to mark as offline, I guess it's mast stops write data from gost sensor to HA DB. Maybe it's can be set as proper of availability. 
For buttons, must be last state, but for sensors, must be configurable.
I am sorry if it's done, and I didn't found it in helps.

### Describe the solution you'd like

Checkbutton prevent data write if availability: offline

### Describe alternatives you've considered

None

### Additional context

None

szymucha94, 2022-11-06:
Can you provide more details including which devices are affected by this?
My setup with z2m makes zigbee aqara devices "unavailable" after a while if device fails to ping, ie. disconnects.

closed 2022-11-10 by VladiGorb

VladiGorb, 2022-11-10:
I cant understand whats wrong, but seems its not z2m error data! Its like HA Show graph of data, that marked in mysqldata_state 7 days ago,  and  last seen in z2m 7 days ago.

Please delete my Issue! Seems  I was wrong.

According to my thoughts, i would write an issue to HA about ghostgraph.

May i asks a litlle question abount z2m MAP tool. 
I have a map, and 3 devices states offline 7 days ago, 
[image]
Its seems diyruz-diyruz_geiger shown with link power. According to legend, offline devices cant be seen as linked to router.