Koenkk/zigbee2mqtt #14830: Error while starting zigbee-herdsman

artemave, 2022-11-05:
### What happened?

Fresh install. Conbee 2 with freshly updated firmware.

Home assistant:
```
Home Assistant 2022.11.1
Supervisor 2022.10.2
Operating System 9.3
Frontend 20221102.1 - latest 
```

Addon config:
```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://localhost:1883
serial:
  port: /dev/ttyACM0
  adapter: deconz
frontend: true
permit_join: true
advanced:
  log_level: debug
```

Fails to start with this error:

```
[19:08:13] INFO: Preparing to start...
[19:08:13] INFO: Socat not enabled
[19:08:15] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2022-11-05 19:08:20: Logging to console and directory: '/config/zigbee2mqtt/log/2022-11-05.19-08-20' filename: log.txt
Zigbee2MQTT:info  2022-11-05 19:08:20: Starting Zigbee2MQTT version 1.28.2 (commit #unknown)
Zigbee2MQTT:info  2022-11-05 19:08:20: Starting zigbee-herdsman (0.14.68)
Zigbee2MQTT:error 2022-11-05 19:08:31: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-11-05 19:08:31: Failed to start zigbee
Zigbee2MQTT:error 2022-11-05 19:08:31: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-11-05 19:08:31: Exiting...
Zigbee2MQTT:error 2022-11-05 19:08:31: undefined
```

I checked the FAQ but to no avail. Thank you for your help.

### What did you expect to happen?

Addon starts.

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

deCONZ_ConBeeII_0x26720700.bin.GCF

### Adapter

Conbee 2

### Debug log

Not sure where to find those...

bummzulu, 2022-11-06:
Did you go through these instructions to check all possible root causes?
https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html

artemave, 2022-11-06:
> Did you go through these instructions to check all possible root causes? https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html

Yes, I've gone through the list and to my best understanding none of that helps.

mdehaas, 2022-11-07:
Can you please login using SSH (addon of developer documentation) and post the output of the command `ls -l /dev/serial/by-id`

artemave, 2022-11-07:
```
[core-ssh ~]$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx    1 root     root            13 Nov  5 19:24 usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2438724-if00 -> ../../ttyACM0
```

mdehaas, 2022-11-07:
Are there other addons or integrations (ZHA) using the stick?

artemave, 2022-11-08:
Yes, actually. It was zha :facepalm: I was looking straight at it and somehow still didn't see it. Maybe that because zha integration tile in the UI doesn't actually say "zigbee home automation" or zha on it. And I was looking for those specific words.

Anyway, thank you very much for your help!

closed 2022-11-08 by artemave

jorgenslings, 2022-12-21:
I got this to work with very basic **/config/zigbee2mqtt/configuration.yaml**

homeassistant: true
permit_join: true
frontend:
  port: 8099
zigbee_herdsman_debug: true
serial:
  port: null
  adapter: deconz