Koenkk/zigbee2mqtt #14832: Tubes router reported offline but last seen reported as minutes ago

ptr727, 2022-11-05:
### What happened?

I noticed that when a power failure occurs, or when I pull the network cable on the tubes coordinator, that the tubes router would be reported as offline.
But what is strange is that the device is also reported as last seen x minutes ago, e.g. offline but last seen 28 minutes ago.

```
Warning 2022-11-04 17:43:56Failed to ping 'Tubes-Router-8F1717' (attempt 1/1, Read 0x00124b00258f1717/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 57150 - 1 - 220 - 0 - 1 after 10000ms))
```

[image]

There is a discussion of the problem [here](https://github.com/tube0013/tube_gateways/discussions/100).

As I was typing this, I changed to debug mode, and I captured this log, where the tubes router came back online:
```
Debug 2022-11-05 13:23:07Received Zigbee message from 'Tubes-Router-8F1717', type 'attributeReport', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 8 with groupID 0
Debug 2022-11-05 13:23:07Device 'Tubes-Router-8F1717' reconnected
Info 2022-11-05 13:23:07MQTT publish: topic 'zigbee2mqtt/Tubes-Router-8F1717/availability', payload 'online'
Info 2022-11-05 13:23:07MQTT publish: topic 'zigbee2mqtt/Tubes-Router-8F1717', payload '{"device":{"friendlyName":"Tubes-Router-8F1717","hardwareVersion":1,"ieeeAddr":"0x00124b00258f1717","manufacturerID":0,"manufacturerName":"TubesZB","model":"tubeszb.router","networkAddress":57150,"powerSource":"Mains (single phase)","softwareBuildID":"20220111","type":"Router","zclVersion":3},"last_seen":"2022-11-05T20:23:07.423Z","linkquality":40}'
Debug 2022-11-05 13:23:09Retrieving state of 'Tubes-Router-8F1717' after reconnect
```

Availability simple is enabled.

### What did you expect to happen?

Tubes router stays online even if power cycled or network cable pulled from coordinator.
If device is reported as offline, then last seen to be more than an hour ago, not minutes ago.

### How to reproduce it (minimal and precise)

Cannot readily reproduce.
Power cycle tubes router.
Disconnect and reconnect coordinator network cable.

### Zigbee2MQTT version

1.28.2 commit: 360a777

### Adapter firmware version

20220219

### Adapter

Tubes zStack3x0

### Debug log

```
Warning 2022-11-04 17:43:56Failed to ping 'Tubes-Router-8F1717' (attempt 1/1, Read 0x00124b00258f1717/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 57150 - 1 - 220 - 0 - 1 after 10000ms))
```

```
Debug 2022-11-05 13:23:07Received Zigbee message from 'Tubes-Router-8F1717', type 'attributeReport', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 8 with groupID 0
Debug 2022-11-05 13:23:07Device 'Tubes-Router-8F1717' reconnected
Info 2022-11-05 13:23:07MQTT publish: topic 'zigbee2mqtt/Tubes-Router-8F1717/availability', payload 'online'
Info 2022-11-05 13:23:07MQTT publish: topic 'zigbee2mqtt/Tubes-Router-8F1717', payload '{"device":{"friendlyName":"Tubes-Router-8F1717","hardwareVersion":1,"ieeeAddr":"0x00124b00258f1717","manufacturerID":0,"manufacturerName":"TubesZB","model":"tubeszb.router","networkAddress":57150,"powerSource":"Mains (single phase)","softwareBuildID":"20220111","type":"Router","zclVersion":3},"last_seen":"2022-11-05T20:23:07.423Z","linkquality":40}'
Debug 2022-11-05 13:23:09Retrieving state of 'Tubes-Router-8F1717' after reconnect
```

jerrm, 2022-11-06:
Was it recently paired?  

Has z2m been restarted since pairing?  If not, try restarting the z2m container. 

See if anything at below seems relevant:

https://github.com/Koenkk/zigbee2mqtt/issues/14312

https://github.com/Koenkk/zigbee2mqtt/issues/13758#issue-1354893710

ptr727, 2022-11-06:
Z2M restarted about 20 hours ago when updated to new version.
But, I just restarted again.

Can you explain a bit more about the reporting changes and meanings?

[image]
[image]

jerrm, 2022-11-06:
Parameters I have tested with:
```
Endpoint: 8
Cluster: genBasic
Attribute: zclVersion
Min rep interval: 0
max rep interval: 560
Min rep change: 0
```

Above assumes the default availability timeout of 10 minutes for an active advice.  10 minutes is actually 600 seconds, but if you push too close to the 600 mark, z2m may attempt the "ping" before the report and the failed ping will mark the device offline.

I tested the system in the "failing ping" state for three days and the 560 was stable in a 70-80 device net. If you wanted to be extra safe and guard against a missed packet, you could do something like 280 seconds so you have at least two tries in a 10 minute interval.

With my Sonoff adapters, reporting hasn't been necessary with the reboot step, but we don't really know what's going on.

ptr727, 2022-11-06:
Thanks for the explanation, if it goes offline again I'll modify the parameters.
It is just suspicious that it would work for weeks and then not work, and it did really seem to be related to temporary power or network interruption.

closed 2022-12-15 by github-actions[bot]

diedrichg, 2023-10-28:
I also have an issue with power loss. When the power is restored it's as though the coordinator stops broadcasting by the time Z2M is able to start. By the time Z2M starts its negotiation the coordinator has stopped listening and herdsman fails and Z2M stops.

Within HA I can ping the coordinator address within the terminal add-on so it's strange that when Z2M starts its search that it can't find the coordinator.

The only fix is to unplug the coordinator, plug it back in, and then start Z2M. The coordinator will then be found without issue.

ptr727, 2023-10-28:
I am now running 5 tubes repeaters, and they seem stable.  
When I reported this issue I did change the parameters to those suggested by @jerrm, `max rep interval: 560`, but I later added more repeaters, and they run default `max rep interval: 3600`, so maybe it is something else, but try `max rep interval: 560`.