Koenkk/zigbee2mqtt #14833: Does not work with nodejs version 19.x

geeregman, 2022-11-05:
### What happened?

updated nodejs from 18.x to 19.0.1  Zigbee2mqtt will not start, says requires nodejs version 18 .... using version 19 .....

### What did you expect to happen?

I expected zigbee2mqtt to run as always.

### How to reproduce it (minimal and precise)

update to nodejs 19.0.1-1

### Zigbee2MQTT version

1.28.2 

### Adapter firmware version

20210708

### Adapter

Slaeh 2652 usb dongle

lux73, 2022-11-06:
to avoid these Problems please consider install via nodejs Repo -> https://github.com/Koenkk/zigbee2mqtt/discussions/13364

with these you are able to pick up an specific Version ;-)

btw the Firmware from your Zigbee Coordiantor is out-of-date

geeregman, 2022-11-06:
Yes, i reverted my nodejs back to 18 and everything works fine, like before.
I guess my issue is just that the latest z2m(1.28.2) will not work with the latest nodejs(19) .......

Will close and keep an eye out for update ...............  thanks ..........

closed 2022-11-06 by geeregman

Koenkk, 2022-11-06:
Should be fixed now https://github.com/Koenkk/zigbee2mqtt/pull/14837 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

geeregman, 2022-11-06:
been using z2m for awhile. Works well.  Thanks for your knowledge and time .................. g