Koenkk/zigbee2mqtt #14835: 2-gang switch not working probably when commands are coming from homebridge

bummzulu, 2022-11-06:
### What happened?

Controlling this 2-gang dimmable switch that uses L1 and L2 (brightness and state commands) to control this device:
https://www.zigbee2mqtt.io/devices/MS-105B.html#moes-ms-105b

When I use the apple home app (over homebridge and z2m plugin) to control the device there is an issue with setting the brightness for L2. The device does not accept changes to the brightness and jumps back to the original brightness level that is already set.

The reason for this is the order in which the properties are set in the payload of the command.
HomeKit sends for the L2 command the properties in the JSON in the order of state and brightness. Zigbee only accepts commands in the order of brightness and state.

Hope this can be changed as we cannot change the way how homekit is composing the messages.

### What did you expect to happen?

Brightness adjustments should be possible and order of the properties should not matter in the JSON payload. 

### How to reproduce it (minimal and precise)

Send a payload to the devices:
{
    "state_l2": "ON",
    "brightness_l2": 200
}

Result: L2 turns on but brightness level of 200 is ignored.

Working message:
{
    "brightness_l2": 200,
    "state_l2": "ON"
}
Result: L2 turns on and brightness level is set to 200.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20210708

### Adapter

Electrolama Zzh!

itavero, 2022-11-21:
Other users of homebridge-z2m are reporting similar behavior with this device.

itavero, 2022-11-21:
Looked at the ["To Zigbee" converter'](https://github.com/Koenkk/zigbee-herdsman-converters/blob/v14.0.670/converters/toZigbee.js#L6818-L6852) for this device and it appears to be a Tuya-based device.
I can imagine that the datapoint change requests are not merged correctly, but then I would expect other Tuya-based devices to also be affected.

closed 2022-12-29 by github-actions[bot]