Koenkk/zigbee2mqtt #14841: [New device support]: Hue Gradient Signe Floor Lamp - Black (915005987601)

3mta3, 2022-11-06:
### Link

https://www.philips-hue.com/en-us/p/hue-white-and-color-ambiance-signe-gradient-floor-lamp/046677803513

### Database entry

{"id":92,"type":"Router","ieeeAddr":"0x001788010c51cdb2","nwkAddr":34747,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"915005987601","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"zclVersion":2}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20210407","swBuildId":"1.86.6","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1667753040726,"defaultSendRequestWhen":"immediate"}

### Comments

Black Signe gradient floor lamp model # 915005987601 - same functionality as 4080248U9. Name is slightly different so I added as new model to devices/philips.js and it is working:

```
    {
        zigbeeModel: ['915005987601'],
        model: '915005987601',
        vendor: 'Philips',
        description: 'Hue Gradient Signe Floor Lamp - Black',
        toZigbee: [tzLocal.gradient_scene, ...hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}).toZigbee],
        exposes: [exposes.enum('gradient_scene', ea.SET, Object.keys(gradientScenes)),
            ...hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}).exposes],
        extend: hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    },
```

Note: adding this model number to the existing device entry also works, but I think 4080248U9 may be previous-gen / non-gradient version (not sure):

```
    {
        zigbeeModel: ['4080248U9', '915005987201', **'915005987601'**],
        model: '4080248U9',
        vendor: 'Philips',
        description: 'Hue White and color ambiance Signe floor light',
        toZigbee: [tzLocal.gradient_scene, ...hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}).toZigbee],
        exposes: [exposes.enum('gradient_scene', ea.SET, Object.keys(gradientScenes)),
            ...hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}).exposes],
        extend: hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    },
```

Let me know which update you believe is most appropriate - if you'll update philips.js with the net-new model, I'll submit a pull request with related doc/image. Thanks!

yestrdave, 2022-12-29:
Hello All I'm a little new to the Z2M world .  I see that the 915005987601 (Black) has been added as a supported device.  I purchased the white one 915005987501 (White).  Im not sure what to do to get it to work.  It seams the part number needs added for the device to work.  For additional reference 929003479901 (Oak). All of these devices are the same except color.

[image]

Any Help would greatly be appreciated.

Koenkk, 2022-12-30:
@yestrdave added! Device will be supported in the 1 January release.

yestrdave, 2022-12-30:
@Koenkk Thank you so much.  I did make an attempt to create a converter for it.  Unfortunately, I couldn't get the .js file correct

j0ech3n, 2024-10-23:
@Koenkk, I'm sorry to revive this thread, but oak (929003479901) was never added. Can this be added? This is the exact same lamp but a different color