Koenkk/zigbee2mqtt #14850: Zigbee2mqtt stops working

pvdburgt-new, 2022-11-07:
### What happened?

After updating zigbee2mqtt it randomly stops, starting again is needed. When I rolled back to HA backup ( I updated HA and zigbee2mqtt and node-red )everything stays working. After updating same problem

### What did you expect to happen?

IT to stay working

### How to reproduce it (minimal and precise)

Cannot reproduce it just stops

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

202110120

### Adapter

zzh

### Debug log

error: srsp - zdo - mgmtpermitjoinreq after 6000ms at timeout._ontimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listontimeout (node:internal/timers:559:17) at processtimers (node:internal/timers:502:7)

pauloon, 2022-11-07:
I'm having this same issue after last update.
Here is what shows up in the log:

<--- Last few GCs --->
[7:0x7fa21c7993c0] 61013533 ms: Mark-sweep 2044.2 (2085.3) -> 2042.2 (2085.3) MB, 2087.8 / 0.0 ms  (average mu = 0.133, current mu = 0.010) allocation failure scavenge might not succeed
[7:0x7fa21c7993c0] 61015639 ms: Mark-sweep 2044.3 (2085.3) -> 2042.2 (2085.3) MB, 2082.8 / 0.0 ms  (average mu = 0.074, current mu = 0.011) allocation failure scavenge might not succeed
<--- JS stacktrace --->
FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory

pvdburgt-new, 2022-11-08:
I did a rollback to an older version and thought it solved the problem. But just after midnight still had the same problem. I will check the logs (debug is now on) and else I will go to an older stable HA backup and will slowly change/update

pauloon, 2022-11-08:
> 

I reinstalled it here from zero, the same version, and I'm checking if it happens again.
I'll keep posted.

Thanks,
Paulo.

closed 2022-12-17 by github-actions[bot]