Koenkk/zigbee2mqtt #14853: Zigbee2MQTT service stops working - FATAL ERROR

pauloon, 2022-11-07:
### What happened?

After last update I've noticed Z2M is stopping service with a fatal error, out of the blue.

I'm using HASS.IO all updated, in a i5 machine with 8 Gb do memory and 128 GB SSD.

### How to reproduce it (minimal and precise)

Just leave it running. After one or two days it stops working (service drops).

### Zigbee2MQTT version

 1.28.2 commit: unknown

### Adapter firmware version

20220219

### Adapter

SONOFF USB Dongle

### Debug log

I did not have the debug log active when this happened.
Posting normal log.
...
Zigbee2MQTT:info  2022-11-07 11:45:15: MQTT publish: topic 'zigbee2mqtt/Smart Plug 15', payload '{"child_lock":"UNLOCK","current":0.04,"energy":6.45,"indicator_mode":"off/on","last_seen":"2022-11-07T11:45:13-03:00","linkquality":102,"power":0,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"update_available":false}'
<--- Last few GCs --->
[7:0x7fa21c7993c0] 61013533 ms: Mark-sweep 2044.2 (2085.3) -> 2042.2 (2085.3) MB, 2087.8 / 0.0 ms  (average mu = 0.133, current mu = 0.010) allocation failure scavenge might not succeed
[7:0x7fa21c7993c0] 61015639 ms: Mark-sweep 2044.3 (2085.3) -> 2042.2 (2085.3) MB, 2082.8 / 0.0 ms  (average mu = 0.074, current mu = 0.011) allocation failure scavenge might not succeed
<--- JS stacktrace --->
FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory

pauloon, 2022-11-08:
Hi,

Just had the same problem again.
Attached the log file:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9960947/log.txt)

Please, help.
Thanks,
Paulo.

pauloon, 2022-11-09:
I reinstalled Z2M all from zero and today it happened again:

2022-11-09T17:11:10.996Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":55137,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":61,"securityuse":0,"timestamp":11657102,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[9,72,2,0,193,1,4,0,1,0]}}
2022-11-09T17:11:10.996Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":72,"manufacturerCode":null,"commandIdentifier":2},"Payload":{"seq":49408,"dpValues":[{"dp":1,"datatype":4,"data":{"type":"Buffer","data":[0]}}]},"Command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}},"address":55137,"endpoint":1,"linkquality":61,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-11-09 14:11:10: Received Zigbee message from 'Presença 2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":49408}' from endpoint 1 with groupID 0
<--- Last few GCs --->
[7:0x7f45b57b73c0] 84373002 ms: Scavenge (reduce) 2042.8 (2083.1) -> 2041.9 (2083.1) MB, 2.5 / 0.0 ms  (average mu = 0.105, current mu = 0.019) allocation failure 
[7:0x7f45b57b73c0] 84373008 ms: Scavenge (reduce) 2043.0 (2083.1) -> 2042.1 (2083.1) MB, 2.7 / 0.0 ms  (average mu = 0.105, current mu = 0.019) allocation failure 
[7:0x7f45b57b73c0] 84373015 ms: Scavenge (reduce) 2043.0 (2083.1) -> 2042.1 (2083.3) MB, 2.6 / 0.0 ms  (average mu = 0.105, current mu = 0.019) allocation failure 
<--- JS stacktrace --->
FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory

@Koenkk ... can you, please, help?

Thank you.
Paulo.

Koenkk, 2022-11-09:
What node version are you using?

pauloon, 2022-11-09:
> What node version are you using?

Thanks for helping me.
Not sure what you mean by "node". Sorry for my ignorance.
Are you talking about NodeRed ? I'm using version 13.5.2.

Follows attached a diagnostic file with some info about my system.
[config_entry-mqtt-5f9bbf81bfa29cb6923246ba774884c4.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/9974211/config_entry-mqtt-5f9bbf81bfa29cb6923246ba774884c4.json.txt)

Please, let me know.

Thanks,
Paulo.

pauloon, 2022-11-11:
> I mean node.js (try running `node --version` on your system)

Hi @Koenkk ,
I tried and it does not work:
[image]

Maybe because I'm using Hass.io and NodeRed?

Thanks,
Paulo.

Koenkk, 2022-11-11:
how are you running z2m, via the addon?

pauloon, 2022-11-11:
> how are you running z2m, via the addon?

Dear @Koenkk ,
Yes, I've installed it via Add-On.
[image]

See if this helps...
[image]

Thanks,
Paulo.

pauloon, 2022-11-11:
If I turn off the watchdog it stops after a few hours.
Just happened again...
We are in a group of more than 2.000 HA users and more people are getting this too.
Looks like Hass.io users...

pauloon, 2022-11-11:
@Koenkk .... Look!
With help from some friends, i think I got the node.js version:
[image]

Will wait for your instructions.

Thanks,
Paulo.

Koenkk, 2022-11-12:
Can you provide the herdsman debug logging, from starting z2m till the error?

pauloon, 2022-11-12:
> 

Hi @Koenkk ,

Of course.
Is that the one I get from Settings > System > Logs > Z2M, copy and save?
[image]

Is there a way to access former logs?

Thanks,
Paulo.

Koenkk, 2022-11-13:
I have not seen this screen before, how big is the log? (does it include everything from start)/

pauloon, 2022-11-13:
> I have not seen this screen before, how big is the log? (does it include everything from start)

Dear @Koenkk ,
This is the screen I mentioned before. I'm not sure if it includes all of it.
I just grabbed all of its contents so you can see:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9997327/log.txt)
Please let me know if there are any other ways to get the logs.

I'm still waiting so it crashes again but since I'm not home these days maybe it "cooled down" a little.

Thanks,
Paulo.

Koenkk, 2022-11-13:
These are not the herdsman debug logging. (they contain `zigbee-herdsman` like in https://github.com/Koenkk/zigbee2mqtt/issues/14853#issuecomment-1312524794).

See https://www.zigbee2mqtt.io/guide/usage/debug.html#home-assistant-os-supervised-addon how to get the logging.

pauloon, 2022-11-13:
> These are not the herdsman debug logging. (they contain `zigbee-herdsman` like in [#14853 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/14853#issuecomment-1312524794)).
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html#home-assistant-os-supervised-addon how to get the logging.

Dear @Koenkk ,

I GOT ANOTHER CRASH!
I copied the entire log:
[log2.txt](https://github.com/Koenkk/zigbee2mqtt/files/9997965/log2.txt)

Please, let me know.

Thanks,
Paulo.

Koenkk, 2022-11-14:
can you turn off `Espelho 2 4gang` and see if the error still occurs?

pauloon, 2022-11-14:
> can you turn off `Espelho 2 4gang` and see if the error still occurs?

I will remove it and restart Z2M, ok?

Thanks,
Paulo.

Koenkk, 2022-11-15:
Ok 👍

XanderTenBoden, 2022-11-15:
I seem to have the same issue here (I opened an issue at https://github.com/Koenkk/zigbee2mqtt/issues/14992 before finding this one). I'm also using node version v16.17.1, so perhaps it has something to do with that indeed? I'm still trying to find the logs.

pauloon, 2022-11-15:
> Ok 👍

I got another crash today, so I guess it is not "Espelho 2 4gang" fault. Now I reseted it again and I'm testing one more.

XanderTenBoden, 2022-11-16:
Mine crashed as well. The Z2M logs gave me the following error:

```Error: SRSP - AF - dataRequestExt after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)```
    

> These are not the herdsman debug logging. (they contain `zigbee-herdsman` like in [#14853 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/14853#issuecomment-1312524794)).
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html#home-assistant-os-supervised-addon how to get the logging.

I am not able to retrieve the herdsman logs using this method, because the docker container just seems to crash in my case. So when I run `docker ps`, there isn't a z2m one running after the crash.

pauloon, 2022-11-16:
Dear @Koenkk ,

After I removed the Espelho 2 4gang, it is still happening.
Follows the last log extracted:
[log4.txt](https://github.com/Koenkk/zigbee2mqtt/files/10025978/log4.txt)

Please, let me know.

Thanks,
Paulo.

XanderTenBoden, 2022-11-16:
@pauloon does the docker container also completely stop in your case?

pauloon, 2022-11-16:
> @pauloon does the docker container also completely stop in your case?

Hi @XanderTenBoden .... How to I check that?
What's the symptom?

Thanks,
Paulo.

pauloon, 2022-11-16:
> @pauloon does the docker container also completely stop in your case?

Hi @XanderTenBoden 
I think the docker is crashing too.
One time that I ran docker ps and didn't see it.
I'm going to let it happen again, and check to make sure.

Thanks,
Paulo.

XanderTenBoden, 2022-11-17:
@pauloon What I did is turning off the watchdog and then wait for it to stop working. As soon as that did, all my zigbee devices in lovelace turned to "unavailable", and the zigbee2mqtt add-on wasn't running anymore.

It seems like the issue has something to do with the z2m docker eating too much RAM. I've enabled the memory sensors and this is what I'm seeing in the history:

[https://tenboden.com/ram.png](url)

pauloon, 2022-11-17:
> @pauloon What I did is turning off the watchdog and then wait for it to stop working. As soon as that did, all my zigbee devices in lovelace turned to "unavailable", and the zigbee2mqtt add-on wasn't running anymore.
> 
> It seems like the issue has something to do with the z2m docker eating too much RAM. I've enabled the memory sensors and this is what I'm seeing in the history:
> 
> https://tenboden.com/ram.png
> 
> I do have a lot of devices connected, about 90. How many do you have? It might explain why the crashes are more frequently for me if you have less than that connected? 🤔

@XanderTenBoden ,
I have 51 devices.
I'm also keeping watchdog disabled, otherwise it restarts Z2M.
I'll try to check the memory history, same as you did.

Thanks,
Paulo.

Koenkk, 2022-11-17:
@pauloon I also see `Presença 4` and `Presença 2` are spamming a lot, can you also try to remove those?

pauloon, 2022-11-17:
> @pauloon I also see `Presença 4` and `Presença 2` are spamming a lot, can you also try to remove those?

Hi @Koenkk ,

I will try turn them off, to stop the traffic.
These are mmWave presence sensors, which I have four. I know they push a lot of data from one expose called "target_distance" that pushes every second, but ONLY when there is someone being detected. As far as I understood it is not possible to turn off this reporting. I added this expose in the "filters" so it reduces MQTT data (I don't use it).
I'm guessing these are not guilty, because the crashes that happened a few days ago I was traveling, and they were reporting much less data.

I'll keep you posted.

Thanks,
Paulo.

pauloon, 2022-11-17:
@Koenkk ,

Just got another crash.
Very fast, this time.

Follows the log:
[log5.txt](https://github.com/Koenkk/zigbee2mqtt/files/10035582/log5.txt)

@XanderTenBoden ,
Indeed, the contained crashed too.
But I did not get the same memory spique that you got.
[image]

Please, let me know,
Thanks,
Paulo.

XanderTenBoden, 2022-11-17:
@pauloon how do you get these logs?

When I use the `docker logs CONTAINER_ID > log.txt 2>&1` command I only get like 15 lines max.

pauloon, 2022-11-17:
> @pauloon how do you get these logs?
> 
> When I use the `docker logs CONTAINER_ID > log.txt 2>&1` command I only get like 15 lines max.

@XanderTenBoden ,

Settings > Add-Ons > Z2M > Logs
[image]

Rgds,
Paulo.

XanderTenBoden, 2022-11-17:
Mine stopped responding as we speak, however the container is still running. Logs I got now are these:

```[19:00:24] INFO: Preparing to start...
[19:00:25] INFO: Socat not enabled
[19:00:27] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:warn  2022-11-17 19:00:33: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-11-17 19:00:33: Allowing new devices to join.
Zigbee2MQTT:warn  2022-11-17 19:00:33: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:error 2022-11-17 19:00:59: Publish 'set' 'state' to 'front_door_light' failed: 'Error: Command 0x00158d00032cadb0/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 39508 - 1 - 6 - 6 - 11 after 10000ms)'
Zigbee2MQTT:error 2022-11-17 19:01:02: Publish 'set' 'brightness' to 'living_room_desk_led_strip' failed: 'Error: Command 0xa4c138e0bfda4062/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:error 2022-11-17 19:01:24: Publish 'set' 'color' to 'living_room_desk_led_strip' failed: 'Error: Command 0xa4c138e0bfda4062/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":21168,"colory":21561}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:error 2022-11-17 20:52:08: Publish 'set' 'state' to 'office_spots' failed: 'Error: Command 0x00158d0006371aab/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 40507 - 1 - 148 - 6 - 11 after 10000ms)'
Zigbee2MQTT:error 2022-11-17 21:53:40: Publish 'set' 'state' to 'office_spots' failed: 'Error: Command 0x00158d0006371aab/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 40507 - 1 - 153 - 6 - 11 after 10000ms)'
Zigbee2MQTT:error 2022-11-17 21:54:20: Publish 'set' 'state' to 'office_spots' failed: 'Error: Command 0x00158d0006371aab/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 40507 - 1 - 164 - 6 - 11 after 10000ms)'```

XanderTenBoden, 2022-11-17:
> Mine crashed as well. The Z2M logs gave me the following error:
>
> ```
> Error: SRSP - AF - dataRequestExt after 6000ms
>     at Timeout._onTimeout [...]

I'm looking at this error now that I had yesterday, and when googling on that I get a result on this page that seems to crash on the exact same line (waitress.ts:64:35):
https://github.com/Koenkk/zigbee2mqtt/issues/9136#issuecomment-944098300

Not sure if I'm onto something though, but perhaps it's worth mentioning?

XanderTenBoden, 2022-11-18:
I've got a new crash and log here:

[log-z2m-xander-18-11-2022.txt](https://tenboden.com/log-z2m-xander-18-11-2022.txt)

pauloon, 2022-11-18:
> I've got a new crash and log here:
> 
> [log-z2m-xander-18-11-2022.txt](https://tenboden.com/log-z2m-xander-18-11-2022.txt)
> 
> @pauloon I seem to have a lot less debugging data. How is your Z2M debugging config set up?

Dear @XanderTenBoden ,

I guess is because I have these parameters configured (Herdsman Debug Active):

Settings > Add-Ons > Z2M > Configuration tab
[image]

And in my Z2M "configuration.yaml":

advanced:
  log_level: debug

Hope this helps.

Thanks,
Paulo.

XanderTenBoden, 2022-11-18:
> Where is the Z2M configuration.yaml located?

Nevermind, found it.

Let's see if I get some more info now as well.

XanderTenBoden, 2022-11-18:
It crashed again. I do have a little bit more logging now, but since the docker container crashed again, I can only retrieve the last lines that are shown in the Z2M add-on's interface in HA:

[https://tenboden.com/log-z2m-xander-18-11-2022-2.txt](logs-2)

@Koenkk  do you notice anything strange in here?

pauloon, 2022-11-18:
@Koenkk @XanderTenBoden 
Just got another crash...

The memory got the behaviour you said before:
[image]

Follows the log:
[log6.txt](https://github.com/Koenkk/zigbee2mqtt/files/10045267/log6.txt)

The container also crashed.
The mmWave Tuya presence sensors were disabled (I don't believe it is a device specific problem).

Please, let me know if I can help anyway.

Thanks,
Paulo.

pauloon, 2022-11-19:
It really do looks like there is a bug that makes something consume memory until it crashes:
[image]

XanderTenBoden, 2022-11-19:
Mine just crashed 2 minutes (!) after restarting it.

[https://tenboden.com/log-z2m-xander-19-11-2022.txt](log-z2m-xander-19-11-2022.txt)

pauloon, 2022-11-19:
@Koenkk ...
Please, help us. I have other fellows that are suffering the same with the HassOS.
Memory use keeps building up:
[image]

Probably it is a glitch or bug to cause this.
If you need remote access to my HA to investigate, we can arrange that.

Thanks,
Paulo.

XanderTenBoden, 2022-11-19:
@pauloon what I've also noticed is that before actually crashing the docker container, my devices stop responding to any interaction. In other words: Z2M is still 'running', but lights can't be toggled and stuff like that. Do you see the same behaviour?

pauloon, 2022-11-19:
> @pauloon what I've also noticed is that before actually crashing the docker container, my devices stop responding to any interaction. In other words: Z2M is still 'running', but lights can't be toggled and stuff like that. Do you see the same behaviour?

I didn't notice that, but it might be possible.
It took a while for me to check the Z2M after noticing that it had crashed.
Maybe it was "frozen" before crashing. Not sure.

Paulo.

pauloon, 2022-11-19:
PS: I'm been talking to people and almost everyone has watchdog on, so I guess there's more people being impacted by this problem, but they just don't notice.
Here, if I leave watchdog on, I don't notice anything because I guess it gets restarted quickly.

Thanks,
Paulo.

XanderTenBoden, 2022-11-19:
> PS: I'm been talking to people and almost everyone has watchdog on, so I guess there's more people being impacted by this problem, but they just don't notice. Here, if I leave watchdog on, I don't notice anything because I guess it gets restarted quickly.
> 
> Thanks, Paulo.

Yes maybe. I found out about the issue because of the non-responsiveness I mentioned before. And because some automations that were sending TTS messages to my Google Home tend to trigger when restarting Z2M.

pauloon, 2022-11-19:
Memory is coming to the limit...
[image]

Let's see if it crashes again....

pauloon, 2022-11-19:
Just crashed again....
[image]

[log7.txt](https://github.com/Koenkk/zigbee2mqtt/files/10047983/log7.txt)

Container crashed too...

pauloon, 2022-11-20:
Memory seems to keep building up constantly:
[image]

Now I'm making another test...
I've made a new install of HassOS on another computer, pretty much clean, just MQTT, M2Z and Node-Red.
The memory seems to have the same behaviour, but in a much slower pace (maybe because I've added only two devices?):
[image]

Could be it an indication of a bug with the HassOS and the Z2M add-on?

XanderTenBoden, 2022-11-20:
@pauloon it seems like i've resolved the issue here.

The issue was that I had some power plugs with consumption-statistics that were pushing these statistics every second. After changing this to once every 10 seconds, my Z2M no longer seems to crash and the memory is at a table level.

pauloon, 2022-11-20:
> @pauloon it seems like i've resolved the issue here.
> 
> The issue was that I had some power plugs with consumption-statistics that were pushing these statistics every second. After changing this to once every 10 seconds, my Z2M no longer seems to crash and the memory is at a table level.

@XanderTenBoden ,
Or maybe you reduced the traffic so it will take much longer to crash?

Like, in the test pc that I've configured, and memory is still building up, only much slower?
[image]

This pc only has one scene button configured, and a smart plug.

If I have one or two plugs reporting at every 5 seconds, shouldn't this be beareable for Z2M?

That is very frustrating.

Rgds,
Paulo.

XanderTenBoden, 2022-11-20:
@pauloon I will watch what happens, but it seems there is no increase in memory consumption anymore. Not even a slight one. In fact, it went down with 40MB over the last 2 hours.

I think what happened (at least here) is that the power consumption reports were too much to handle for my coordinator. So it builds up a queue in memory. That might also explain why my lights stopped working after a certain amount of time, because the queue was simply getting too long. Not 100% sure this is what happened, but it makes sense in my head.

It might also mean that in your case, maybe your network is too big for the type of coordinator you are using? But that's only a wild guess though.

dunkelz, 2022-11-20:
@XanderTenBoden I have same problem as you guys. This must be some regression as it was working fine until 1.28.0-1, and is still running since I reverted to that version.

pauloon, 2022-11-20:
But 50 devices too big?

Anyway, a brand new install with only 2 devices (yes, one is reporting power consumption), and it is still building up memory.
Still looks like a bug to me...
@Koenkk could take a look at this for us.

Thanks,
Paulo.

XanderTenBoden, 2022-11-20:
@pauloon maybe you can try what happens if you disable the power consumption reporting? Could be that something has changed in the handling of that since the new release?

@dunkelz I didn't have problems before that release either. However downgrading to 1.28.x didn't resolve the problem for me either 🤷‍♂️

Koenkk, 2022-11-20:
What are the models of the plugs you removed from the network? There is a memory leak somewhere (and this should be fixed)

XanderTenBoden, 2022-11-20:
@Koenkk [TS011F_plug_3](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3)

Koenkk, 2022-11-20:
@XanderTenBoden 

Could you check if the issue is fixed with the following external converter (energy measurement won't work anymore)

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    fingerprint: [].concat(...TS011Fplugs.map((manufacturerName) => {
        return [160, 69, 68, 65, 64].map((applicationVersion) => {
            return {modelID: 'TS011F', manufacturerName, applicationVersion};
        });
    })),
    model: 'TS011F_plug_3',
    description: 'Smart plug (with power monitoring by polling)',
    vendor: 'TuYa',
    whiteLabel: [{vendor: 'VIKEFON', model: 'TS011F'}, {vendor: 'BlitzWolf', model: 'BW-SHP15'},
        {vendor: 'Avatto', model: 'MIUCOT10Z'}, {vendor: 'Neo', model: 'NAS-WR01B'}],
    ota: ota.zigbeeOTA,
    fromZigbee: [fz.on_off, fzLocal.electrical_measurement_skip_duplicate, fzLocal.metering_skip_duplicate, fz.ignore_basic_report,
        fz.tuya_switch_power_outage_memory, fz.ts011f_plug_indicator_mode, fz.ts011f_plug_child_mode],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.ts011f_plug_indicator_mode, tz.ts011f_plug_child_mode],
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    options: [exposes.options.measurement_poll_interval()],
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage'),
        exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
            .withDescription('Plug LED indicator mode'), e.child_lock()],
    // onEvent: (type, data, device, options) =>
    //     tuya.onEventMeasurementPoll(type, data, device, options, true, device.applicationVersion === 160),
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

pauloon, 2022-11-20:
Update:
[image]

Always up! Hehehe

XanderTenBoden, 2022-11-21:
@Koenkk I will try this when I'm home after work and let you know!

[51 more replies]

closed 2022-11-29 by Koenkk