Koenkk/zigbee2mqtt #14859: [New device support]: Zigbee Tuya 1-63A Circuit Breaker TS0601 _TZE200_abatw3kj (ZMB9-Z-4P product)

Izram1, 2022-11-07:
### Link

https://gomedia.net.pl/pl/p/Bezpiecznik-Nadpradowy-Zigbee-4P-3-Fazy-Tuya-1-63A/750784

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0xa4c13820e5ef826c","nwkAddr":12676,"manufId":4417,"manufName":"_TZE200_abatw3kj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"/\u000b\u0000\u0000\u0016/\u000b\u0000\u0000\u0016\u0007\u0000\u0000\u0000\u0016j\b\u0000\u0000e/\t\u0000\u0000\u0016/\t\u0000\u0000e�\n\u0000\u0000\u0016�\n\u0000\u0000e","65506":34,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE200_abatw3kj","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d9c9ca","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1667736707347,"defaultSendRequestWhen":"immediate"}

### Comments

I followed the path suggested by Artpc fellow, but definitely this doesn't use all features of the device as the other one was a 1 phase breaker. As there are many devices like this I assume someone has already managed this. 

Below logs that are looking correct:
`debug 2022-11-06 11:52:19: Received MQTT message on 'homeassistant/switch/0xa4c13820e5ef826c/switch/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c13820e5ef826c/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xa4c13820e5ef826c/set","device":{"identifiers":["zigbee2mqtt_0xa4c13820e5ef826c"],"manufacturer":"SINOTIMER","model":"Circuit breaker 4P with meter (ZCB16-2P)","name":"0xa4c13820e5ef826c"},"name":"0xa4c13820e5ef826c","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0xa4c13820e5ef826c","unique_id":"0xa4c13820e5ef826c_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
debug 2022-11-06 11:52:19: Received MQTT message on 'homeassistant/sensor/0xa4c13820e5ef826c/voltage/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c13820e5ef826c/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c13820e5ef826c"],"manufacturer":"SINOTIMER","model":"Circuit breaker 4P with meter (ZCB16-2P)","name":"0xa4c13820e5ef826c"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","name":"0xa4c13820e5ef826c voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c13820e5ef826c","unique_id":"0xa4c13820e5ef826c_voltage_zigbee2mqtt","unit_of_measurement":"V","value_template":"{{ value_json.voltage }}"}'
debug 2022-11-06 11:52:19: Received MQTT message on 'homeassistant/sensor/0xa4c13820e5ef826c/power/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c13820e5ef826c/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c13820e5ef826c"],"manufacturer":"SINOTIMER","model":"Circuit breaker 4P with meter (ZCB16-2P)","name":"0xa4c13820e5ef826c"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","name":"0xa4c13820e5ef826c power","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c13820e5ef826c","unique_id":"0xa4c13820e5ef826c_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
debug 2022-11-06 11:52:19: Received MQTT message on 'homeassistant/sensor/0xa4c13820e5ef826c/current/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c13820e5ef826c/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c13820e5ef826c"],"manufacturer":"SINOTIMER","model":"Circuit breaker 4P with meter (ZCB16-2P)","name":"0xa4c13820e5ef826c"},"device_class":"current","enabled_by_default":false,"entity_category":"diagnostic","name":"0xa4c13820e5ef826c current","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c13820e5ef826c","unique_id":"0xa4c13820e5ef826c_current_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current }}"}'
debug 2022-11-06 11:52:19: Received MQTT message on 'homeassistant/sensor/0xa4c13820e5ef826c/energy/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c13820e5ef826c/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c13820e5ef826c"],"manufacturer":"SINOTIMER","model":"Circuit breaker 4P with meter (ZCB16-2P)","name":"0xa4c13820e5ef826c"},"device_class":"energy","enabled_by_default":true,"name":"0xa4c13820e5ef826c energy","state_class":"total_increasing","state_topic":"zigbee2mqtt/0xa4c13820e5ef826c","unique_id":"0xa4c13820e5ef826c_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
debug 2022-11-06 11:52:19: Received MQTT message on 'homeassistant/sensor/0xa4c13820e5ef826c/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c13820e5ef826c/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c13820e5ef826c"],"manufacturer":"SINOTIMER","model":"Circuit breaker 4P with meter (ZCB16-2P)","name":"0xa4c13820e5ef826c"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0xa4c13820e5ef826c linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c13820e5ef826c","unique_id":"0xa4c13820e5ef826c_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'` 

basically it shows Voltage (incorrectly), Power (unchecked), Current (unchecked), State with ON/OFF control (works fine). 

There is also a lot of errors, I assume different channels sending data, but cannot be recognized. 
`debug 2022-11-06 11:52:20: Received Zigbee message from '0xa4c13820e5ef826c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":16}],"seq":2304}' from endpoint 1 with groupID 0
info  2022-11-06 11:52:20: RECEIVED DP #16 -- VALUE = false
info  2022-11-06 11:52:20: MQTT publish: topic 'zigbee2mqtt/0xa4c13820e5ef826c', payload '{"current":null,"energy":null,"linkquality":78,"power":null,"state":"OFF","voltage":null}'
debug 2022-11-06 11:52:20: Received Zigbee message from '0xa4c13820e5ef826c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":1}],"seq":2560}' from endpoint 1 with groupID 0
info  2022-11-06 11:52:20: RECEIVED DP #1 -- VALUE = 0
info  2022-11-06 11:52:20: MQTT publish: topic 'zigbee2mqtt/0xa4c13820e5ef826c', payload '{"current":null,"energy":0,"linkquality":78,"power":null,"state":"OFF","voltage":null}'
debug 2022-11-06 11:52:20: Received Zigbee message from '0xa4c13820e5ef826c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":5,"dp":9}],"seq":2816}' from endpoint 1 with groupID 0
info  2022-11-06 11:52:20: RECEIVED DP #9 -- VALUE = 0
warn  2022-11-06 11:52:20: zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":2816,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
debug 2022-11-06 11:52:20: Received Zigbee message from '0xa4c13820e5ef826c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,96,0,0,0,0,0,0],"type":"Buffer"},"datatype":0,"dp":6}],"seq":3072}' from endpoint 1 with groupID 0
info  2022-11-06 11:52:20: RECEIVED DP #6 -- VALUE = 	     
[... cut ...]
    vendor: 'SINOTIMER',
    extend: extend.switch(),
    description: 'Circuit breaker 4P with meter',
    fromZigbee: [fzLocal.tuya_dinrail_switch, ],
    toZigbee: [tzLocal.state],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.switch().setAccess('state', ea.STATE_SET), e.voltage(), e.power(), e.current(), e.energy()],
};

module.exports = definition;
```

closed 2022-12-16 by github-actions[bot]

clumsy-stefan, 2024-03-06:
Anyone made any progress on this? I have a similar device which identifies as `_TZE200_wbhaespm`.

Any help would be gratly appreciated!

Izram1, 2024-03-07:
I returned the device 😉 that's my progress

clumsy-stefan, 2024-03-07:
;-)
At least I could get the switch working... I'm still trying to find the measurements (which is secondary to me)...

clumsy-stefan, 2024-03-08:
if someone's interested, I do have a working tempate which at least shows all reported values and enables the switching of the breaker. ALso I do have the original ZB-Protocol documentation with DP etc. from this device from the manufacturer (in chinese and machine-translated in english). But unfortunately my programming knowledge is limited... I don't succeed in omplementing any of the settable DP's... 

So, if anyone's intrested in developing this further, just let me know!

Template:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const extend = require('zigbee-herdsman-converters/lib/extend');
const legacy = require('zigbee-herdsman-converters/lib/legacy');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_wbhaespm',
    description: '3 Phase breaker with leakage protection and power meter',
    extend: [],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [  e.switch().setAccess('state', ea.STATE_SET),
                tuya.exposes.voltageWithPhase('X'), tuya.exposes.voltageWithPhase('Y'), tuya.exposes.voltageWithPhase('Z'),
                tuya.exposes.powerWithPhase('X'), tuya.exposes.powerWithPhase('Y'), tuya.exposes.powerWithPhase('Z'),
                tuya.exposes.currentWithPhase('X'), tuya.exposes.currentWithPhase('Y'), tuya.exposes.currentWithPhase('Z'),
                // Change the description according to the specifications of the device
                e.energy().withDescription('Total forward active energy'), e.temperature(),
                e.binary('trip', ea.STATE_SET, 'ON', 'OFF').withDescription('Trip'),
            ],
    meta: {
            tuyaDatapoints: [
                [1, 'energy', tuya.valueConverter.divideBy100],
                [6, null, tuya.valueConverter.phaseVariant2WithPhase('X')],
                [7, null, tuya.valueConverter.phaseVariant2WithPhase('Y')],
                [8, null, tuya.valueConverter.phaseVariant2WithPhase('Z')],
                [9, 'alarm', tuya.valueConverter.raw],
                [13, 'remaining', tuya.valueConverter.raw],
                [16, 'state', tuya.valueConverter.onOff],
                [17, 'alarm_set_1', tuya.valueConverter.raw],
                [18, 'alarm_set_2', tuya.valueConverter.raw],
                [19, 'serial', tuya.valueConverter.raw],
                [21, 'trip', tuya.valueConverterBasic.lookup({ 'ON': true, 'OFF': false })],
                [101, 'set', tuya.valueConverter.raw],
                [102, 'temperature', tuya.valueConverter.divideBy10],
            ],
        },
};

module.exports = definition;
```