Koenkk/zigbee2mqtt #14867: TV02-Zigbee: Disable preset_mode reset to auto when switching to system_mode heat

narendrastark, 2022-11-07:
### Is your feature request related to a problem? Please describe

I was used to control my TRVs by using home assistant's scheduler component alongside with the scheduler card. The scheduler component only allows to set different temperatures by choosing the system_mode, which means every time I change temperatures based on the time it triggers a message to set the system_mode to heat which resets the preset_mode to auto which overwrites the desired temperature.

### Describe the solution you'd like

Setting system_mode to heat should not reset the preset_mode.

### Describe alternatives you've considered

Reaching out to the scheduler component developers to enable just switching the temperature without switching or retriggering the system_mode.

### Additional context

n/a

vladyspavlov, 2022-11-09:
TRV resets preset mode by itself. Before, it was hardcoded in the converter to stay on manual preset. For now, the converter just follows the original device logic.

gomble, 2022-11-09:
> ### Is your feature request related to a problem? Please describe
> I was used to control my TRVs by using home assistant's scheduler component alongside with the scheduler card. The scheduler component only allows to set different temperatures by choosing the system_mode, which means every time I change temperatures based on the time it triggers a message to set the system_mode to heat which resets the preset_mode to auto which overwrites the desired temperature. That hasn't been the case few releases back and to be honest I do not really get the purpose of resetting the preset_mode when setting system_mode to heat. But I am happy to be taught a better lesson
> 
> ### Describe the solution you'd like
> Setting system_mode to heat should not reset the preset_mode.
> 
> ### Describe alternatives you've considered
> Reaching out to the scheduler component developers to enable just switching the temperature without switching or retriggering the system_mode. But setting temperature without operation mode may lead to the thermostat completely ignoring the command (if it is in operation mode off).
> 
> ### Additional context
> n/a

i have the same problem

Hessenpower01, 2022-11-13:
> TRV resets preset mode by itself. Before, it was hardcoded in the converter to stay on manual preset. For now, the converter just follows the original device logic.
> 
> I think it can be easily fixed with automation, which will be triggered on the `preset` change and take it back if necessary.

I have the same problem and also tried to solve it this way. Problem ist that the device also changes temperature when switching between "auto" and "manual".

vladyspavlov, 2022-11-13:
> I have the same problem and also tried to solve it this way. Problem ist that the device also changes temperature when switching between "auto" and "manual".

It's also an original device logic. And the same behavior with temperature was present before updating the converter, with hardcoded manual preset. There is no any changes in this part.

Hessenpower01, 2022-11-13:
> It's also an original device logic. And the same behavior with temperature was present before updating the converter, with hardcoded manual preset. There is no any changes in this part.

Thx for the info. Does this mean that the device is acting like this on it's own and that this behavior is unchangeable?

vladyspavlov, 2022-11-14:
> Thx for the info. Does this mean that the device is acting like this on it's own and that this behavior is unchangeable?

The full discussion can be found here https://github.com/Koenkk/zigbee-herdsman-converters/pull/4750.
In short, it's better to follow default TRV's logic than rewrite it to something else, and then anyone can adjust it to their own needs and wishes. Before making changes to the converter, I tested it with Tuya gateway, and, as far as I remember, it works in the same way (and quite honestly, it does make sense).

You can write your own logic in the converter and load it externally to the zigbee2mqtt, or create something on top of the current logic using your automation system. I chose the second way in my setup.

holantomas, 2022-11-15:
@vladyspavlov so if I get it right. You go and fu*ckup all others work? So now I can throw all my TRVs to the trash because I don't know nothing about z2m converters and don't have time to study it.

"You can write your own logic in the converter" ... if you want to have clean code, you can but than by your self, write back previous logic. People here using it. Now when here in EU temperatures start hitting 0 and lower, is great after needing to server reinstall see that somebody hyperactive f*cked my heating when I have small kid in house. Thank you for your work that no body wants. I just spend all night to solve this and now I read that somebody played ... thanks no sleep no heating for nothing. Something like BC you don't know?

swiewiora, 2022-11-20:
> @vladyspavlov so if I get it right. You go and fu*ckup all others work? So now I can throw all my TRVs to the trash because I don't know nothing about z2m converters and don't have time to study [...]

It is unbelievable how much entitlement some people can have to others work for which they didn't even pay. Why using free open source project and insult the authors? Justifying anger due to personal incompetence and shielding yourself using "small kid" flag is so disgusting. There are many ways to avoid this situation and it is your and your only fault that in your house there is no heating. Just stick to TuYa proprietary system and let's hope your kids won't end up being like you.

holantomas, 2022-11-20:
> It is unbelievable how much entitlement some people can have to others work for which they didn't even pay. Why using free open source project and insult the authors? Justifying anger due to [...]

And it standard for you to pay for work you didn't wanted? @vladi1234 did awesome work to get TRVs behave understandably and logically - no matter how ugly code looks. He then said to him, "don't do this" but he did it. Then you have internet full of people asking what happend, because changelog not saying anything about BC Break. Should all non-developer or non-js-dev or just parents which don't have time to check git diff of every OSS SW they using(this is why we have changelog or BC Break log) stuck on Tuya as you said?

As I said, he can do what he want, but there are ways to keep out of BC Break or notice users about deprecated things and how to achieve old behaving. The next part is playing with type of thing like that in winter is really bad idea. OSS is not about that all users have to be developers - in the end all SW is for users mainly. Why the hack we have final flags, encapsulation and other security options that are there to keep other users/devs out of internal logic when they have to be devs and know what they doing (that's your point right?)? After I saw this, it was easy to downgrade. But not everybody knows how to do it and are scared that will break rest of their home configuration - mainly there are no instructions how to downgrade safely. That's not pro-user's behaving and I hope no body have to work with you in dev team and don't need to use your SW if you have this type of thinking a.k.a. "it's OSS, so spend next 10 years to learn how to be Dev, how to DevOps and then fix it to your self".

vladi1234, 2022-11-20:
> And it standard for you to pay for work you didn't wanted? @vladi1234 did awesome work to get TRVs behave understandably and logically - no matter how ugly code looks. He then said to him, **"don't [...]

Thank you @holantomas  for the kind words,
I am very disappointed with these changes.
I also tried everything I could to avoid this problem, but nevertheless it happened.

vladi1234, 2022-11-23:
Hello everyone,

i tried to remove errors from the user code "vladyspavlov", but I also found that this does not make sense, since in total it has more than 10 errors embedded in the code, which makes our thermostat stupid.

I am planning to reanimate and integrate my extended code which is now bug free and includes new features.

Has a working `system_mode` for `Homeassistant`.
Similarly, it has a functional `schedule`.
Some other small corrections.

If you are interested, please raise your finger to convince the administrator.

Hessenpower01, 2022-11-23:
> Hello everyone,
>
> i tried to remove errors from the user code "vladyspavlov", but I also found that this does not make sense, since in total it has more than 10 errors embedded in the code, which [...]

Finger is raised as high as possible

Koenkk, 2022-11-24:
Lets discuss this further in #15164

closed 2022-11-24 by Koenkk

scalt, 2022-12-08:
Another user impacted by this change. There's no logic in setting the valve to auto if a user sets it to something else. This cannot be changed by automation as temperature goes back to a schedule when entering the auto mode.

Please keep mode in whatever the user choose without forcing it back to auto after turning on heating. 

Tx!