Koenkk/zigbee2mqtt #14868: [Feature request]: Network Key and Pan ID auto generated during install to minimize insecure installs

bharvey88, 2022-11-07:
### Is your feature request related to a problem? Please describe

While learning more about zigbee and z2m in the home assistant discord server I have been made aware that my install is far from secure. Per: https://www.zigbee2mqtt.io/advanced/zigbee/03_secure_network.html#change-zigbee-network-encryption-key it says the install uses a default encryption key and pan ID. It also states we are unable to generate the key ourselves when using the addon inside home assistant OS. I find this default install behavior to be very worrying and I request that there be a way to automatically generate both a random network key and pan ID to make this a noob-friendly installation experience.

### Describe the solution you'd like

Automatically generate randomized network key and pan ID during installation process.

### Describe alternatives you've considered

Alternatively, a **required** set of options for both network key and pan ID in the configuration tab with accompanying documentation in the documentation tab of the addon.

### Additional context

No additional context to give, let me know if you want any added information.

MattWestb, 2022-11-08:
Also Extended PAN-ID need being unique if having more network in the radio distance or can getting very strange things happening with the networks.

closed 2022-12-17 by github-actions[bot]

bharvey88, 2023-01-15:
bump this feature request is still wanted