Koenkk/zigbee2mqtt #14872: MOES BRT-100-TRV - Set point value with decimal

bamnoru, 2022-11-08:
### What happened?

When changing on the device the set point value, it's possible to increase/decrease the value by 0.5 step.
But, the current setpoint value is rounded without decimal.

### What did you expect to happen?

The set point value should be read with decimal

### How to reproduce it (minimal and precise)

Change the set point value on the device with a decimal (18.5 for example), and read the value current_heating_setpoint.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20211116

### Adapter

CC2531

### Debug log

Setting the temperature to 18.5 :

info  2022-11-08 08:58:51: MQTT publish: topic 'zigbee2mqtt/Radiateur SdB', payload '{"battery":97,"boost_heating":"OFF","boost_heating_countdown":0,"boost_heating_countdown_time_set":300,"child_lock":"UNLOCK","current_heating_setpoint":22,"eco_mode":"OFF","eco_temperature":18,"linkquality":47,"local_temperature":20.5,"local_temperature_calibration":-1,"max_temperature":25,"min_temperature":15,"position":25,"preset":"manual","programming_mode":"06:00/20°C  11:30/21°C  13:30/22°C  17:30/23°C  06:00/24°C  12:00/23°C  14:30/22°C  17:30/21°C  06:00/19°C  12:30/20°C  14:30/21°C  18:30/20°C","running_state":"heat","system_mode":"heat","update":{"state":"idle"},"update_available":false,"valve_state":"OPEN","window":"CLOSED","window_detection":"ON"}'

debug 2022-11-08 08:59:13: Received Zigbee message from 'Radiateur SdB', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,18],"type":"Buffer"},"datatype":2,"dp":2}],"seq":50}' from endpoint 1 with groupID 0

info  2022-11-08 08:59:13: MQTT publish: topic 'zigbee2mqtt/Radiateur SdB', payload '{"battery":97,"boost_heating":"OFF","boost_heating_countdown":0,"boost_heating_countdown_time_set":300,"child_lock":"UNLOCK","current_heating_setpoint":18,"eco_mode":"OFF","eco_temperature":18,"linkquality":47,"local_temperature":20.5,"local_temperature_calibration":-1,"max_temperature":25,"min_temperature":15,"position":25,"preset":"manual","programming_mode":"06:00/20°C  11:30/21°C  13:30/22°C  17:30/23°C  06:00/24°C  12:00/23°C  14:30/22°C  17:30/21°C  06:00/19°C  12:30/20°C  14:30/21°C  18:30/20°C","running_state":"heat","system_mode":"heat","update":{"state":"idle"},"update_available":false,"valve_state":"OPEN","window":"CLOSED","window_detection":"ON"}'

Ivano62, 2022-11-09:
With physical input on the device the step is 0.5 °C

xworld21, 2022-11-10:
Unfortunately those 0.5 values are not exposed over Zigbee. From the point of view of Zigbee2MQTT, the valve supports integer values only, so unless the firmware changes behaviour (to e.g. accept or report 2*temperature instead of temperature), the issue cannot be fixed by Z2M.

Ivano62, 2022-11-13:
Considering that the OTA update with Z2M seems not to work properly, is there another possibility to check for the availability of a new firmware version?

closed 2022-11-17 by bamnoru