Koenkk/zigbee2mqtt #14884: Aqara P1  (MS -S02 or RTCGQ14LM) - all relevant data isn't coming through

Jens-Wymeersch, 2022-11-08:
### What happened?

I just bought 5 new P1 Aqara motion sensors (https://www.zigbee2mqtt.io/devices/RTCGQ14LM.html). 
The device(s) pair normally and are recognised correctly. However, occupancy, illumination, motion_sensitivity, detection_interval, trigger_indication and device_temperature are N/A or the button doesn't work.

[image]

The values that are visible are 

[image]

### What did you expect to happen?

I expected that all values would be visible, instead of a few that have hardly any meaning. 

### How to reproduce it (minimal and precise)

As I have 5 of these devices, I tried with at least 2 - same result and tried it with my demo environment. Also resulting in the same outcome.

### Zigbee2MQTT version

1.28.2 

### Adapter firmware version

20220219

### Adapter

ZZH!

### Debug log

```
info  2022-11-08 17:05:34: Starting interview of '0x54ef4410004e37c4'
info  2022-11-08 17:05:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410004e37c4","ieee_address":"0x54ef4410004e37c4","status":"started"},"type":"device_interview"}'
debug 2022-11-08 17:05:36: Device '0x54ef4410004e37c4' announced itself
info  2022-11-08 17:05:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410004e37c4","ieee_address":"0x54ef4410004e37c4"},"type":"device_announce"}'
debug 2022-11-08 17:05:37: Received MQTT message on 'homeassistant/binary_sensor/0x54ef4410004e37c4/occupancy/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x54ef4410004e37c4"],"manufacturer":"Xiaomi","model":"Aqara P1 human body movement and illuminance sensor (RTCGQ14LM)","name":"0x54ef4410004e37c4","sw_version":"0.0.0_0005"},"device_class":"motion","name":"0x54ef4410004e37c4 occupancy","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x54ef4410004e37c4","unique_id":"0x54ef4410004e37c4_occupancy_zigbee2mqtt","value_template":"{{ value_json.occupancy }}"}'
debug 2022-11-08 17:05:37: Received MQTT message on 'homeassistant/sensor/0x54ef4410004e37c4/illuminance_lux/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x54ef4410004e37c4"],"manufacturer":"Xiaomi","model":"Aqara P1 human body movement and illuminance sensor (RTCGQ14LM)","name":"0x54ef4410004e37c4","sw_version":"0.0.0_0005"},"device_class":"illuminance","enabled_by_default":true,"name":"0x54ef4410004e37c4 illuminance lux","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef4410004e37c4","unique_id":"0x54ef4410004e37c4_illuminance_lux_zigbee2mqtt","unit_of_measurement":"lx","value_template":"{{ value_json.illuminance }}"}'
debug 2022-11-08 17:05:37: Received MQTT message on 'homeassistant/sensor/0x54ef4410004e37c4/illuminance/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x54ef4410004e37c4"],"manufacturer":"Xiaomi","model":"Aqara P1 human body movement and illuminance sensor (RTCGQ14LM)","name":"0x54ef4410004e37c4","sw_version":"0.0.0_0005"},"device_class":"illuminance","enabled_by_default":false,"name":"0x54ef4410004e37c4 illuminance","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef4410004e37c4","unique_id":"0x54ef4410004e37c4_illuminance_zigbee2mqtt","unit_of_measurement":"lx","value_template":"{{ value_json.illuminance }}"}'
debug 2022-11-08 17:05:37: Received MQTT message on 'homeassistant/sensor/0x54ef4410004e37c4/motion_sensitivity/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x54ef4410004e37c4"],"manufacturer":"Xiaomi","model":"Aqara P1 human body movement and illuminance sensor (RTCGQ14LM)","name":"0x54ef4410004e37c4","sw_version":"0.0.0_0005"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:tune","name":"0x54ef4410004e37c4 motion sensitivity","state_topic":"zigbee2mqtt/0x54ef4410004e37c4","unique_id":"0x54ef4410004e37c4_motion_sensitivity_zigbee2mqtt","value_template":"{{ value_json.motion_sensitivity }}"}'
debug 2022-11-08 17:05:37: Received MQTT message on 'homeassistant/select/0x54ef4410004e37c4/motion_sensitivity/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x54ef4410004e37c4/set/motion_sensitivity","device":{"identifiers":["zigbee2mqtt_0x54ef4410004e37c4"],"manufacturer":"Xiaomi","model":"Aqara P1 human body movement and illuminance sensor (RTCGQ14LM)","name":"0x54ef4410004e37c4","sw_version":"0.0.0_0005"},"entity_category":"config","icon":"mdi:tune","name":"0x54ef4410004e37c4 motion sensitivity","options":["low","medium","high"],"state_topic":"zigbee2mqtt/0x54ef4410004e37c4","unique_id":"0x54ef4410004e37c4_motion_sensitivity_zigbee2mqtt","value_template":"{{ value_json.motion_sensitivity }}"}'
debug 2022-11-08 17:05:37: Received MQTT message on 'homeassistant/sensor/0x54ef4410004e37c4/detection_interval/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x54ef4410004e37c4"],"manufacturer":"Xiaomi","model":"Aqara P1 human body movement and illuminance sensor (RTCGQ14LM)","name":"0x54ef4410004e37c4","sw_version":"0.0.0_0005"},"enabled_by_default":false,"name":"0x54ef4410004e37c4 detection interval","state_topic":"zigbee2mqtt/0x54ef4410004e37c4","unique_id":"0x54ef4410004e37c4_detection_interval_zigbee2mqtt","unit_of_measurement":"s","value_template":"{{ value_json.detection_interval }}"}'
debug 2022-11-08 17:05:37: Received MQTT message on 'homeassistant/number/0x54ef4410004e37c4/detection_interval/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x54ef4410004e37c4/set/detection_interval","device":{"identifiers":["zigbee2mqtt_0x54ef4410004e37c4"],"manufacturer":"Xiaomi","model":"Aqara P1 human body movement and illuminance sensor (RTCGQ14LM)","name":"0x54ef4410004e37c4","sw_version":"0.0.0_0005"},"max":65535,"min":2,"name":"0x54ef4410004e37c4 detection interval","state_topic":"zigbee2mqtt/0x54ef4410004e37c4","unique_id":"0x54ef4410004e37c4_detection_interval_zigbee2mqtt","unit_of_measurement":"s","value_template":"{{ value_json.detection_interval }}"}'
debug 2022-11-08 17:05:37: Received MQTT message on 'homeassistant/switch/0x54ef4410004e37c4/switch_trigger_indicator/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x54ef4410004e37c4/set/trigger_indicator","device":{"identifiers":["zigbee2mqtt_0x54ef4410004e37c4"],"manufacturer":"Xiaomi","model":"Aqara P1 human body movement and illuminance sensor (RTCGQ14LM)","name":"0x54ef4410004e37c4","sw_version":"0.0.0_0005"},"name":"0x54ef4410004e37c4 trigger_indicator","payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/0x54ef4410004e37c4","unique_id":"0x54ef4410004e37c4_switch_trigger_indicator_zigbee2mqtt","value_template":"{% if value_json.trigger_indicator %} true {% else %} false {% endif %}"}'
debug 2022-11-08 17:05:37: Received MQTT message on 'homeassistant/sensor/0x54ef4410004e37c4/device_temperature/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x54ef4410004e37c4"],"manufacturer":"Xiaomi","model":"Aqara P1 human body movement and illuminance sensor (RTCGQ14LM)","name":"0x54ef4410004e37c4","sw_version":"0.0.0_0005"},"device_class":"temperature","enabled_by_default":true,"entity_category":"diagnostic","name":"0x54ef4410004e37c4 device temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef4410004e37c4","unique_id":"0x54ef4410004e37c4_device_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.device_temperature }}"}'
debug 2022-11-08 17:05:37: Received MQTT message on 'homeassistant/sensor/0x54ef4410004e37c4/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x54ef4410004e37c4"],"manufacturer":"Xiaomi","model":"Aqara P1 human body movemen [... cut ...]
debug 2022-11-08 17:05:39: Received Zigbee message from '0x54ef4410004e37c4', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID 0
info  2022-11-08 17:05:39: MQTT publish: topic 'zigbee2mqtt/0x54ef4410004e37c4', payload '{"battery":100,"detection_interval":null,"device_temperature":null,"illuminance":null,"linkquality":185,"motion_sensitivity":null,"occupancy":null,"trigger_indicator":null,"update":{"state":null},"update_available":null,"voltage":3000}'
```

Jens-Wymeersch, 2022-11-09:
@Koenkk Can you please advice what to do in this case ?

Koenkk, 2022-11-09:
I don't have experience with these device but I know they require a newer adapter firmware. You are running the correct one but it could it be that you paired these devices while still on an older fw? ( <= 20210708) . If yes then try repairing them.

Jens-Wymeersch, 2022-11-09:
@Koenkk I bought the devices a few weeks ago. I have 2 environments currently (both at the current firmware version) as I'm planning to move from one to another. In both I have the same problem... Also I tested multiple devices. 
Anything that can be done ? Maybe in the recent months they changed slightly.

Koenkk, 2022-11-10:
Can you try to reconfigure them? (click the yellow refresh icon on the frontend, it might be needed to wake them up short before doing the reconfigure by pressing a button on the sensor)

Jens-Wymeersch, 2022-11-10:
@Koenkk thanks for your quick reply. Unfortunately not resulting in success. 
At some point this error popped up : 

```
error 2022-11-10 18:46:39: Failed to configure '0x54ef44100053de26', attempt 1 (Error: Read 0x54ef44100053de26/1 genPowerCfg(["batteryVoltage"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 8894 - 1 - 211 - 1 - 1 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7))
info  2022-11-10 18:46:39: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x54ef44100053de26"},"error":"Failed to configure (Read 0x54ef44100053de26/1 genPowerCfg([\"batteryVoltage\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 8894 - 1 - 211 - 1 - 1 after 10000ms))","status":"error","transaction":"b1ek9-1"}'
```

Koenkk, 2022-11-11:
Can you try to re-pair the device close to the coordinator and try again?

Jens-Wymeersch, 2022-11-13:
@Koenkk Thank you for your reply. The devices are less then 1m away. Any other ideas ?

Koenkk, 2022-11-13:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.motion.ac02'],
    model: 'RTCGQ14LM',
    vendor: 'Xiaomi',
    whiteLabel: [{vendor: 'Xiaomi', model: 'MS-S02'}],
    description: 'Aqara P1 human body movement and illuminance sensor',
    fromZigbee: [fz.aqara_occupancy_illuminance, fz.aqara_opple, fz.battery],
    toZigbee: [tz.aqara_detection_interval, tz.aqara_motion_sensitivity, tz.RTCGQ14LM_trigger_indicator],
    exposes: [e.occupancy(), e.illuminance_lux().withProperty('illuminance'),
        e.illuminance().withUnit('lx').withDescription('Measured illuminance in lux'),
        exposes.enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high'])
            .withDescription('. Press pairing button right before changing this otherwise it will fail.'),
        exposes.numeric('detection_interval', ea.ALL).withValueMin(2).withValueMax(65535).withUnit('s')
            .withDescription('Time interval for detecting actions. ' +
                'Press pairing button right before changing this otherwise it will fail.'),
        exposes.binary('trigger_indicator', ea.ALL, true, false).withDescription('When this option is enabled then ' +
            'blue LED will blink once when motion is detected. ' +
            'Press pairing button right before changing this otherwise it will fail.'),
        e.device_temperature(), e.battery(), e.battery_voltage()],
    meta: {battery: {voltageToPercentage: '3V_2850_3000'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        // await endpoint.read('genPowerCfg', ['batteryVoltage']);
        await endpoint.read('aqaraOpple', [0x0102], {manufacturerCode: 0x115f});
        await endpoint.read('aqaraOpple', [0x010c], {manufacturerCode: 0x115f});
        await endpoint.read('aqaraOpple', [0x0152], {manufacturerCode: 0x115f});
    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Jens-Wymeersch, 2022-11-13:
@Koenkk Thank you for your reply. I'm getting the following error after adding the converter via the UI. 

```
/app/node_modules/zigbee-herdsman-converters/index.js:78
    definition.toZigbee.push(tz.scene_store, tz.scene_recall, tz.scene_add, tz.scene_remove, tz.scene_remove_all, tz.read, tz.write,
                        ^
TypeError: Cannot read properties of undefined (reading 'push')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:78:25)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:108:18)
```

Jens-Wymeersch, 2022-11-13:
@Koenkk based on conversation on discord, i see that the current supported firmware is Firmware _0.0.00005 dated Dec 7 2021 and not what I have Jun 9 2022 | 0.0.0_0006.
Does this help ?

Jens-Wymeersch, 2022-11-14:
@Koenkk I see some improvement somehow. The reading from the device(s) moved from N/A to Null. Still no reading but at least the state shows all fields. Now getting the reading out of the device ?

Koenkk, 2022-11-14:
Can you confirm that the contents of `ext_converter.js` exactly match https://github.com/Koenkk/zigbee2mqtt/issues/14884#issuecomment-1312741386 ? I just tested this converter and it does not throw this error in my case.

Jens-Wymeersch, 2022-11-15:
@Koenkk thanks for your reply. I finally was able to get the converter working but still didn't get the hoped result. Here are the logs from the moment I re-paired the device

```
info  2022-11-15 07:29:53: Accepting joining not in blocklist device '0x54ef44100053de26'
info  2022-11-15 07:29:53: Device '0x54ef44100053de26' joined
info  2022-11-15 07:29:55: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef44100053de26","ieee_address":"0x54ef44100053de26"},"type":"device_joined"}'
info  2022-11-15 07:29:55: Starting interview of '0x54ef44100053de26'
info  2022-11-15 07:29:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef44100053de26","ieee_address":"0x54ef44100053de26","status":"started"},"type":"device_interview"}'
debug 2022-11-15 07:29:56: Device '0x54ef44100053de26' announced itself
info  2022-11-15 07:29:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef44100053de26","ieee_address":"0x54ef44100053de26"},"type":"device_announce"}'
debug 2022-11-15 07:29:58: Received Zigbee message from 'Presence Sensor Living', type 'attributeReport', cluster 'aqaraOpple', data '{"322":1}' from endpoint 1 with groupID 0
debug 2022-11-15 07:29:58: lumi.motion.ac01: Processed data into payload {"presence":true}
info  2022-11-15 07:29:58: MQTT publish: topic 'zigbee2mqtt/Presence Sensor Living', payload '{"approach_distance":"medium","device_temperature":28,"linkquality":51,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":6,"presence":true,"presence_event":"enter","update":{"state":"idle"},"update_available":null}'
debug 2022-11-15 07:29:58: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.ac02"}' from endpoint 1 with groupID 0
debug 2022-11-15 07:29:58: Skipping message, definition is undefined and still interviewing
debug 2022-11-15 07:29:58: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
debug 2022-11-15 07:29:58: Skipping message, definition is undefined and still interviewing
debug 2022-11-15 07:29:58: Received Zigbee message from 'Presence Sensor Living', type 'attributeReport', cluster 'aqaraOpple', data '{"323":6}' from endpoint 1 with groupID 0
debug 2022-11-15 07:29:58: lumi.motion.ac01: Processed data into payload {"presence_event":"approach"}
info  2022-11-15 07:29:58: MQTT publish: topic 'zigbee2mqtt/Presence Sensor Living', payload '{"approach_distance":"medium","device_temperature":28,"linkquality":51,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":6,"presence":true,"presence_event":"approach","update":{"state":"idle"},"update_available":null}'
debug 2022-11-15 07:29:59: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug 2022-11-15 07:29:59: Skipping message, definition is undefined and still interviewing
debug 2022-11-15 07:29:59: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2022-11-15 07:29:59: Skipping message, definition is undefined and still interviewing
debug 2022-11-15 07:29:59: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'genBasic', data '{"appVersion":6}' from endpoint 1 with groupID 0
debug 2022-11-15 07:29:59: Skipping message, definition is undefined and still interviewing

debug 2022-11-15 07:29:59: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2022-11-15 07:29:59: Skipping message, definition is undefined and still interviewing
debug 2022-11-15 07:29:59: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2022-11-15 07:29:59: Skipping message, definition is undefined and still interviewing
debug 2022-11-15 07:29:59: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'genBasic', data '{"dateCode":"Jun  9 2022"}' from endpoint 1 with groupID 0
debug 2022-11-15 07:29:59: Skipping message, definition is undefined and still interviewing
debug 2022-11-15 07:29:59: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"0.0.0_0006"}' from endpoint 1 with groupID 0
debug 2022-11-15 07:29:59: Skipping message, definition is undefined and still interviewing
info  2022-11-15 07:29:59: Successfully interviewed '0x54ef44100053de26', device has successfully been paired
info  2022-11-15 07:29:59: Device '0x54ef44100053de26' is supported, identified as: Xiaomi Aqara P1 human body movement and illuminance sensor (RTCGQ14LM)
info  2022-11-15 07:30:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara P1 human body movement and illuminance sensor","exposes":[{"access":1,"description":"Indicates whether the device detected occupancy","name":"occupancy","property":"occupancy","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Measured illuminance in lux","name":"illuminance_lux","property":"illuminance","type":"numeric","unit":"lx"},{"access":1,"de [... cut ...] able":null,"voltage":null}'

debug 2022-11-15 07:30:03: lumi.motion.ac02: Processed data into payload {"motion_sensitivity":"medium"}
info  2022-11-15 07:30:03: MQTT publish: topic 'zigbee2mqtt/0x54ef44100053de26', payload '{"battery":null,"detection_interval":30,"device_temperature":null,"illuminance":null,"linkquality":185,"motion_sensitivity":"medium","occupancy":null,"trigger_indicator":null,"update":{"state":null},"update_available":null,"voltage":null}'
```
Any ideas what to do from here ?

Jens-Wymeersch, 2022-11-15:
@Koenkk Attached you find the logs. 

```
info  2022-11-15 21:32:14: Accepting joining not in blocklist device '0x54ef44100053de26'
debug 2022-11-15 21:32:14: Device '0x54ef44100053de26' announced itself
info  2022-11-15 21:32:15: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef44100053de26","ieee_address":"0x54ef44100053de26"},"type":"device_announce"}'

debug 2022-11-15 21:32:17: Retrieving state of '0x54ef44100053de26' after reconnect

debug 2022-11-15 21:32:47: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"0x54ef44100053de26","transaction":"pwgfj-3"}'
info  2022-11-15 21:32:47: Configuring '0x54ef44100053de26'

error 2022-11-15 21:33:10: Failed to configure '0x54ef44100053de26', attempt 1 (Error: Read 0x54ef44100053de26/1 aqaraOpple([258], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 40397 - 1 - 176 - 64704 - 1 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7))
info  2022-11-15 21:33:10: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x54ef44100053de26"},"error":"Failed to configure (Read 0x54ef44100053de26/1 aqaraOpple([258], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 40397 - 1 - 176 - 64704 - 1 after 10000ms))","status":"error","transaction":"pwgfj-3"}'

```

Jens-Wymeersch, 2022-11-17:
@Koenkk I'm only able to do a long press (re-pairing mode) - I got this error following re-configuration after long press

I'm less than 50 cm from the coordinator, but still the same result. What I do see is that the interview process is extremely quick.

Koenkk, 2022-11-17:
@Jens-Wymeersch to do the full interview, force remove the device through the z2m frontend, restart z2m and re-pair.

Jens-Wymeersch, 2022-11-19:
@Koenkk here are the results but no success
```
info  2022-11-19 10:37:15: Accepting joining not in blocklist device '0x54ef44100053de26'
debug 2022-11-19 10:37:15: Device '0x54ef44100053de26' announced itself
info  2022-11-19 10:37:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef44100053de26","ieee_address":"0x54ef44100053de26"},"type":"device_announce"}'
debug 2022-11-19 10:37:18: Retrieving state of '0x54ef44100053de26' after reconnect

debug 2022-11-19 10:37:26: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"0x54ef44100053de26","transaction":"g6h29-2"}'
info  2022-11-19 10:37:26: Configuring '0x54ef44100053de26'
debug 2022-11-19 10:37:26: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'aqaraOpple', data '{"258":30}' from endpoint 1 with groupID 0
debug 2022-11-19 10:37:26: lumi.motion.ac02: Processed data into payload {"detection_interval":30}
info  2022-11-19 10:37:26: MQTT publish: topic 'zigbee2mqtt/0x54ef44100053de26', payload '{"battery":null,"detection_interval":30,"device_temperature":null,"illuminance":null,"linkquality":185,"motion_sensitivity":"medium","occupancy":null,"trigger_indicator":false,"update":{"state":null},"update_available":null,"voltage":null}'
debug 2022-11-19 10:37:26: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'aqaraOpple', data '{"268":2}' from endpoint 1 with groupID 0
debug 2022-11-19 10:37:26: lumi.motion.ac02: Processed data into payload {"motion_sensitivity":"medium"}
info  2022-11-19 10:37:26: MQTT publish: topic 'zigbee2mqtt/0x54ef44100053de26', payload '{"battery":null,"detection_interval":30,"device_temperature":null,"illuminance":null,"linkquality":185,"motion_sensitivity":"medium","occupancy":null,"trigger_indicator":false,"update":{"state":null},"update_available":null,"voltage":null}'
debug 2022-11-19 10:37:26: Received Zigbee message from '0x54ef44100053de26', type 'readResponse', cluster 'aqaraOpple', data '{"338":0}' from endpoint 1 with groupID 0
debug 2022-11-19 10:37:26: lumi.motion.ac02: Processed data into payload {"trigger_indicator":false}
info  2022-11-19 10:37:26: MQTT publish: topic 'zigbee2mqtt/0x54ef44100053de26', payload '{"battery":null,"detection_interval":30,"device_temperature":null,"illuminance":null,"linkquality":185,"motion_sensitivity":"medium","occupancy":null,"trigger_indicator":false,"update":{"state":null},"update_available":null,"voltage":null}'
info  2022-11-19 10:37:26: Successfully configured '0x54ef44100053de26'

```

Jens-Wymeersch, 2022-11-20:
@Koenkk Thank you for all your help. After my migration and after a lot of indescribable errors, I switched from zzh! stick. 

Result : not only I was able to pair a few devices that I was struggling with the whole day but also those damn motion sensors.

Thank you again for your constant support on this and many other cases. Z2M is truly a great solution and is the core of my home automation platform for now and in the future !

closed 2022-11-20 by Jens-Wymeersch