Koenkk/zigbee2mqtt #14888: [New device support]:

johnnydabbbb, 2022-11-08:
### Link

https://www.aqara.com/smart-wall-switch-H1-with-neutral_spec

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x04cf8cdf3c848a45","nwkAddr":14970,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.rkna01","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9,1794,2820],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"zclVersion":3}},"aqaraOpple":{"attributes":{"554":3}}},"binds":[{"cluster":10,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b74af2","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":24,"stackVersion":2,"hwVersion":1,"dateCode":"11-26-2021","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-983567100},"lastSeen":1667929159187,"defaultSendRequestWhen":"immediate"}

### Comments

Below is what I have done:

Searched on supported device page and found that only ZNXNKG02LM (Wireless) version is supported instead of this ZNXNKG01LM (Wired with Neutral) version.

Tried following the support new device page, now running the dev branch, with reference to the already supported wireless version's definition and custom external converters codes in xiaomi.js. Spent a couple days copy and pasting + typing.

Result:

Managed to get the device paired and get most of the functions (those the same as the wireless version) exposed however there is no response shown on "exposes" and "state" page of the device. All these functions I tried to expose are shown as "unavailable" on HA MQTT Integration. I simply dont see any response/debug log from the MQTT console when actions are triggered on the device.

I am not an IT professional, with limited programming knowledge I cannot get this device to work. I have many Aqara devices linked to an Aqara M2 hub as well as homekit originally and currently trying to port everything into HA using Zigbee2MQTT with a Sonoff dongle so that everything is centralized and more advanced automations could be programmed. This is actually the only Aqara device I have at home that is currently not supported out of the box. I have 5-6 of this installed in my whole place, it would be great if I could have this last piece of the puzzle solved to have a 100% working Smart Home Solution. Please help me :'-(

johnnydabbbb, 2022-11-08:
Here is what I have in ZNXNKG01LM.js:

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const constants = require('zigbee-herdsman-converters/lib/constants');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');
const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');

const xiaomiExtend = {
    light_onoff_brightness_colortemp: (options={disableColorTempStartup: true}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        fromZigbee: extend.light_onoff_brightness_colortemp(options).fromZigbee.concat([
            fz.xiaomi_bulb_interval, fz.ignore_occupancy_report, fz.ignore_humidity_report,
            fz.ignore_pressure_report, fz.ignore_temperature_report,
        ]),
    }),
};

const preventReset = async (type, data, device) => {
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
    await device.getEndpoint(1).write('genBasic', payload, options);
};

const fzLocal = {
    aqara_trv: {
        cluster: 'aqaraOpple',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            Object.entries(msg.data).forEach(([key, value]) => {
                switch (parseInt(key)) {
                case 0x0271:
                    result['system_mode'] = {1: 'heat', 0: 'off'}[value];
                    break;
                case 0x0272:
                    result['preset'] = {2: 'away', 1: 'auto', 0: 'manual'}[value];
                    break;
                case 0x0273:
                    result['window_detection'] = {1: 'ON', 0: 'OFF'}[value];
                    break;
                case 0x0274:
                    result['valve_detection'] = {1: 'ON', 0: 'OFF'}[value];
                    break;
                case 0x0277:
                    result['child_lock'] = {1: 'LOCK', 0: 'UNLOCK'}[value];
                    break;
                case 0x0279:
                    result['away_preset_temperature'] = (value / 100).toFixed(1);
                    break;
[... 169 lines cut ...]

const definition = {
    zigbeeModel: ['lumi.switch.rkna01'], // The model ID from: Device with modelID 'lumi.remote.rkba01' is not supported.
    model: 'ZNXNKG01LM', // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi', // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara Smart Knob H1 (With Neutral)', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.xiaomi_on_off_action, fz.xiaomi_multistate_action, fz.xiaomi_basic, fz.aqara_opple, fz.aqara_knob_rotation], // We will add this later
    toZigbee: [tz.aqara_opple_operation_mode], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.battery_voltage(),e.action(['single', 'double', 'hold', 'release', 'start_rotating', 'rotation', 'stop_rotating']),
            exposes.enum('operation_mode', ea.ALL, ['event', 'command']).withDescription('Button mode'),
            exposes.numeric('action_rotation_angle', ea.STATE).withUnit('*').withDescription('Rotation angle'),
            exposes.numeric('action_rotation_angle_speed', ea.STATE).withUnit('*').withDescription('Rotation angle speed'),
            exposes.numeric('action_rotation_percent', ea.STATE).withUnit('%').withDescription('Rotation percent'),
            exposes.numeric('action_rotation_percent_speed', ea.STATE).withUnit('%').withDescription('Rotation percent speed'),
            exposes.numeric('action_rotation_time', ea.STATE).withUnit('ms').withDescription('Rotation time'),
             ], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
    },
    
};

module.exports = definition;

closed 2022-12-17 by github-actions[bot]

ecointet, 2024-03-31:
support for this product (aqara knob with neutral) would be great for dimmable lights.

nutbox76, 2025-03-23:
in the aqara app you can debound the 3 channels so the relay doesnt activate when pressed once is this possible in Z2M ,i know other aqara switches have the decoupled option so its got to be aint it?
Thank you