Koenkk/zigbee2mqtt #14890: Zigbee2MQTT Addon with Sonoff Dongle won’t start

wilwizard1979, 2022-11-08:
### What happened?

Hi, I have flashed the latest firmware into a new SONOFF Dongle and followed the instructions to install and configure Zigbee2MQTT but I got the following log when trying to start the Add-On:

> Zigbee2MQTT
[21:46:54] INFO: Preparing to start...
[21:46:55] INFO: Socat not enabled
[21:46:55] INFO: Zigbee Herdsman debug logging enabled
[21:46:56] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/devices/icasa.js:1
<
^
SyntaxError: Unexpected token '<'
    at Object.compileFunction (node:vm:360:18)
    at wrapSafe (node:internal/modules/cjs/loader:1055:15)
    at Module._compile (node:internal/modules/cjs/loader:1090:27)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1180:10)
    at Module.load (node:internal/modules/cjs/loader:1004:32)
    at Function.Module._load (node:internal/modules/cjs/loader:839:12)
    at Module.require (node:internal/modules/cjs/loader:1028:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/app/node_modules/zigbee-herdsman-converters/index.js:118:30)
    at Module._compile (node:internal/modules/cjs/loader:1126:14)`

I'm using Sonoff 3.0 USB Dongle Plus, Chipset CC2652.
Upgrade was done to the latest version available in your repository

This is the information I can see in All Hardware option from Home Assistant:

> Subsystem:
tty
Device path:
/dev/ttyUSB0
ID:
/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0
Attributes:
DEVLINKS: >-
  /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0
  /dev/serial/by-path/pci-0000:00:06.0-usb-0:2:1.0-port0

I posted this first in Home Assistant comminity and I thought it would be the right time to report as an issue here. In the Thread below more information of the investigation can be found:
[https://community.home-assistant.io/t/zigbee2mqtt-addon-with-sonoff-dongle-wont-start/484867/7](url)

Thanks a lot in advance for your help

### What did you expect to happen?

I expected that the Add-On would normally start

### How to reproduce it (minimal and precise)

Not sure if it is an issue that can be reproduced in other environment.

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P CC2652

### Debug log

> info  2022-10-28 23:02:27: Logging to console and directory: '/config/zigbee2mqtt/log/2022-10-28.23-02-27' filename: log.txt
info  2022-10-28 23:02:28: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
info  2022-10-28 23:02:28: Starting zigbee-herdsman (0.14.62)
error 2022-10-28 23:02:30: Error while starting zigbee-herdsman
error 2022-10-28 23:02:30: Failed to start zigbee
error 2022-10-28 23:02:30: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2022-10-28 23:02:30: Exiting...
error 2022-10-28 23:02:30: Error: spawn udevadm ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:283:19)
    at onErrorNT (node:internal/child_process:478:16)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)

wilwizard1979, 2022-11-08:
Hi, I just noticed that the message I receive from the Log is the same if I have the Dongle connected or not. So this gives me the impression that the Dongle is not seen by HA, but this is not the case when I see it listed in the All Hardware:

[image]

The same disappears when I disconnect the Dongle. So I'm not sure.

It is the second time I think that there may be a hardware issue. The first was the fact that it was never showing up in the Integrations as a "Discovered" device. I just would like to have more evidences in order to ask for a replacement.

mdehaas, 2022-11-08:
If you insert the dongle and restart HA. Is the dongle discovered by ZHA? If so the dongle is most likely not the problem. 
I see debugging is on. Just to be sure… there are no external converters you are trying to load/create? Also has the situation ever worked or is all this newly installed?

wilwizard1979, 2022-11-09:
> 

answering your questions:
The dongle is not discovered by ZHA even after re-start
I don't know what do you mean with external converters, so probably no. I just have ZHA running with a Sonoff ZB Bridge and 1 Tuya BLE + Zigbee coordinator.
The situation never worked. It is my first setup of Z2M after buying this dongle.

wilwizard1979, 2022-11-11:
Yes, the dongle is directly connected, without extensions.
I tried to troubleshoot without the other devices (like mouse and a second ZBBridge coordinator I have for ZHA).
No difference. But I found something that I will share next.

wilwizard1979, 2022-11-11:
I am about to give up on having Zigbee2MQTT in Home Assistant as I'm intending to run a separate Virtual Box with Ubuntu and MQTT runing separate.
In this process, I was following the Linux installation instruction of the Zigbee2MQTT website. When starting the Zigbee2MQTT Service I received a very similar message:

`.... Error while starting zigbee-herdsman error`

But! I received something more useful in addition:
`Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyUSB0' at SerialPort.`

That gave me the chance to investigate further on this and it seems to be a knownreported issue in Github: [https://github.com/Koenkk/zigbee2mqtt/issues/12218](url)

I followed the resolution for this case adding my $USER to the dialout group (in UBUNTU)
After this I restested in Ubuntu and it worked

### **I still would like to make it work in Home Assistant!**, 
but I don't know how to do the equivalent $USER addition to the Dialout group in Home Assistant (if this is even possible...).
Any experience or idea?

Rustymage, 2022-11-12:
> Yes, the dongle is directly connected, without extensions. I tried to troubleshoot without the other devices (like mouse and a second ZBBridge coordinator I have for ZHA). No difference. But I found something that I will share next.

I found a direct conncetion sometimes caused issues due to interference and I needed to use an extension cable and keep the adapter in the open and away from other wireless transmitting devices.

Rustymage, 2022-11-12:
> I am about to give up on having Zigbee2MQTT in Home Assistant as I'm intending to run it in a separate Virtual Box with Ubuntu, Mosquitto and Zigbee2MQTT. In this process, I was following the Linux installation instruction of the Zigbee2MQTT website [https://www.zigbee2mqtt.io/guide/installation/01_linux.html#configuring](url). When starting the Zigbee2MQTT Service I received a very similar message:
> 
> `.... Error while starting zigbee-herdsman error`
> 
> But! I received something more useful in addition: `Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyUSB0' at SerialPort.`
> 
> That gave me the chance to investigate further on this and it seems to be a known reported issue in Github: [https://github.com/Koenkk/zigbee2mqtt/issues/12218](url)
> 
> I followed the resolution for this case adding my $USER to the dialout group (in UBUNTU) After this I restested Ubuntu and it worked
> ### **I still would like to make it work in Home Assistant!**,
> 
> but I don't know how to do the equivalent $USER addition to the Dialout group in Home Assistant (if this is even possible...). Any experience or idea?

What's the output of this https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port

wilwizard1979, 2022-11-12:
> I found a direct conncetion sometimes caused issues due to interference and I needed to use an extension cable and keep the adapter in the open and away from other wireless transmitting devices.

I will search for one. Any recommendation of maximum length? I'm doing tests in the basement and will switch all devices around off

wilwizard1979, 2022-11-12:
> What's the output of this https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port

I tested the this command in my HA Terminal:
`test -w /dev/ttyUSB0 && echo success || echo failure`
and received
`success`

To be sure I disconnected the Dongle from the USB and tested the command again. I received then `failure`. So it means it is detecting the Dongle.

I could not test the other Linux commands of this documentation in Home Assistant (eg.: give permission, etc)

Anything else?

Rustymage, 2022-11-12:
> I will search for one. Any recommendation of maximum length? I'm doing tests in the basement and will switch all devices around off

I use this 

[image]

Just ensure it's of reasonable quality.

It may not fix your issue but it's recommended not to have your adapter plugged directly into the mobo, at least that's what I've learnt and read about.

wilwizard1979, 2022-11-12:
> It may not fix your issue but it's recommended not to have your adapter plugged directly into the mobo, at least that's what I've learnt and read about.

Thanks a lot. I could not find exactly the same brand in Germany, but I bought one with a good reputation.

wilwizard1979, 2022-11-12:
Just to complete the investigation, I tested without any device around, except the notebook which the USB is plugged in. The result is the same. Just need to test with the cable extender now, arriving on Monday.

Rustymage, 2022-11-12:
> Just to complete the investigation, I tested without any device around, except the notebook which the USB is plugged in. The result is the same. Just need to test with the cable extender now, arriving on Monday.

I found a conbee I had would not work plugged directly into a RPi. With an extension, it was fine.

I don't believe you answered this q - https://github.com/Koenkk/zigbee2mqtt/issues/14890#issuecomment-1312416340

wilwizard1979, 2022-11-12:
I did it up there: [https://github.com/Koenkk/zigbee2mqtt/issues/14890#issuecomment-1312433856](url)

wilwizard1979, 2022-11-12:
I did answer up there:

> I tested the this command in my HA Terminal: `test -w /dev/ttyUSB0 && echo success || echo failure` and received `success`
>
> To be sure I disconnected the Dongle from the USB and tested the command [...]

Rustymage, 2022-11-12:
> I did answer up there:

Thanks, I missed that.

So it's seen by the HA instance, but won't start Z2M. It's seen and operates using a VM & Z2M. So the adapter works.

Have you shared your configuration.yaml?

wilwizard1979, 2022-11-12:
> Have you shared your configuration.yaml?

Not yet. Here we go:

**Configuration.yaml inside Zigbee2MQTT:**
`homeassistant: true`

**Configuration.yaml from HA:**
Nothing in relation to Zigbee2MQTT

**Configuration in the Addon:**
```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core.mosquitto
serial:
  port: /dev/ttyusb0
zigbee_herdsman_debug: true
```

Rustymage, 2022-11-13:
Maybe silly, but have you tried `/dev/ttyUSB0`

wilwizard1979, 2022-11-13:
> Maybe silly, but have you tried `/dev/ttyUSB0`

Yes, just tried. Not working...

wilwizard1979, 2022-11-13:
@ciotlosm, could you please take a look on this case?

wilwizard1979, 2022-11-13:
Just to complement the history, I want to add that I tried now several configurations from the documentation linked below: [https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html](url)
Unfortunately the result was the same for the several combinations I've tried.

Starting to believe that there is something wrong with the Dongle hardware version. With the many component shortage issues, it might be that this hardware version would have something special.

Will wait for some more few days to see if someone could help, otherwise I will return it.

wilwizard1979, 2022-11-14:
> I found a direct conncetion sometimes caused issues due to interference and I needed to use an extension cable and keep the adapter in the open and away from other wireless transmitting devices.

Hi @Rustymage , I did the same test with the cable extension but the result is unfortunately the same.

wilwizard1979, 2022-11-24:
Hi All, just to leave my last update on this thread.
In the end I decided to give up on the Sonoff CC2652 Dongle with MQTT and decided to go to the new Sonoff EFR32MG21 Dongle using ZHA.
Pretty happy with that. Thanks all for trying to support.

closed 2022-11-25 by wilwizard1979