Koenkk/zigbee2mqtt #14894: Important latency with Legrand wireless dimmer

jiquem, 2022-11-09:
### What happened?

Facing important latency (2-3 seconds) with Legrand 067774 & 067773, I implemented binding between wireless dimmer and hue lights. However I am still experiencing latency.
From time to time, I have more than 10 seconds delay on Legrand switches who are far from the coordinator. (I have 50 routers)

### What did you expect to happen?

I expect to have no latency when using binding.
In this discussion I’ve opened on Jeedom forum it’s mentionned that it's working with deconz/conbee II : https://community.jeedom.com/t/latence-avec-les-interrupteurs-legrand/94427

### How to reproduce it (minimal and precise)

Simply use a Legrand 067774 & 067773 in direct binding

### Zigbee2MQTT version

1.28.2-dev commit: 5e2df18

### Adapter firmware version

20220219

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator

### Debug log

Binding implemented I have these in the log when I push a button

Info 2022-11-09 13:49:28MQTT publish: topic 'zigbee2mqtt/legrand_switch_double_beajeanmi_dressing_1', payload '{"action":null,"battery":60,"last_seen":1667998168821,"linkquality":65,"voltage":2800}'
Info 2022-11-09 13:49:28MQTT publish: topic 'zigbee2mqtt/legrand_switch_double_beajeanmi_dressing_1', payload '{"action":null,"battery":60,"last_seen":1667998168903,"linkquality":61,"voltage":2800}'
Info 2022-11-09 13:49:29MQTT publish: topic 'zigbee2mqtt/legrand_switch_double_beajeanmi_dressing_1', payload '{"action":null,"battery":60,"last_seen":1667998169553,"linkquality":61,"voltage":2800}'
Info 2022-11-09 13:49:29MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"legrand_switch_double_beajeanmi_dressing_1","ieee_address":"0x0004740000b927d2"},"type":"device_announce"}'
Info 2022-11-09 13:49:30MQTT publish: topic 'zigbee2mqtt/legrand_switch_double_beajeanmi_dressing_1', payload '{"action":null,"battery":60,"last_seen":1667998170732,"linkquality":51,"voltage":2800}'
Info 2022-11-09 13:49:30MQTT publish: topic 'zigbee2mqtt/legrand_switch_double_beajeanmi_dressing_1', payload '{"action":null,"battery":60,"last_seen":1667998170738,"linkquality":51,"voltage":2800}'
Info 2022-11-09 13:49:30MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"legrand_switch_double_beajeanmi_dressing_1","ieee_address":"0x0004740000b927d2"},"type":"device_announce"}'
Info 2022-11-09 13:49:31MQTT publish: topic 'zigbee2mqtt/legrand_switch_double_beajeanmi_dressing_1', payload '{"action":null,"battery":60,"last_seen":1667998172021,"linkquality":51,"voltage":2800}'
Info 2022-11-09 13:49:32MQTT publish: topic 'zigbee2mqtt/legrand_switch_double_beajeanmi_dressing_1', payload '{"action":null,"battery":60,"last_seen":1667998172250,"linkquality":47,"voltage":2800}'

jiquem, 2022-11-12:
I noticed that direct binding is working fine when Z2M is switched off : the latency is quite good. And when I switch Z2M on again, the latency is bad. @Celestianx would it be possible to test if you have some time please ?

celestianx, 2022-11-12:
I am away from home but I will.
I did not know that leaving z2m on could induce delays but maybe it’s juste because you are shutting down any other zigbee traffic?
I have recently purchased a zigbee sniffer and have setup a wireless switch with z2m and with legrand to see if there is any difference but did not have the time yet to analyze

One thing that I saw is the frame about Ubisys which never appeared to me as normal and I think this is a z2m bug of wrongly interpreter cluster

jiquem, 2022-11-12:
Thanks @celestianx 
I spend the whole day moving my 100 devices to deconz. There is still a few latency but much less. This is quite strange.
However deconz is far less userfriendly to configure especially for binding and clusters

celestianx, 2022-11-13:
> Thanks @celestianx I spend the whole day moving my 100 devices to deconz. There is still a few latency but much less. This is quite strange. However deconz is far less userfriendly to configure especially for binding and clusters

You switched back to deconz in the end? Quite personally, if the Legrand app was not that bad to handle lights compared to other options, I would have never dwelled into the open world of smart homes :) It's powerful but really time consuming

jiquem, 2022-11-13:
Yes, I will see if it’s better during one or two weeks and after that, I will decide.

bsheepdiy, 2022-11-16:
I confirm this latency. Switches that are far from the coordinator have a delay (3sec at least, 10 sec when they are not used for 4 days). This delay was not here using deconz/conbee2. I don't know if its a coordinator firmware bug or z2m...

My Environment: 
SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator / firmware 20220928
z2m 1.28.1

jiquem, 2022-11-16:
I am now running with deconz/conbee2 since since this weekend and the latency is around 0.5s even for switches far from the coordinator.
However, z2m support all my devices, deconz/conbee2 not ;( so I may switch back again. (which is quite painful with 100 devices)

celestianx, 2022-11-17:
> I am now running with deconz/conbee2 since since this weekend and the latency is around 0.5s even for switches far from the coordinator. However, z2m support all my devices, deconz/conbee2 not ;( so I may switch back again. (which is quite painful with 100 devices)

Which device does deconz/conbee2 not support? Legrand or others? Is it really that hard to add device support? I have been doing it for 2 Legrand devices on z2m and sure, it takes some time to understand but should be doable.

celestianx, 2022-11-17:
While all of these is early testing. What I can report:
- Latency occurs only when device has been "sleeping" for a while. Is that so for everyone?
- Direct device binding is removing the latency but z2m has to be shut down or you have to remove the coordinator binding in device settings in z2m (it seems that even removing coordinator binding and add it back so it appears after direct device binding works - have to test that again)

In my initial traffic sniffing observations having one switch configured with a Legrand Gateway and another with z2m, the switch behaves differently
- Legrand Gateway seems to setup a group to bind switch and controlled devices and every single time the device is pressed, it seems to successively sent several "toggle on/off" and "device announce" messages on the network (~5)
=> That makes me wonder whether the code we identified as causing the #14256 issue and removed was there to address this when device is configured within a group.
- With z2m, default binding is direct binding with the coordinator. I see 1-2 `manuSpecificUbisysDimmerSetup` cluster message which makes me thing something is wrong, and somehow z2m cannot use the manufacturerCode as Ubisys and Legrand share  the same cluster id.
- All of that makes me think that maybe the switch firmware expects to work with a "group" vs. "direct binding". One additional cue to that is some people had pointed out (#10862) that creating a virtual group was fixing the referenced #14256 issue

bsheepdiy, 2022-11-17:
I think you get it, we will need to use groups to bind.
When I used conbee/deconz, by default, it creates a group for each lights.

If we have no other solution, it will be good to complete the zigbee documentation as a suggestion to avoid this latency.

celestianx, 2022-11-17:
Sure let's see if that proves to be the root cause. Whoever can test first (won't be able to test before this week-end most likely)
I do hope such a simple solution will work: having a virtual groups with the coordinator in it as I tend to prefer to let my Hue lights be managed by the Hue bridge (brings more features, updates, etc.) and connect both with HASS

jiquem, 2022-11-17:
Hello,
I will try this tomorrow, the time to move some dimmers and lights to z2m. (I can't tonight, otherwise my wife may probably kill me ;))
BTW, my Aqara TRV isn't supported  yet (but a patch is coming) in deconz

celestianx, 2022-11-17:
I tested it as I got back with 2 setups:
- Switch binded to an empty group in z2m, leaving also the default binding with the coordinator
- Switch binded to a group in z2m with two devices, leaving also the default binding with the coordinator
Will keep testing but it seems to have addressed the latency issue. Please test on your end and let me know

celestianx, 2022-11-18:
Also, if you are interested, I have developed a Blueprint for HASS for both switches so that, once added in z2m, you just use the UI to select the switch and lights and you can even finetune brightness steps and transitions:
[image]

For 067773:
[image]

For 067774:
[image]

I intend to add support for outlet switch as well

jiquem, 2022-11-18:
Hello,
In confirm that it's working fine : I have no more latency on the 5 dimmer I used to test.
I just face another issue that I will try to reproduce this weekend. I bind a switch with 3 differents group. It worked find at 7am... and then stopped working at 10am. The binding didn't disappear in z2m, so I reperformed it... and will see if it fails again.

celestianx, 2022-11-18:
Cool! Just out of curiosity, why would you need to bind a switch to 3 different groups?

jiquem, 2022-11-18:
I created three groups to manage 3 areas within my living room :
- kitchen : 6 lights
- dinning : 2 lights
- sofa : 7 lights

Each group is linked to a specific dimmer
On top of that, I have a dimmer to switch on / switch off everything

celestianx, 2022-11-18:
> I created three groups to manage 3 areas within my living room :
> 
> * kitchen : 6 lights
> * dinning : 2 lights
> * sofa : 7 lights
> 
> Each group is linked to a specific dimmer On top of that, I have a dimmer to switch on / switch off everything

Ok so the groups are not related. I thought this was the source of your problem.

jiquem, 2022-11-18:
Yes, you understood well

jiquem, 2022-11-18:
Thanks for the blueprint @celestianx.
However, while importing them, clicking on your links, I've got an error 
[image]

celestianx, 2022-11-19:
> Thanks for the blueprint @celestianx. However, while importing them, clicking on your links, I've got an error [image]

My repository was private :) Should work now!

celestianx, 2022-11-19:
@bsheepdiy : did you have a chance to test on your side and confirm this solves it?

bsheepdiy, 2022-11-21:
Everything is working flawlessy for me using group and empty group.

celestianx, 2022-11-21:
@Koenkk : 
Is there any way for us to submit changes to these URLs to provide additional configuration steps to avoid any latency?:
- [067773](https://www.zigbee2mqtt.io/devices/067773.html)
- [067774](https://www.zigbee2mqtt.io/devices/067774.html)

celestianx, 2022-11-25:
Hi @jiquem,
I have updated the documentation for 067774:
```
As part of its configuration, the remote expects to have a "group" binding configured for each of its button. Otherwise, it will consider itself not fully configured and send unnecessary commands on the zigbee network resulting in noticeable delay when controlling your device(s).
In a nutshell:
- When using devices bindings directly in zigbee2mqtt, create groups with your target devices for each button and bind related groups accordingly.
- When using automations to control your devices (e.g. homeassistant), create a virtual group (empty) and bind it to the buttons. Beware that having both the coordinator and group bound to a remote's button will publish 2 MQTT messages
```

For 067773, I haven't noticed any latencies and thus did not update it yet.
I think you can close the issue!

jiquem, 2022-11-25:
Thanks @celestianx !
Issue is closed
I may open another one since binding suddenly disappeared on 3 of my dimmer this week (not a the same time even not the same day)
I had to reintegrate them in Z2M and reperform the binding

closed 2022-11-25 by jiquem

celestianx, 2022-11-26:
Sure. As for me, I don't have any issue just for your to know. Everything has been stable.
My only issue is with my blueprint, I think it captures 2 MQTT messages, one from the coordinator, one from the group and sometimes it turns on and off the light instantaneously
Have you had the time to play around with the blueprint?

jiquem, 2022-11-26:
Hello @celestianx, no I didn't test them yet. Will do next weekend

jiquem, 2022-11-28:
@celestianx : it doesn't list my devices

[image]

celestianx, 2022-11-28:
When you take a look at your device manufacture/model in HASS, what does it say?
``` 
 switch:
      name: Switch
      selector:
        device:
          manufacturer: "Legrand"
          model: "Wireless remote switch (067773)"
    target:
```

jiquem, 2022-12-02:
Hi @celestianx , here is what I have
[image]

celestianx, 2022-12-03:
I am also about to publish new versions of the blueprints I have been working with multiple targets support, including both lights and switches support

jiquem, 2022-12-03:
Thanks @celestianx I confirm blueprints are working now !

celestianx, 2022-12-04:
> Thanks @celestianx I confirm blueprints are working now !

I have published updated blueprints. It now supports both lights and switches as well as selecting multiple targets

Loic691, 2022-12-07:
Hi
I have noticed the same issue but with latency variable between 0 to 60 or 120 seconds..
I'm not sure it is the same issue...

giannoug, 2023-03-14:
Same delay here, but I can't seem to figure out how empty group binding works. Can anyone share a screenshot of how a working binding should look like?

[image]

celestianx, 2023-03-16:
You just have to create a group in the "Groups" tab in z2m top menu. I think you did it right with your "empty" group.

[image]

And what you did seems about right. Maybe, unbind coordinator with LevelCtrl and OnOff. I have noticed in some use-case it would also add delays if it was both bound to a Coordinator and a Group.

As for forcing the updates, I have asked to one of my friend who works at Legand Netatmo:
1. You have to wake up the device for about 10 times for it to request for an update, meaning you have to push the button, wait 1 min, push the button etc. I did it for a set of devices this week-end, it works even though it's really bothersome.
2. If you manage to update to "20220722", my friend told me now you can force the update by pushing both up and down button for about 20s until a magenta led blinks

giannoug, 2023-03-16:
Thanks for the confirmation, at least I'm sure I got the groups part right!

I managed to update the two switches using the press every 10s method (you said 1m but it's actually 10s between presses). That worked until they didn't update more, so I removed and added them back to z2m. I have confirmed they are updated in the Legrand app (can't remember what version though), but they still report 20211001 and version 6 on z2m. They now expose toggle and a lot more options so something is changed.

It seems that the lag is gone by binding to an empty group, but I still see manuSpecificUbisysDimmerSetup messages in logs when the switch wakes up.

celestianx, 2023-03-17:
> Thanks for the confirmation, at least I'm sure I got the groups part right!
>
> I managed to update the two switches using the press every 10s method (you said 1m but it's actually 10s between [...]

Good! I know it was around a few seconds from the z2m logs but to make sure and give simple directions, I rounded up to a min ;o) For me, the latest version for double gangs is 36 (Legrand app) and for simple gang  is 63 (Legrand app) 
As for the manuSpecificUbisysDimmerSetup, it seems to have disappeared for me

giannoug, 2023-03-17:
> For me, the latest version for double gangs is 36 (Legrand app) and for simple gang is 63 (Legrand app)
Thanks for that! I believe I saw these versions too. I had to remove everything because the scene switch seems to break the Netatmo integration in HA for some reason.

jiquem, 2023-03-17:
You can find a followup of FW versions on https://community.jeedom.com/t/legrand-with-netatmo-firmware-des-equipements/75218