Koenkk/zigbee2mqtt #14895: IKEA Tradfri shortcut button battery drain

maframan, 2022-11-09:
### What happened?

Battery drain with the IKEA Tradfri shortcut button

### What did you expect to happen?

Battery normal usage

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-E (EFR32MG21)

### Debug log

N/A

maframan, 2022-11-09:
For your information, for test only, i've upgraded the coordinator Sonoff ZBDongle-E at this firmware version: '{"meta":{"maintrel":"1 ","majorrel":"7","minorrel":"1","product":9,"revision":"7.1.1.0 build 273"},"type":"EZSP v9"}'.
Now i'm monitoring the behavior

Koenkk, 2022-11-09:
Same as https://github.com/Koenkk/zigbee2mqtt/issues/13681

closed 2022-11-09 by Koenkk