Koenkk/zigbee2mqtt #14909: MOES BRT-100-TRV doesn't update vlve position

Ivano62, 2022-11-09:
### What happened?

When the room temperature drops below the set point the valve opens, the "valve_state" attribute changes value from CLOSED to OPEN, but the opening percentage, "position" attribute, remains fixed at **0%**.
Before version 1.28.2 everything worked fine
[image]

### What did you expect to happen?

The "position" attribute indicates the valve opening percentage

### Zigbee2MQTT version

1.28.2-dev commit: 672da64

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

catalinescu-sudo, 2022-11-12:
Hi, please take a look at my issue https://github.com/Koenkk/zigbee2mqtt/issues/14940 do you have the same problem? Thank you!

Regarding your issue I can tell that my valve position is getting updated
[image]

Ivano62, 2022-11-13:
After latest update position seems working correctly.

harristom, 2022-12-18:
I see a similar issue - valve is reported as closed but with position 25%

[image]

closed 2023-01-26 by github-actions[bot]

almirus, 2023-12-13:
Is it normal behavior to frequently turn on and off?

[image]