Koenkk/zigbee2mqtt #14914: Xiaomi WXKG02LM_rev2 not reacting on the first click

jakob42, 2022-11-10:
### What happened?

Hi,

I switched back to z2m after a stint with a conbee II and their deconz software. Replaced the conbee with a ZBDongle-P, everything works fine and all devices are supported (they werent with deconz). But I ran into a problem that my battery powered wall switches often don't react on the first click. (I noticed it at least with the following switches: [Xiaomi WXKG02LM_rev2](https://www.zigbee2mqtt.io/devices/WXKG02LM_rev2.html#xiaomi-wxkg02lm_rev2), [TuYa TS0044](https://www.zigbee2mqtt.io/devices/TS0044.html#tuya-ts0044), [IKEA E1743](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743) and [Philips 324131092621](https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621))

```
zigbee2mqtt      | Zigbee2MQTT:debug 2022-10-28 07:58:44: Received Zigbee message from 'Schalter Wohnzimmer', type 'raw', cluster 'genOnOff', data '{"data":[1,5,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:info  2022-10-28 07:58:44: MQTT publish: topic 'zigbee2mqtt/Schalter Wohnzimmer', payload '{"action":null,"battery":100,"last_seen":1666936724678,"linkquality":32,"voltage":3000}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-10-28 07:58:45: Received Zigbee message from 'Schalter Wohnzimmer', type 'raw', cluster 'genOnOff', data '{"data":[1,5,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:info  2022-10-28 07:58:45: MQTT publish: topic 'zigbee2mqtt/Schalter Wohnzimmer', payload '{"action":null,"battery":100,"last_seen":1666936725681,"linkquality":32,"voltage":3000}'
```

`action: null` obviosly doesn't trigger anything in my HA or nodered.

Sorry, I didn't try this on dev, I don't have a test environment and am afraid of breaking something. If this is really necessary for that kind of problem, please let me know.

### What did you expect to happen?

 Half a minute later, same switch, same button:

```
zigbee2mqtt      | Zigbee2MQTT:debug 2022-10-28 07:59:25: Received Zigbee message from 'Schalter Wohnzimmer', type 'raw', cluster 'genOnOff', data '{"data":[1,6,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:info  2022-10-28 07:59:25: MQTT publish: topic 'zigbee2mqtt/Schalter Wohnzimmer', payload '{"action":"4_single","battery":100,"last_seen":1666936765342,"linkquality":32,"voltage":3000}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-10-28 07:59:25: MQTT publish: topic 'zigbee2mqtt/Schalter Wohnzimmer', payload '{"action":"","battery":100,"last_seen":1666936765342,"linkquality":32,"voltage":3000}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-10-28 07:59:25: MQTT publish: topic 'zigbee2mqtt/Schalter Wohnzimmer/action', payload '4_single'
```

The switches worked on the first click with deconz.

### How to reproduce it (minimal and precise)

Wait for some time (about 1-2 hours I guess) and then try to use the switch. The first click, sometimes even the second doesn't work, couple seconds later everything is fine.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20220219

### Adapter

ZBDongle-P

jakob42, 2022-11-10:
Feels the same as in this issue, first tests suggest that it solved my issue with the [TS0044](https://www.zigbee2mqtt.io/devices/TS0044.html#tuya-ts0044).
https://github.com/Koenkk/zigbee2mqtt/issues/8072

jakob42, 2022-11-10:
Doesn't seem to be solved for my TS0044 after all 😞

guille79es, 2022-11-10:
Weird, I just updated my setup to 1.28.2 and this same model of switch now shows 
This plugin does not support zigbee device with model "WXKG02LM_rev2" yet
Frontend looks ok and says supported?

lux73, 2022-11-12:
can't confirm this Problem - i had two of them in my Setup. They're running smooth & reliable.

running native z2m v1.28.2 on Debian 11.5 @ slaesh CC2652RB with 40+ Devices

it could be a Problem with your Environment - do yoiu running ZHA or native z2m?

jakob42, 2022-11-12:
I'm running z2m dockerized, not zha. The switch is a couple of years old, I'm wondering if it might be a different version?

I've replaced the battery for the first time a couple of month ago, but it was running fine afterwards. The change came directly when switching from conbee+deconz to ZBDongle-P+z2m...

Zigbee2MQTT version
[1.28.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.2) commit: [360a777](https://github.com/Koenkk/zigbee2mqtt/commit/360a777)
Coordinator type
zStack3x0
Coordinator revision
20220219
Coordinator IEEE Address
0x00124b0024c9f240
Frontend version
0.6.114
Stats
Total 34
By device type
End devices: 20
Router: 14
By power source
Battery: 20
Mains (single phase): 14
By vendor
LUMI: 9
IKEA of Sweden: 8
MLI: 5
OSRAM: 2
Philips: 2
_TZ3000_pnzfdr9y: 2
_TZ2000_a476raq2: 1
eWeLink: 1
_TZ3000_odygigth: 1
_TZ3000_wkai4ga5: 1
_TZ3000_vp6clf9d: 1
_TZ3000_8ybe88nf: 1
By model
lumi.weather: 3
TS0201: 2
lumi.remote.b286acn01: 2
RWL021: 2
ZBT-ExtendedColor: 2
TS0044: 2
TRADFRI on/off switch: 2
TRADFRI signal repeater: 2
lumi.sensor_motion.aq2: 2
lumi.sensor_ht: 2
TS0101: 2
TRADFRI bulb E27 WS opal 980lm: 1
TH01: 1
Plug 01: 1
TRADFRI remote control: 1
TS0505A: 1
CCT Lighting: 1
RGB-CCT: 1
TRADFRI bulb E14 WS opal 600lm: 1
ZBT-Remote-ALL-RGBW: 1
TRADFRI bulb E27 WW 806lm: 1
LIGHTIFY Indoor Flex RGBW: 1

lux73, 2022-11-12:
```
Zigbee2MQTT version
    [1.28.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.2) commit: [e66fe853](https://github.com/Koenkk/zigbee2mqtt/commit/e66fe853)

Coordinator type
    zStack3x0

Coordinator revision
    20220219

Coordinator IEEE Address
    0x00124b002390da8c

Frontend version
    0.6.114

Stats

1. Total 42
2. By device type
   End devices: 32
   Router: 10
3. By power source
   Battery: 31
   Mains (single phase): 10
   DC Source: 1
4. By vendor
   LUMI: 25
   Philips: 6
   dresden elektronik: 3
   Eurotronic: 2
   Danfoss: 2
   XIAOMI: 2
   IKEA of Sweden: 1
   aqara: 1
5. By model
   lumi.weather: 6
   lumi.sensor_magnet.aq2: 6
   lumi.sen_ill.mgl01: 5
   LTW010: 4
   lumi.sensor_motion.aq2: 4
   FLS-PP3: 2
   SPZB0001: 2
   lumi.remote.b286acn01: 2
   eTRV0100: 2
   lumi.motion.agl04: 2
   LTW012: 1
   FLS-CT: 1
   LTW013: 1
   lumi.vibration.aq1: 1
   lumi.router: 1
   TRADFRI on/off switch: 1
   lumi.motion.ac01: 1
```

lux73, 2022-11-12:
Thats my Switch:

```
Friendly name
    doppelwippe-kueche

Description

Last seen
    2 minutes ago
Availability
    Disabled
Device type
    EndDevice
Zigbee Model
    lumi.remote.b286acn01
Zigbee Manufacturer
    LUMI
Description
    Aqara double key wireless wall switch (2018 model)
Support status

    Supported
IEEE Address
    0x00158d00027c3445
Network address
    0x3D41
Firmware build date
    20180809
Firmware version
    3000-0001
Manufacturer
    Xiaomi
Model
    WXKG02LM_rev2
Power
    Battery 100%
Interview completed
    True
```

lux73, 2022-11-12:
in addition i have:

`legacy: false`

in my device.yaml for my two Double Rocker - don't know if that helps

closed 2022-12-20 by github-actions[bot]