Koenkk/zigbee2mqtt #14917: Cannot set properties of undefined (setting 'installed_version')

vladyspavlov, 2022-11-10:
### What happened?

When attempting to launch the latest zigbee2mqtt edge, an error occurred. It seems like an error appeared after this PR https://github.com/Koenkk/zigbee2mqtt/pull/14905

### How to reproduce it (minimal and precise)

Start zigbee2mqtt edge

### Zigbee2MQTT version

1.28.2-dev (commit #98a6a2d)

### Adapter firmware version

20220219

### Adapter

Sonoff CC2652P

### Debug log

```
TypeError: Cannot set properties of undefined (setting 'installed_version')
    at HomeAssistant.adjustMessageBeforePublish (/app/lib/extension/homeassistant.ts:1465:51)
    at Controller.publishEntityState (/app/lib/controller.ts:266:49)
    at Object.publishLastSeen (/app/lib/util/utils.ts:348:9)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:157:19)
```

Koenkk, 2022-11-11:
Can you provide your z2m `configuration.yaml`? The update property should always be mocked.

vladyspavlov, 2022-11-11:
<details>
<summary>configuration.yaml</summary>

```
external_converters: []
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  user: addons
  password: mqttpass
  server: mqtt://core-mosquitto:1883
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_eae9d457011aec11a0bf77e5f01c6278-if00-port0
advanced:
  log_level: debug
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
  last_seen: ISO_8601
device_options:
  retain: true
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  transmit_power: 15
availability: true
```
</details>

Koenkk, 2022-11-12:
An that of `devices.yml`?

vladyspavlov, 2022-11-12:
<details>
<summary>devices.yaml</summary>

```
'0xb4e3f9fffe393f5f':
  friendly_name: '0xb4e3f9fffe393f5f'
  homeassistant: {}
  optimistic: true
  retain: true
'0x00124b0023ae3549':
  friendly_name: '0x00124b0023ae3549'
'0xa4c13809448f82aa':
  friendly_name: '0xa4c13809448f82aa'
  homeassistant: {}
  optimistic: true
  retain: true
'0xa4c13850a037aabf':
  friendly_name: '0xa4c13850a037aabf'
  homeassistant: {}
  optimistic: true
  retain: true
'0xa4c13892fdbfa1b0':
  friendly_name: '0xa4c13892fdbfa1b0'
  optimistic: true
  retain: true
'0xa4c138f3ff9bf15a':
  friendly_name: '0xa4c138f3ff9bf15a'
  homeassistant: {}
  optimistic: true
  retain: true
'0xa4c138cf87a03e5f':
  friendly_name: '0xa4c138cf87a03e5f'
  homeassistant: {}
  optimistic: true
  retain: true
'0xa4c138e830c29bfe':
  friendly_name: '0xa4c138e830c29bfe'
  optimistic: true
  retain: true
'0xa4c138ba88cc1fd5':
  friendly_name: '0xa4c138ba88cc1fd5'
  optimistic: true
  retain: true
  humidity_calibration: 8
  temperature_calibration: 1
'0xa4c138598140ce79':
  friendly_name: '0xa4c138598140ce79'
  description: ''
'0xa4c138c66ae7a7b9':
  friendly_name: '0xa4c138c66ae7a7b9'
'0xa4c138f6315abdd1':
  friendly_name: '0xa4c138f6315abdd1'
'0xa4c138fc3a0523b6':
  friendly_name: '0xa4c138fc3a0523b6'
'0xa4c1389103841357':
  friendly_name: '0xa4c1389103841357'
  optimistic: true
  retain: true
'0xa4c138198a25e0be':
  friendly_name: '0xa4c138198a25e0be'
  homeassistant: {}
  optimistic: true
  retain: true
'0xa4c138b81740ab3f':
  friendly_name: '0xa4c138b81740ab3f'
  optimistic: true
  retain: true
'0xa4c1381786aa311c':
  friendly_name: '0xa4c1381786aa311c'
  humidity_calibration: 3
  temperature_calibration: 0.5
'0xa4c1386aae3b7e96':
  friendly_name: '0xa4c1386aae3b7e96'
  humidity_calibration: -6
  temperature_calibration: 0.5
  homeassistant: {}
  optimistic: true
  retain: true
'0xa4c1383a484a46d9':
  friendly_name: '0xa4c1383a484a46d9'
  humidity_calibration: 2
  temperature_calibration: 0.5
'0xa4c1382e6b5da7ce':
  friendly_name: '0xa4c1382e6b5da7ce'
  humidity_calibration: -4
  temperature_calibration: 0.75
'0xa4c1389295a0d426':
  friendly_name: '0xa4c1389295a0d426'
'0xa4c1381503399c24':
  friendly_name: '0xa4c1381503399c24'
'0xa4c1382db2b9e0c9':
  friendly_name: '0xa4c1382db2b9e0c9'
  humidity_calibration: -7
'0xa4c138c58b6e2bcc':
  friendly_name: '0xa4c138c58b6e2bcc'
'0xa4c138c5e92250da':
  friendly_name: '0xa4c138c5e92250da'
  humidity_calibration: -7
  temperature_calibration: 0.5
'0xa4c1382cfb807ca3':
  friendly_name: '0xa4c1382cfb807ca3'
  humidity_calibration: -6
  temperature_calibration: 0.5
'0x00124b0024cc2fdc':
  friendly_name: '0x00124b0024cc2fdc'
'0x00124b0024fc2c72':
  friendly_name: '0x00124b0024fc2c72'
'0x00124b002512032f':
  friendly_name: '0x00124b002512032f'
'0x00124b002512c10f':
  friendly_name: '0x00124b002512c10f'
  homeassistant: {}
  optimistic: true
  retain: true
'0x00124b0025113dae':
  friendly_name: '0x00124b0025113dae'
  homeassistant: {}
  optimistic: true
  retain: true
'0x00124b0025001af3':
  friendly_name: '0x00124b0025001af3'
  homeassistant: {}
  optimistic: true
  retain: true
'0xa4c13868925b0057':
  friendly_name: '0xa4c13868925b0057'
  homeassistant: {}
  optimistic: true
  retain: true
'0x00124b002512a5e6':
  friendly_name: '0x00124b002512a5e6'
  homeassistant: {}
  optimistic: true
  retain: true
'0x60a423fffe7ff4b8':
  friendly_name: '0x60a423fffe7ff4b8'
  homeassistant: {}
  optimistic: true
  retain: true
'0xcc86ecfffe9335fa':
  friendly_name: '0xcc86ecfffe9335fa'
'0x540f57fffede0446':
  friendly_name: '0x540f57fffede0446'
'0x84ba20fffe2abfca':
  friendly_name: '0x84ba20fffe2abfca'
  homeassistant: {}
  optimistic: true
  retain: true
'0x00158d0002aef2f8':
  friendly_name: '0x00158d0002aef2f8'
'0x00158d0007003f67':
  friendly_name: '0x00158d0007003f67'
'0x00158d00075f852b':
  friendly_name: '0x00158d00075f852b'
  occupancy_timeout: 60
  homeassistant: {}
  optimistic: true
  retain: true
'0x00158d0007755727':
  friendly_name: '0x00158d0007755727'
  occupancy_timeout: 60
  homeassistant: {}
  optimistic: true
  retain: true
'0xa4c138f1713bd8b8':
  friendly_name: '0xa4c138f1713bd8b8'
'0xa4c13876ea7d2e46':
  friendly_name: '0xa4c13876ea7d2e46'
'0xa4c138515db00afe':
  friendly_name: '0xa4c138515db00afe'
'0xa4c138d31ec97a48':
  friendly_name: '0xa4c138d31ec97a48'
'0xa4c138f7bebe9e61':
  friendly_name: '0xa4c138f7bebe9e61'
'0xa4c1381fae83ee6b':
  friendly_name: '0xa4c1381fae83ee6b'
'0xa4c1381c61f002c6':
  friendly_name: '0xa4c1381c61f002c6'
'0xa4c138f006efcfb7':
  friendly_name: '0xa4c138f006efcfb7'
'0xa4c1387d03894fed':
  friendly_name: '0xa4c1387d03894fed'
'0xa4c138b02df7b005':
  friendly_name: '0xa4c138b02df7b005'
'0xa4c1380edab0e27e':
  friendly_name: '0xa4c1380edab0e27e'
'0xa4c13833cfce9a3b':
  friendly_name: '0xa4c13833cfce9a3b'
'0xb4e3f9fffe11af53':
  friendly_name: '0xb4e3f9fffe11af53'
'0xa4c138a10e59867f':
  friendly_name: '0xa4c138a10e59867f'
'0xcc86ecfffec1f69f':
  friendly_name: '0xcc86ecfffec1f69f'
'0xa4c138f4013d28e0':
  friendly_name: '0xa4c138f4013d28e0'
'0xa4c138e10e92deea':
  friendly_name: '0xa4c138e10e92deea'
'0xa4c1383a1dc32bb8':
  friendly_name: '0xa4c1383a1dc32bb8'
'0xa4c1387f07c7cccd':
  friendly_name: '0xa4c1387f07c7cccd'
'0xa4c138594675c9f9':
  friendly_name: '0xa4c138594675c9f9'
```
</details>

Koenkk, 2022-11-13:
Fixed the issue!

closed 2022-12-21 by github-actions[bot]