Koenkk/zigbee2mqtt #14920: Bouffalolab RMC002 "Power" Listed as "Unknown in Devices List

coolspot18, 2022-11-10:
### What happened?

This unbranded Zigbee Smart Outlet is listed as ? for Power when it is a plug-in device:

[image]

Is this correct or just a minor display issue?

### What did you expect to happen?

It should display as a plug-in device.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee USB 3.0 Plus w/ zStack 3x0

coolspot18, 2022-11-11:
Here it is:

`{
	"id": 19,
	"type": "Router",
	"ieeeAddr": "0x7cb94c618a4c0000",
	"nwkAddr": 60727,
	"manufId": 65535,
	"manufName": "Bouffalolab",
	"powerSource": "Unknown",
	"modelId": "Bouffalolab",
	"epList": [1],
	"endpoints": {
		"1": {
			"profId": 260,
			"epId": 1,
			"devId": 0,
			"inClusterList": [0, 3, 4, 5, 6],
			"outClusterList": [0, 3, 4, 5],
			"clusters": {
				"genBasic": {
					"attributes": {
						"modelId": "Bouffalolab",
						"manufacturerName": "Bouffalolab",
						"powerSource": 0,
						"zclVersion": 8,
						"appVersion": 1,
						"stackVersion": 0,
						"hwVersion": 0,
						"dateCode": "20210610-BL",
						"swBuildId": ""
					}
				},
				"genOnOff": {
					"attributes": {
						"onOff": 0
					}
				}
			},
			"binds": [{
				"cluster": 6,
				"type": "endpoint",
				"deviceIeeeAddress": "0x00124b0026b698e5",
				"endpointID": 1
			}],
			"configuredReportings": [{
				"cluster": 6,
				"attrId": 0,
				"minRepIntval": 0,
				"maxRepIntval": 3600,
				"repChange": 0
			}],
			"meta": {}
		}
	},
	"appVersion": 1,
	"stackVersion": 0,
	"hwVersion": 0,
	"dateCode": "20210610-BL",
	"swBuildId": "",
	"zclVersion": 8,
	"interviewCompleted": true,
	"meta": {
		"configured": 1466202208
	},
	"lastSeen": 1668150869742,
	"defaultSendRequestWhen": "immediate"
} {
	"id": 20,
	"type": "Router",
	"ieeeAddr": "0x7cb94c6198b40000",
	"nwkAddr": 19380,
	"manufId": 65535,
	"manufName": "Bouffalolab",
	"powerSource": "Unknown",
	"modelId": "Bouffalolab",
	"epList": [1],
	"endpoints": {
		"1": {
			"profId": 260,
			"epId": 1,
			"devId": 0,
			"inClusterList": [0, 3, 4, 5, 6],
			"outClusterList": [0, 3, 4, 5],
			"clusters": {
				"genBasic": {
					"attributes": {
						"modelId": "Bouffalolab",
						"manufacturerName": "Bouffalolab",
						"powerSource": 0,
						"zclVersion": 8,
						"appVersion": 1,
						"stackVersion": 0,
						"hwVersion": 0,
						"dateCode": "20210610-BL",
						"swBuildId": ""
					}
				},
				"genOnOff": {
					"attributes": {
						"onOff": 0
					}
				}
			},
			"binds": [{
				"cluster": 6,
				"type": "endpoint",
				"deviceIeeeAddress": "0x00124b0026b698e5",
				"endpointID": 1
			}],
			"configuredReportings": [{
				"cluster": 6,
				"attrId": 0,
				"minRepIntval": 0,
				"maxRepIntval": 3600,
				"repChange": 0
			}],
			"meta": {}
		}
	},
	"appVersion": 1,
	"stackVersion": 0,
	"hwVersion": 0,
	"dateCode": "20210610-BL",
	"swBuildId": "",
	"zclVersion": 8,
	"interviewCompleted": true,
	"meta": {
		"configured": 1466202208
	},
	"lastSeen": 1668148202963,
	"defaultSendRequestWhen": "immediate"
}`

danieledwardgeorgehitchcock, 2022-11-11:
Interesting.. I'll create an external converter for you to try in a bit

danieledwardgeorgehitchcock, 2022-11-11:
Try [this](https://gist.github.com/danieledwardgeorgehitchcock/52d36e5eac8e484f4fafb1b50d528ab8) (you may need to remove your device and restart Z2M first)

closed 2022-11-11 by Koenkk

coolspot18, 2022-11-11:
> Try [this](https://gist.github.com/danieledwardgeorgehitchcock/52d36e5eac8e484f4fafb1b50d528ab8) (you may need to remove your device and restart Z2M first)

Looks good - copied the .js file into my zigbee2mqtt folder, added the filename to the external converters page, restarted the add-on and the proper power source shows up.

Thanks!