Koenkk/zigbee2mqtt #14921: Z2M keeps using IEEE address of old stick so it cannot be used as router

lslomczynski, 2022-11-10:
### What happened?

Hello wonderful Z2M comunity :-)

I've been using a CC2531 USB stick as Coordinator and for a long run it was a pain as after some time (weks / months) it was becoming unresponsive and I needed to re-flash it over and over again.
After an upgrade to SONOFF CC2652P USB stick, despite deleting /data/coordinator_backup.json and /data/database.db, the new stick always came back to those files with IEEE of old CC2531. Same address was also visible in Z2M webpanel (Settings -> About).

It was fine up until yesterday when I decided to give the CC2531 a second chance as a router.
After re-flashing with router firmware and pairing to my network Z2M "went bannanas" as it suddenly had 2 devices with same IEEE.
The router was not visible ot the Z2M map, all items were shaking and flying around the screen like they were a drillhammer operators... Also the CC2531 router did not show up on the Devices list...

Getting to the point:
- I cant change the IEEE of my CC2531 as I do not have the CC Debugger BOX - I've been using the "alternartive flash method" all the time (Raspberry Pi here)
- I was able to read my SONOFF IEEE using TI Flash Programmer 2. Second IEEE was FFF....
- I have stopped the Z2M service (systemctl)
- In the /data/coordinator_backup.json I changed my old IEEE to new one under positions: "coordinator_ieee", "extended_pan_id"
- I did the same with /data/database.db
- After Z2M service start all changes taht I made were reverted to the old IEEE values...

I wonder where this old IEEE is being kept and why Z2M does not use the new one, even after deletion of abovementioned files... I even ran a grep through my entire 16GB SD card looking for that IEEE hidden somewhere and despite obvious locations like syslog or kernel log (plugging / unplugging) it didn't show up anywhere...

I am looking forward to your reply :-)

PS: I've found similar issues described here but, forgive me, neither of those that I saw ended with a solution other than "change IEEE of CC2531" while a proper solution, in my opinion, would be to use the new sticks' IEEE instead of the old one. Or maybe I am wrong? :-)

### What did you expect to happen?

I expected that I could use my old CC2531 flashed with router software without an IEEE address colision on the software layer.

### How to reproduce it (minimal and precise)

Use a CC2652P USB stick (or probably any other stick) as a new Coordinator and try to use the old Coordinator stick flashed with router software as router.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

CC2652P is 20210708,  CC2531 is 20200929 (router)

### Adapter

CC2652P

### Debug log

Nothing extraordinary shows up in the log. After pairing the CC2531 router I saw only this:
```
info 2022-11-06 14:04:53MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Coordinator","ieee_address":"0x00124b0014d9a936"},"type":"device_announce"}'
info 2022-11-06 14:04:53MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Coordinator"},"type":"device_announced"}'
```

jerrm, 2022-11-11:
I would consider this expected behavior.  My understanding is changing the coordinator ieee would break the network.

I don't know what triggers z2m to restore the coordinator from the backup, but if it had your zigbee net would likely be broken.  I suspect the devices would have needed re-pairing at minimum.  

No experience with a cc2531 stick, so can't help there.  If there isn't an easy way to change the ieee, I'm not sure it's worth the trouble for a $6 stick.

danieledwardgeorgehitchcock, 2022-11-11:
If you have just swapped out your coordinators without re-pairing your devices, this is mentioned [here](https://www.zigbee2mqtt.io/guide/faq/#what-does-and-does-not-require-repairing-of-all-devices) so I'd recommend scrapping your current network and re-building it with the CC2562P

lslomczynski, 2022-11-11:
@jerrm thank you for your reply :-)
Yes, I totally agree that this is the best solution for majority of users as it is fail-safe, allows for hussle free dongle swaps, promotes network stability and so on. I can imagine that re-pairing dozens of devices would be a huge disadvantage...
On the other side - I am at the beginnig of my ZigBee "adventure" with not so many devices yet (but with plans for the future) so I wouldn't mind re-pairing (if needed).
Also I like to keep things clean and I would love to use my present devices' IEEE instead of mimicking the old one :-)
I just couldn't find a way to do it so I posted here. You can see that the date form the short log that I pasted above is 4 days away from the github post date, so I really tried hard :-)

jerrm, 2022-11-11:
If you're resigned to a rebuild, delete coordinator_backup.json and reflash the stick.   I think that should force things.

closed 2022-11-16 by lslomczynski