Koenkk/zigbee2mqtt #14922: Configuration Error with TuYa WSD500A

its-fr4nki, 2022-11-10:
### What happened?

I added a new TuYa WSD500A to my Zigbee2MQTT system. 
The temperature sensor was integrated correctly, but no values are received from it.
In addition, this message appears in the log:

`022-11-11 00:03:33Failed to configure 'Temp 00', attempt 1 (Error: Read 0xa4c13858da15ae1e/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:658:23) at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:20) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:472:28) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/tuya.js:1613:13) at Configure.configure (/app/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))`

`Info 2022-11-11 00:03:33MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Temp 00"},"error":"Failed to configure (Read 0xa4c13858da15ae1e/1 genBasic([\"manufacturerName\",\"zclVersion\",\"appVersion\",\"modelId\",\"powerSource\",65534], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received))","status":"error","transaction":"u1jwq-5"}'`

### What did you expect to happen?

I expect the sensor to transmit the temperature to Zigbee2MQTT correctly. This is officially supported.

### How to reproduce it (minimal and precise)

I deleted the sensor and re-taught it with the same result. Also, I get the same result with a second same sensor.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

0x26720700

### Adapter

ConBee2

allcaponne, 2022-12-11:
Hello [его-fr4nki](https://github.com/its-fr4nki) Can you rezolved this problem?

TheChemicalWorkshop, 2022-12-18:
my all 4 sensors report N/A on all 4 values, tried repairing, hitting the reset button etc
they all say they are supported and finished configuring, havent looked into it yet (i simply lack the time)

any chance we can get them working?

TheChemicalWorkshop, 2022-12-18:
`Zigbee2MQTT:debug 2022-12-18 19:47:15: Device 'temp sensor in my hand' announced itself
Zigbee2MQTT:info  2022-12-18 19:47:15: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"temp sensor in my hand","ieee_address":"0xa4c138623059523b"},"type":"device_announce"}'
Zigbee2MQTT:debug 2022-12-18 19:47:17: Retrieving state of 'temp sensor in my hand' after reconnect
Zigbee2MQTT:debug 2022-12-18 19:47:20: Received Zigbee message from 'temp sensor in my hand', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-12-18 19:47:20: Received Zigbee message from 'temp sensor in my hand', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0`

here is a log i managed to get, default config (automatic)

DrDuk, 2022-12-26:
I've just bought one of these and I have the same problem.  Tried installing edge and no joy. Thanks

olaor, 2023-01-06:
Additional info:

My devices have the following model and manufacturer:
Zigbee Model TS0201
Zigbee Manufacturer _TZ3000_fllyghyj

I note that _TZ3000_bguser20 is said to be working.

DrDuk, 2023-01-06:
This was solved in https://github.com/Koenkk/zigbee2mqtt/issues/15729#event-8132492615

You need to define an external converter:
https://gist.github.com/Koenkk/a5ef6a151f1cc63a01183b47e3404541

Has to be run on Edge or the latest dev build

I can confirm that it works for  model 'TS0601' and manufacturer name '_TZE200_yjjdcqsq'

olaor, 2023-01-06:
It is already defined in node_modules/zigbee-herdsman-converters/devices/tuya.js

 {
        fingerprint: [
            {modelID: 'TS0201', manufacturerName: '_TZ3000_bguser20'},
            {modelID: 'TS0201', manufacturerName: '_TZ3000_fllyghyj'},
            {modelID: 'TS0201', manufacturerName: '_TZ3000_yd2e749y'},
            {modelID: 'TS0201', manufacturerName: '_TZ3000_6uzkisv2'},
            {modelID: 'TS0201', manufacturerName: '_TZ3000_xr3htd96'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_9yapgbuv'},
        ],
        model: 'WSD500A',
        vendor: 'TuYa',
        description: 'Temperature & humidity sensor',
        fromZigbee: [fzLocal.TS0201_battery, fz.temperature, fz.humidity],
        toZigbee: [],
        exposes: [e.battery(), e.temperature(), e.humidity(), e.battery_voltage()],
        configure: tuya.configureMagicPacket,
    },

Running 1.29.1 .

olaor, 2023-01-06:
Switched to dev branch as per https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html , still no temperature.

olaor, 2023-01-06:
Works very well in deconz, btw.

OneCreek, 2023-01-09:
> Switched to dev branch as per https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html , still no temperature.

for me it works: [image] with setup according to https://github.com/ioBroker/ioBroker.zigbee/issues/1649 except the voltage bug.

Frogowitsch, 2023-01-13:
I've got the same problem with the TS0201 _TZ3000_fllyghyj. On edge and still no go. Any progress on the subject ?

Frogowitsch, 2023-01-18:
My TS0201 _TZ3000_fllyghyj gets recognized, no problem. It also sends all values, but only for 2 or 3 mins, after that its dead. No more updates. Is this also a known problem ?

TheChemicalWorkshop, 2023-01-29:
any chance getting them working in zigbee2mqtt? or at least removing them from the page or marking as unsupported?

martas99, 2023-02-20:
> It is already defined in node_modules/zigbee-herdsman-converters/devices/tuya.js
>
> { fingerprint: [ {modelID: 'TS0201', manufacturerName: '_TZ3000_bguser20'}, {modelID: 'TS0201', manufacturerName: [...]

Hi, sorry for my ignorance - how should I add this piece of code to the configuration?

martas99, 2023-03-23:
Issue is solved[ here](https://github.com/Koenkk/zigbee2mqtt/discussions/16366)

closed 2023-05-01 by github-actions[bot]