Koenkk/zigbee2mqtt #14923: Hue Outdoor Motion Outdoor 9290030674: No image available for imageType '283'

m0wlheld, 2022-11-11:
### What happened?

Requested check for an OTA update for a recently added Hue Outdoor Motion Sensor [9290030674](https://www.zigbee2mqtt.io/devices/9290030674.html#philips-9290030674). Response in flyout and log was:

> Failed to check if update available for '{device id}' (No image available for imageType '283')

### What did you expect to happen?

Something like 'No update available' or 'Update available'.

### How to reproduce it (minimal and precise)

* Buy / get a Hue Outdoor Motion Sensor [9290030674](https://www.zigbee2mqtt.io/devices/9290030674.html#philips-9290030674) and pair it according to instructions
* Open Z2M frontend, goto tab OTA and request check for updates (reads: "Prüfen auf neue Updates" in German).

### Zigbee2MQTT version

1.28.1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

```
zigbee2mqtt  | Zigbee2MQTT:debug 2022-11-11 12:35:27: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"MY_DEVICE_ID","transaction":"0btwb-3"}'
zigbee2mqtt  | Zigbee2MQTT:info  2022-11-11 12:35:27: Checking if update available for 'MY_DEVICE_ID'
zigbee2mqtt  | Zigbee2MQTT:debug 2022-11-11 12:35:27: Check if update available for 'MY_DEVICE_ID' (SML004)
zigbee2mqtt  | Zigbee2MQTT:debug 2022-11-11 12:35:27: Using endpoint '2'
zigbee2mqtt  | Zigbee2MQTT:debug 2022-11-11 12:35:29: Received Zigbee message from 'MY_DEVICE_ID', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":33568006,"imageType":283,"manufacturerCode":4107}' from endpoint 2 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:debug 2022-11-11 12:35:29: Got OTA request '{"fieldControl":0,"manufacturerCode":4107,"imageType":283,"fileVersion":33568006}'
zigbee2mqtt  | Zigbee2MQTT:debug 2022-11-11 12:35:29: ZigbeeOTA: downloaded main index
zigbee2mqtt  | Zigbee2MQTT:info  2022-11-11 12:35:29: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"MY_DEVICE_ID"},"error":"Failed to check if update available for 'MY_DEVICE_ID' (No image available for imageType '283')","status":"error","transaction":"0btwb-3"}'
zigbee2mqtt  | Zigbee2MQTT:error 2022-11-11 12:35:29: Failed to check if update available for 'MY_DEVICE_ID' (No image available for imageType '283')
zigbee2mqtt  | Zigbee2MQTT:debug 2022-11-11 12:35:29: Error: No image available for imageType '283'
zigbee2mqtt  |     at getImageMeta (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:113:15)
zigbee2mqtt  |     at processTicksAndRejections (node:internal/process/task_queues:96:5)
zigbee2mqtt  |     at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:265:18)
zigbee2mqtt  |     at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:256:23)
zigbee2mqtt  |     at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:192:39)
```

m0wlheld, 2022-12-12:
Still present.

ftrueck, 2023-01-02:
Still present

m0wlheld, 2023-01-02:
Have you checked with most recent stable version 1.29.0? This issue has been addressed generally, possibly resolving yours, too.

pannal, 2023-01-04:
> Have you checked with most recent stable version 1.29.0? This issue has been addressed generally, possibly resolving yours, too.

Yep, still same issue with the latest stable.

m0wlheld, 2023-01-04:
Must confirm, still present with 1.29.0. Note: This only affects Hue Outdoor Motion Sensor modell [9290030674](https://www.zigbee2mqtt.io/devices/9290030674.html#philips-9290030674), while instances of modell [9290019758](https://www.zigbee2mqtt.io/devices/9290030674.html#philips-9290030674) do properly respond to OTA requests.

pannal, 2023-01-04:
For me it's happening with [9290030675](https://www.zigbee2mqtt.io/devices/9290030675.html#philips-9290030675)

m0wlheld, 2023-01-04:
I updated the subject, as the [9290030675](https://www.zigbee2mqtt.io/devices/9290030675.html#philips-9290030675) is not an **outdoor** motion sensor (or at least it is not designed to be). I also have an indoor motion sensor of modell [9290012607](https://www.zigbee2mqtt.io/devices/9290012607.html#philips-9290012607) which does not suffer from that error.

Could be that more recent versions of both indoor and outdoor sensors are affected?

So far, the error has been reported for modells [9290030674](https://www.zigbee2mqtt.io/devices/9290030674.html#philips-9290030674) and  [9290030675](https://www.zigbee2mqtt.io/devices/9290030675.html#philips-9290030675).

cor73x, 2023-01-06:
I have same error for [9290030675](https://www.zigbee2mqtt.io/devices/9290030675.html#philips-9290030675)

Gerben321, 2023-01-28:
Yep same here with 9290030675.

Xirt, 2023-02-27:
> I updated the subject, as the [9290030675](https://www.zigbee2mqtt.io/devices/9290030675.html#philips-9290030675) is not an **outdoor** motion sensor (or at least it is not designed to be). I also [...]

I have both new models and can confirm the issue for both. For the older indoor motion sensor (9290012607) the OTA updates are working fine.

jkerekes99, 2023-03-12:
I can confirm I just purchased a whole bunch of the Hue Motion Sensor V2s, tried to update them via Z2M and get the "no image available" message.

I really need to update them as they have the bug where they don't report LUX after motion clears.
The latest firmware is meant to fix this.

Frustratingly I purchased a Hue Bridge just so I could update them, however now I have found I have to have a hue light connected to the bridge before I can add or update any sensors! Argh Philips!

sirjmann92, 2023-04-10:
Getting the same error in zigbee2mqtt with a new SML003 (9290030675 indoor sensor) using a SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-E

FW version: 2.53.6
FW build date: 20210830
zigbee2mqtt 1.30.3 commit: 24c6b2e
Updated firmware on dongle: 7.1.4.0 build 389 
EZSP v9
FE 0.6.127

daimon82, 2023-05-05:
I think the "No image available" error is expected for this sensor.
From Z2M OTA documentation:

"Not all manufacturers make their updates available, below is a (not-complete) list of manufacturer/devices that support it:
...
Philips Hue devices (not every firmware is made available by them, in case not you will see the following exception in the log No image available for ...)"

Xirt, 2023-06-08:
I just upgraded to version 1.31.2 and can confirm the issue is still present.

Koenkk, 2023-06-08:
Someone has to grab the image from the official Hue hub, @TheJulianJES do you maybe have the image?

TheJulianJES, 2023-06-08:
I've checked again, but the 2nd gen motion sensors `9290030674` / `SML004` do not have any updates published yet.
`2.53.6` / `33568006` is the latest.

Xirt, 2023-06-08:
I believe the issue is that an error is thrown instead of indicating that finding  (e g. "No update available"). Isn't the root cause different from simply adding an update file that can be picked up?

TheJulianJES, 2023-06-08:
Yeah, I think Z2M works by specifying that a device supports OTA. And for "supporting OTA", Z2M requires at least one update file of any version for that device. Otherwise, you'll see that error.

IMO, it shouldn't be an error, but the device should still be marked with "OTA support".
As soon as an update is available, I'll make sure to PR it into the `zigbee-ota` repo.

Koenkk, 2023-06-08:
@TheJulianJES we can add the current firmware to the repo, that will fix this error.

TheJulianJES, 2023-06-09:
Sadly Hue only provides updates and not the image the device shipped with originally.

Koenkk, 2023-06-09:
I see, I've temporarily disabled the OTA feature for this device.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-09 by Koenkk

Xirt, 2023-06-09:
Sounds good, just wanted to make sure we cover the complete scope as you mention device and not devices (plural).

Throughout the discussion we found this issue is occurring for both models [9290030674](https://www.zigbee2mqtt.io/devices/9290030674.html#philips-9290030674) and [9290030675](https://www.zigbee2mqtt.io/devices/9290030675.html#philips-9290030675). Thanks for checking / confirming.

Koenkk, 2023-06-10:
@Xirt also fixed it for the [9290030675](https://www.zigbee2mqtt.io/devices/9290030675.html#philips-9290030675)

Timsche2210, 2024-03-15:
Do you have any idea why my SML003 (9290030675) is not listed in the OTA?
Detail say:
Firmware-Datum
20210830
Firmware-Version
2.53.6

Current software from Z2M is 1.36.0-1

Problem is, that - after receiving first values - the values don't change...

kris-anderson, 2024-04-08:
> Do you have any idea why my SML003 (9290030675) is not listed in the OTA? Detail say: Firmware-Datum 20210830 Firmware-Version 2.53.6
> 
> Current software from Z2M is 1.36.0-1
> 
> Problem is, that - after receiving first values - the values don't change...

I was wondering the same thing. I just installed one of these, and the low and medium motion sensor settings cause no motion to be detected. My first thought was to update the firmware, but the device isn't listed on the OTA tab.