Koenkk/zigbee2mqtt #14925: Can't pair new devices. Devices joins but interview never finishes (red exclamation mark)

eduardoll00, 2022-11-11:
### What happened?

Recently after a few upgrades, everytime I try to pair a new device it appears listed on the z2m devices screen but it keeps the red exclamation mark forever. So all the devices are now always failing the interview process. I've tried with 3 different devices that were working before (o removed from the network and tried to pair again)

### What did you expect to happen?

Devices is paired successfully and fully operational 

### How to reproduce it (minimal and precise)

Just try to add a new device 

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

0x26720700

### Adapter

Conbee2

### Debug log

Error 2022-11-11 08:58:53Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/15 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:58:53Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:58:53Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/15 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:58:53Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:58:56Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/15 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:58:56Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:58:56Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/15 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:58:56Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:58:59Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/15 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:58:59Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:58:59Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/15 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:58:59Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:59:03Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/15 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:59:03Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:59:03Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/15 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:59:03Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:59:06Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/15 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:59:06Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 08:59:06Publish 'set' 'state' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/15 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
[... cut ...]
Error 2022-11-11 09:00:06Publish 'set' 'color' to '0x00124b001ba74175' failed: 'Error: Command 0x00124b001ba74175/11 lightingColorCtrl.moveToColor({"transtime":4,"colorx":37486,"colory":25493}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-11 09:00:26Failed to interview '0x00124b001ba74175', device has not successfully been paired

znanev, 2022-11-11:
I've got the same problem since updating to the latest Dev version yesterday. Dongrading to the latest release (from master) fixes the issues.

What seems to happen is the z2m process dies and restarts itself, hence the newly added devices don't complete the interview process.

Following is extract from the syslog, concerning the crash of the z2m process:

```
Nov 11 12:02:16 RasPiBridge npm[1387]: Zigbee2MQTT:warn  2022-11-11 12:02:16: Device '0x00124b00228123a8' left the network
Nov 11 12:02:24 RasPiBridge npm[1387]: /opt/zigbee2mqtt/lib/extension/homeassistant.ts:1464
Nov 11 12:02:24 RasPiBridge npm[1387]:         if (entity.isDevice() && entity.definition.ota) {
Nov 11 12:02:24 RasPiBridge npm[1387]:                                                    ^
Nov 11 12:02:24 RasPiBridge npm[1387]: TypeError: Cannot read properties of null (reading 'ota')
Nov 11 12:02:24 RasPiBridge npm[1387]:     at HomeAssistant.adjustMessageBeforePublish (/opt/zigbee2mqtt/lib/extension/homeassistant.ts:1464:52)
Nov 11 12:02:24 RasPiBridge npm[1387]:     at Controller.publishEntityState (/opt/zigbee2mqtt/lib/controller.ts:266:49)
Nov 11 12:02:24 RasPiBridge npm[1387]:     at Object.publishLastSeen (/opt/zigbee2mqtt/lib/util/utils.ts:348:9)
Nov 11 12:02:24 RasPiBridge npm[1387]:     at EventEmitter.<anonymous> (/opt/zigbee2mqtt/lib/controller.ts:165:29)
Nov 11 12:02:24 RasPiBridge npm[1387]:     at EventEmitter.emit (node:events:525:35)
Nov 11 12:02:24 RasPiBridge npm[1387]:     at EventBus.emitLastSeenChanged (/opt/zigbee2mqtt/lib/eventBus.ts:52:22)
Nov 11 12:02:24 RasPiBridge npm[1387]:     at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.ts:66:27)
Nov 11 12:02:24 RasPiBridge npm[1387]:     at Controller.emit (node:events:513:28)
Nov 11 12:02:24 RasPiBridge npm[1387]:     at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
Nov 11 12:02:24 RasPiBridge npm[1387]:     at Controller.onDeviceJoined (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:557:14)
Nov 11 12:02:25 RasPiBridge systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Nov 11 12:02:25 RasPiBridge systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Nov 11 12:02:25 RasPiBridge systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Nov 11 12:02:25 RasPiBridge systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 4.
Nov 11 12:02:25 RasPiBridge systemd[1]: Stopped zigbee2mqtt.
Nov 11 12:02:25 RasPiBridge systemd[1]: Started zigbee2mqtt.
Nov 11 12:02:27 RasPiBridge npm[1737]: > zigbee2mqtt@1.28.2-dev start
Nov 11 12:02:27 RasPiBridge npm[1737]: > node index.js
```

I first thought that this might be due to not using the latest Home Assistant version (as seen the crash  happens in the homeassistant.ts), but I've just upgraded to the latest (2022.11.2) and removing and re-pairing the same device (a CC2652P based router) procuces the same crash.

eduardoll00, 2022-11-11:
> I've got the same problem since updating to the latest Dev version yesterday. Dongrading to the latest release (from master) fixes the issues.
> 
> What seems to happen is the z2m process dies and restarts itself, hence the newly added devices don't complete the interview process.
> 
> Following is extract from the syslog, concerning the crash of the z2m process:
> 
> ```
> Nov 11 12:02:16 RasPiBridge npm[1387]: Zigbee2MQTT:warn  2022-11-11 12:02:16: Device '0x00124b00228123a8' left the network
> Nov 11 12:02:24 RasPiBridge npm[1387]: /opt/zigbee2mqtt/lib/extension/homeassistant.ts:1464
> Nov 11 12:02:24 RasPiBridge npm[1387]:         if (entity.isDevice() && entity.definition.ota) {
> Nov 11 12:02:24 RasPiBridge npm[1387]:                                                    ^
> Nov 11 12:02:24 RasPiBridge npm[1387]: TypeError: Cannot read properties of null (reading 'ota')
> Nov 11 12:02:24 RasPiBridge npm[1387]:     at HomeAssistant.adjustMessageBeforePublish (/opt/zigbee2mqtt/lib/extension/homeassistant.ts:1464:52)
> Nov 11 12:02:24 RasPiBridge npm[1387]:     at Controller.publishEntityState (/opt/zigbee2mqtt/lib/controller.ts:266:49)
> Nov 11 12:02:24 RasPiBridge npm[1387]:     at Object.publishLastSeen (/opt/zigbee2mqtt/lib/util/utils.ts:348:9)
> Nov 11 12:02:24 RasPiBridge npm[1387]:     at EventEmitter.<anonymous> (/opt/zigbee2mqtt/lib/controller.ts:165:29)
> Nov 11 12:02:24 RasPiBridge npm[1387]:     at EventEmitter.emit (node:events:525:35)
> Nov 11 12:02:24 RasPiBridge npm[1387]:     at EventBus.emitLastSeenChanged (/opt/zigbee2mqtt/lib/eventBus.ts:52:22)
> Nov 11 12:02:24 RasPiBridge npm[1387]:     at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.ts:66:27)
> Nov 11 12:02:24 RasPiBridge npm[1387]:     at Controller.emit (node:events:513:28)
> Nov 11 12:02:24 RasPiBridge npm[1387]:     at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
> Nov 11 12:02:24 RasPiBridge npm[1387]:     at Controller.onDeviceJoined (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:557:14)
> Nov 11 12:02:25 RasPiBridge systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
> Nov 11 12:02:25 RasPiBridge systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
> Nov 11 12:02:25 RasPiBridge systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
> Nov 11 12:02:25 RasPiBridge systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 4.
> Nov 11 12:02:25 RasPiBridge systemd[1]: Stopped zigbee2mqtt.
> Nov 11 12:02:25 RasPiBridge systemd[1]: Started zigbee2mqtt.
> Nov 11 12:02:27 RasPiBridge npm[1737]: > zigbee2mqtt@1.28.2-dev start
> Nov 11 12:02:27 RasPiBridge npm[1737]: > node index.js
> ```
> 
> I first thought that this might be due to not using the latest Home Assistant version (as seen the crash happens in [homeassistant.ts](https://github.com/Koenkk/zigbee2mqtt/blob/98a6a2dfe815388dc2383eb0e0ddaf2555783de9/lib/extension/homeassistant.ts#L1464)), but I've just upgraded to the latest (2022.11.2) and removing and re-pairing the same device (a CC2652P based router) procuces the same crash.

What is the version that is running well? Is there a easy way to downgrade? I can't find this option in the HA's addons screen

znanev, 2022-11-11:
@eduardoll00 If you can downgrade to the latest release (1.28.2), then you'll be able to pair your devices. The bug is introduced in the the development branch.

I'm not sure how you can do this from within HA's addon, as I'm not using addons and Z2M is running as stand alone application on a different Raspberry Pi.

eduardoll00, 2022-11-11:
@znanev I'm wondering if my issue is not a different one because I'm not using the beta releases from the development branch

znanev, 2022-11-11:
@eduardoll00 Might be a different one indeed, if you're on the latest release. In my case, Z2M "dies" when interviewing devices and immediately restarts itself - I'm not seeing the same from your logs.

So these might be two separate issues, which might be benefitial to split.

znanev, 2022-11-15:
Just chiming in to report, that my issue is fixed by commit 489d1de (tested with latest code from dev - 16398b78).

cvicari, 2022-11-18:
I apparently had the same problem as @eduardoll00, using
zigbee2mqtt version 1.28.2 - commit: [360a777]
adapter EZSP v8, Coordinator revision 6.10.3.0 build 297

This happens with a couple of devices. This is the excerpt of my failure log while trying to pair an AQSZB-110 (https://www.zigbee2mqtt.io/devices/AQSZB-110.html). This device used to work, some day ago I tried to pair it after a problem on my PAN.
I do not see "debug" lines from zigbee-herdsman so maybe I could not activate its debug log. 

Zigbee2MQTT:info  2022-11-18 12:07:15: Device '0x0015bc00360010d6' joined
Zigbee2MQTT:info  2022-11-18 12:07:15: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00360010d6","ieee_address":"0x0015bc00360010d6"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-11-18 12:07:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0015bc00360010d6"},"type":"device_connected"}'
Zigbee2MQTT:info  2022-11-18 12:07:15: MQTT publish: topic 'zigbee2mqtt/0x0015bc00360010d6/availability', payload 'online'
Zigbee2MQTT:info  2022-11-18 12:07:15: MQTT publish: topic 'zigbee2mqtt/0x0015bc00360010d6', payload '{"last_seen":"2022-11-18T12:07:15+01:00"}'
Zigbee2MQTT:info  2022-11-18 12:07:15: Starting interview of '0x0015bc00360010d6'
Zigbee2MQTT:info  2022-11-18 12:07:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00360010d6","ieee_address":"0x0015bc00360010d6","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-11-18 12:07:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0015bc00360010d6"},"type":"pairing"}'
Zigbee2MQTT:info  2022-11-18 12:07:16: MQTT publish: topic 'zigbee2mqtt/0x0015bc00360010d6', payload '{"last_seen":"2022-11-18T12:07:16+01:00"}'
Zigbee2MQTT:info  2022-11-18 12:07:22: MQTT publish: topic 'zigbee2mqtt/CC2531 router', payload '{"last_seen":"2022-11-18T12:07:22+01:00","led":false,"linkquality":164}'
Zigbee2MQTT:error 2022-11-18 12:07:25: Failed to interview '0x0015bc00360010d6', device has not successfully been paired

cvicari, 2022-12-01:
I tried again, here is my zigbee-herdsman debug log, I think this is the relevant excerpt of the log, full log file is attached.

2022-12-01T10:42:30.516Z zigbee-herdsman:adapter:ezsp:driv Node_Desc_rsp  frame: {"type":"Buffer","data":[14,128,124,199,0]}
2022-12-01T10:42:30.518Z zigbee-herdsman:adapter:ezsp:debg Node descriptor request for '51068' failed (RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds), retry
2022-12-01T10:42:30.519Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2022-12-01T10:42:30.520Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2022-12-01T10:42:30.520Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x0015bc00360010d6', attempt 5
2022-12-01T10:42:30.522Z zigbee-herdsman:adapter:ezsp:debg Requesting 'Node Descriptor' for '51068'
2022-12-01T10:42:30.523Z zigbee-herdsman:adapter:ezsp:driv ZDO Node_Desc_req params: {"dstaddr":51068}
2022-12-01T10:42:30.526Z zigbee-herdsman:adapter:ezsp:ezsp ==> setExtendedTimeout: {"remoteEui64":{"_value":{"type":"Buffer","data":[0,21,188,0,54,0,16,214]}},"extendedTimeout":true}
2022-12-01T10:42:30.527Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setExtendedTimeout","_id_":126,"_isRequest_":true,"remoteEui64":{"_value":{"type":"Buffer","data":[0,21,188,0,54,0,16,214]}},"extendedTimeout":true}
2022-12-01T10:42:30.528Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,7,0): 4800017e00d610003600bc150001
2022-12-01T10:42:30.530Z zigbee-herdsman:adapter:ezsp:uart --> [070a21a92a2ac3a259a24a99bf559307827e]
2022-12-01T10:42:30.531Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
2022-12-01T10:42:30.539Z zigbee-herdsman:adapter:ezsp:uart <-- [710aa1a92a2a9b487e]
2022-12-01T10:42:30.540Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,1,0): 710aa1a92a2a9b487e
2022-12-01T10:42:30.540Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2022-12-01T10:42:30.541Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2022-12-01T10:42:30.542Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4880017e00
2022-12-01T10:42:30.543Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x7e: {"_id_":126,"_cls_":"setExtendedTimeout","_isRequest_":false}
2022-12-01T10:42:30.544Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (1) success
2022-12-01T10:42:30.546Z zigbee-herdsman:adapter:ezsp:ezsp ==> sendUnicast: {"type":0,"indexOrDestination":51068,"apsFrame":{"clusterId":2,"profileId":0,"sequence":15,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":320},"messageTag":16,"message":{"type":"Buffer","data":[15,124,199]}}
2022-12-01T10:42:30.547Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"sendUnicast","_id_":52,"_isRequest_":true,"type":0,"indexOrDestination":51068,"apsFrame":{"clusterId":2,"profileId":0,"sequence":15,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":320},"messageTag":16,"message":{"type":"Buffer","data":[15,124,199]}}
2022-12-01T10:42:30.549Z zigbee-herdsman:adapter:ezsp:uart --> DATA (1,0,0): 4900013400007cc7000002000000400100000f10030f7cc7
2022-12-01T10:42:30.550Z zigbee-herdsman:adapter:ezsp:uart --> [100b21a9602a15ce9e944a27aa5592099d4e27a4fdcd68f73a8f1e7e]
2022-12-01T10:42:30.551Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (2)
2022-12-01T10:42:30.561Z zigbee-herdsman:adapter:ezsp:uart <-- [020ba1a9602a15f6b8ad7e]
2022-12-01T10:42:30.562Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,2,0): 020ba1a9602a15f6b8ad7e
2022-12-01T10:42:30.563Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2022-12-01T10:42:30.564Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2022-12-01T10:42:30.565Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 49800134000044
2022-12-01T10:42:30.567Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x34: {"_id_":52,"_cls_":"sendUnicast","_isRequest_":false,"status":0,"sequence":68}
2022-12-01T10:42:30.568Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (2) success
2022-12-01T10:42:32.203Z zigbee-herdsman:adapter:ezsp:uart <-- [120bb1a96b2a15ce9e944a27aa5592099c4e27effdce67f0e07e]
2022-12-01T10:42:32.204Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,2,0): 120bb1a96b2a15ce9e944a27aa5592099c4e27effdce67f0e07e
2022-12-01T10:42:32.204Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2022-12-01T10:42:32.205Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2022-12-01T10:42:32.206Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4990013f00007cc70000020000004000000044100000
2022-12-01T10:42:32.208Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x3f: {"_id_":63,"_cls_":"messageSentHandler","_isRequest_":false,"type":0,"indexOrDestination":51068,"apsFrame":{"profileId":0,"clusterId":2,"sourceEndpoint":0,"destinationEndpoint":0,"options":64,"groupId":0,"sequence":68},"messageTag":16,"status":0,"message":{"type":"Buffer","data":[]}}
2022-12-01T10:42:32.216Z zigbee-herdsman:adapter:ezsp:uart <-- [220bb1a97d312a15b25996ca25aa5593499c40e76791099874f8c9e3f53b7d5e77c37e]
2022-12-01T10:42:32.217Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,2,0): 220bb1a9112a15b25996ca25aa5593499c40e76791099874f8c9e3f53b7e77c37e0000
2022-12-01T10:42:32.217Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (3)
2022-12-01T10:42:32.218Z zigbee-herdsman:adapter:ezsp:uart --> [83401b7e]
2022-12-01T10:42:32.219Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 499001450000000002800000000100000ec0cc7cc7ffff050f807cc7004864
[... cut ...]
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:567:17)'
2022-12-01T10:42:32.619Z zigbee-herdsman:controller:error Interview failed for '0x0015bc00360010d6 with error 'Error: Interview failed because can not get node descriptor ('0x0015bc00360010d6')'
Zigbee2MQTT:error 2022-12-01 11:42:32: Failed to interview '0
x0015bc00360010d6', device has not successfully been paired

[log-herdsman-aqszb-110-cannot_pair.txt.gz](https://github.com/Koenkk/zigbee2mqtt/files/10131651/log-herdsman-aqszb-110-cannot_pair.txt.gz)

cvicari, 2023-01-01:
I updated z2m to the latest version just today, and I confirm that the problem is still there

adamhassel, 2023-01-17:
Also an issue for me, with an Ikea motion sensor.

scontomio, 2023-01-20:
@cvicari I have the same problem with AQSZB-110 and adapter EZSP v8 - Sonoff ZBDongle-E

adamhassel, 2023-02-20:
My issues are also with Sonoff ZB Dongle E. Right now, z2m is not complaining that the motion sensor is not paired. Instead, it's simply dropping it from its list of devices. Later, it'll reappear.

closed 2023-03-31 by github-actions[bot]

wwwdesign, 2024-01-22:
Any news on the matter? I got the same problem on:

> Core 2024.1.5
> Supervisor 2023.12.1
> Operating System 11.4
> Frontend 20240104.0

and Zigbee2MQTT 1.35.1-1

manuelcoders, 2024-01-22:
I solved the problem by updating the adapter firmware.