Koenkk/zigbee2mqtt #14926: [New device support]: Bosch Radiator Thermostat II

niklasarnitz, 2022-11-11:
### Link

https://www.bosch-smarthome.com/at/de/produkte/geraete/heizkoerper-thermostat/

### Database entry

{"id":22,"type":"EndDevice","ieeeAddr":"0x18fc26000002a2d7","nwkAddr":39513,"manufId":4617,"manufName":"BOSCH","powerSource":"Battery","modelId":"RBSH-TRV0-ZB-EU","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,4,32,513,516,2821],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"RBSH-TRV0-ZB-EU","manufacturerName":"BOSCH","powerSource":3,"zclVersion":8,"appVersion":1,"stackVersion":5,"hwVersion":1,"dateCode":"20220627","swBuildId":"3.02.05"}},"genPollCtrl":{"attributes":{"checkinInterval":6480}},"hvacThermostat":{"attributes":{"20496":0}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c20f21","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":5,"hwVersion":1,"dateCode":"20220627","swBuildId":"3.02.05","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1668172257367,"defaultSendRequestWhen":"active"}

### Comments

Here are some things I found out and got working already:
- It pairs successfully after adding the Install Code
- I cannot get it to read the temperature or write the temperature. Maybe I'm just not understanding the concepts of z2mqtt that well.
Here is my first draft of a config:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['RBSH-TRV0-ZB-EU'],
    model: 'Radiator thermostat II',
    vendor: 'Bosch',
    description: 'Radiator thermostat',
    fromZigbee: [],
    toZigbee: [fz.thermostat, fz.hvac_user_interface],
    exposes: [
	   e.child_lock(),
	   exposes.climate()
	   	.withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET)
	   	.withLocalTemperature()
    ],
};

module.exports = definition;
```
Comment: The setpoint range and interval should be correct.

itkama, 2022-11-11:
This seems to be a duplicate of #14005 - though more people with this device should hopefully speed things up :)

danieledwardgeorgehitchcock, 2022-11-11:
You might want to add a `fromZigbee` converter to read back the temperature from the device.

If you have the Z2M GUI enabled, you can see what clusters are supported by the endpoints, in the device clusters tab.

You can also see what the device reports back using the device dev tab and reading from the various clusters.

danieledwardgeorgehitchcock, 2022-11-11:
Try adding `fz.thermostat` to your `fromZigbee` section

itkama, 2022-11-11:
@danieledwardgeorgehitchcock 
I've played around a bit with the dev tab you mentioned.

Via Endpoint 1, Cluster hvacThermostat I can read the following attributes:
maxHeatSetpointLimit
minHeatSetpointLimit
localTemp
occupiedHeatingSetpoint 

I can also set the last one if I write instead of reading it.
For the last one I get this answer:
`Read result of 'hvacThermostat': {"occupiedHeatingSetpoint":2000}`
It's the current heating setpoint which is set at 20°C.

How would I now add these commands to converter file @niklasarnitz started above?

niklasarnitz, 2022-11-11:
I'll be trying this tomorrow.
But the last thread entry by @danieledwardgeorgehitchcock should propably work.

itkama, 2022-11-11:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['RBSH-TRV0-ZB-EU'],
    model: 'Radiator Thermostat II',
    vendor: 'Bosch',
    description: 'Radiator Thermostat',
    fromZigbee: [fz.thermostat, fz.hvac_user_interface, fz.battery],
    toZigbee: [tz.thermostat_occupied_heating_setpoint],
    exposes: [
        e.battery(),
	    exposes.climate()
	    .withSetpoint('occupied_heating_setpoint', 5, 30, 0.5, ea.STATE_SET)
	    .withLocalTemperature()
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['hvacThermostat', 'genPowerCfg']);
        await reporting.thermostat_occupied_heating_setpoint(endpoint);
        await reporting.thermostatTemperature(endpoint);
        await reporting.batteryVoltage(endpoint);
        await reporting.battery(endpoint);
    },
};

module.exports = definition;`

This is my current result. Changing the setpoint works, temperature gets reported and also the battery percentage:
[image]

What would definetly be nice to have: A current running state of the device to have it display that (in my case) in Home Assistant.

itkama, 2022-11-11:
`const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ["RBSH-TRV0-ZB-EU"],
    model: "Radiator Thermostat II",
    vendor: "Bosch",
    description: "Radiator Thermostat",
    fromZigbee: [fz.thermostat, fz.hvac_user_interface, fz.battery],
    toZigbee: [tz.thermostat_occupied_heating_setpoint],
    exposes: [
        e.battery(),
	    exposes.climate()
	    .withSetpoint('occupied_heating_setpoint', 5, 30, 0.5, ea.STATE_SET)
	    .withLocalTemperature()
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['hvacThermostat', 'genPowerCfg']);
        await reporting.thermostat_occupied_heating_setpoint(endpoint);
        await reporting.thermostatTemperature(endpoint);
        await reporting.batteryVoltage(endpoint);
        await reporting.battery(endpoint);
    },
};

module.exports = definition;`

niklasarnitz, 2022-11-12:
When using @itkama 's config, I get a few different errors:

When I try and refresh the temperature:
```
Debug 2022-11-12 03:19:28 Received MQTT message on 'zigbee2mqtt/0x18fc26000002a2d7/get' with data '{"local_temperature":""}'
Error 2022-11-12 03:19:28 No converter available for 'local_temperature' ("")
```

And it seems that the binding  process of genPowerCfg fails:
```
x18fc26000005976f/1 genPowerCfg from '0x00124b0024c20f21/1' failed (AREQ - ZDO - bindRsp after 10000ms)
zigbee2mqtt_1        |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
zigbee2mqtt_1        |     at listOnTimeout (node:internal/timers:559:17)
zigbee2mqtt_1        |     at processTimers (node:internal/timers:502:7))
zigbee2mqtt_1        | Zigbee2MQTT:info  2022-11-12 03:12:37: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x18fc26000005976f"},"error":"Failed to configure (Bind 0x18fc26000005976f/1 genPowerCfg from '0x00124b0024c20f21/1' failed (AREQ - ZDO - bindRsp after 10000ms))","status":"error","transaction":"uxdrr-1"}'
```
Setting the temperature works.

Here's a screenshot of my GUI:
[image]

itkama, 2022-11-12:
Interestingly I get the same error when trying to refresh the temperature manually. But the thermostat seems to be doing that itself quite frequently in the background. But those messages from the thermostat only started appearing once I poked it via the dev console.

niklasarnitz, 2022-11-12:
Refresh now works for me. Also the Local Temp works.

But here is another error:
```
zigbee2mqtt_1        | Zigbee2MQTT:error 2022-11-12 17:00:27: Failed to configure 'EG_Zimmer_Straße_Thermostat_Rechts', attempt 2 (ReferenceError: reporting is not defined
zigbee2mqtt_1        |     at Object.configure (/app/dist/util/externally-loaded.js:23:9)
zigbee2mqtt_1        |     at Configure.configure (/app/lib/extension/configure.ts:115:37)
zigbee2mqtt_1        |     at EventEmitter.<anonymous> (/app/lib/extension/configure.ts:80:62)
zigbee2mqtt_1        |     at EventEmitter.emit (node:events:525:35)
zigbee2mqtt_1        |     at EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:52:22)
zigbee2mqtt_1        |     at Controller.<anonymous> (/app/lib/zigbee.ts:66:27)
zigbee2mqtt_1        |     at Controller.emit (node:events:513:28)
zigbee2mqtt_1        |     at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
zigbee2mqtt_1        |     at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:727:18)
zigbee2mqtt_1        |     at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70))
```

danieledwardgeorgehitchcock, 2022-11-12:
> Refresh now works for me. Also the Local Temp works.
>
> But here is another error:
>
> ```
> zigbee2mqtt_1        | Zigbee2MQTT:error 2022-11-12 17:00:27: Failed to configure [...]

Add `const reporting = require('zigbee-herdsman-converters/lib/reporting');` into your declarations at the top of your file

itkama, 2022-11-12:
@danieledwardgeorgehitchcock I'm getting this error now:
`Failed to configure 'Manuel Thermostat', attempt 3 (TypeError: reporting.thermostat_occupied_heating_setpoint is not a function at Object.configure (/app/dist/util/externally-loaded.js:25:25) at Configure.configure (/app/lib/extension/configure.ts:115:13))`

danieledwardgeorgehitchcock, 2022-11-12:
replace `await reporting.thermostat_occupied_heating_setpoint(endpoint);` with `await reporting.thermostatOccupiedHeatingSetpoint(endpoint);`

danieledwardgeorgehitchcock, 2022-11-12:
It's telling you what the issue is. `localTemperatureCalibration` is not reportable from the device

itkama, 2022-11-12:
The question would be why it is telling me this? Via the console of this device it is possible:
[image]

danieledwardgeorgehitchcock, 2022-11-12:
You're reading an attribute there, not asking it to report it given a threshold or interval. Try doing the same thing manually in the reporting tab and you'll likely get the same error...

mmattel, 2022-11-20:
Is there any progress so that the new device info can be added and published in a near future?

niklasarnitz, 2022-11-20:
I have a working device config now.

I’ll create a PR shortly.

Koenkk, 2022-11-20:
Added by @niklasarnitz (in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4983)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-20 by Koenkk

mmattel, 2022-11-21:
@Koenkk do you have an idea when a new stable release will be published?

BBJake, 2022-11-25:
[image]

I added some more functions for the device and mapped the system_mode to the Bosch configuration_mode

BBJake, 2022-11-25:
```
const herdsman = require('zigbee-herdsman');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const utils = require("zigbee-herdsman-converters//lib/utils");
const assert = require('assert');
const e = exposes.presets;
const ea = exposes.access;

const boschManufacturer = {manufacturerCode: 0x1209};

const operatingModes = {
    'automatic': 0,
    'manual': 1,
    'pause': 5,
};

const stateOffOn = {
    'OFF': 0,
    'ON': 1,
};

const displayOrientation = {
    'normal': 0,
    'flipped': 1,
};

const tzLocal = {
    bosch_thermostat: {
        key: ['window_open','boost','system_mode'],
        convertSet: async (entity, key, value, meta) => {
            if (key === 'window_open') {
		value = value.toUpperCase();
                utils.validateValue(value, Object.keys(stateOffOn));
                const index = stateOffOn[value];
                await entity.write('hvacThermostat', {0x4042: {value: index, type: herdsman.Zcl.DataType.enum8}}, boschManufacturer);
                return {state: {window_open: value}};
            }
            if (key === 'boost') {
                value = value.toUpperCase();
                utils.validateValue(value, Object.keys(stateOffOn));
                const index = stateOffOn[value];
                await entity.write('hvacThermostat', {0x4043: {value: index, type: herdsman.Zcl.DataType.enum8}}, boschManufacturer);
                return {state: {boost: value}};
            }
            if (key === 'system_mode') {
		// Map system_mode (Off/Auto/Heat) to Boschg operating mode
                value = value.toLowerCase();

		let opMode = 1; // OperatingMode 1 = Manual (Default)
		if (value=='off') { opMode = 5 } // OperatingMode 5 = Pause 
		else if (value=='auto') { opMode = 0 } // OperatingMOde 1 = Automatic 

                await entity.write('hvacThermostat', {0x4007: {value: opMode, type: herdsman.Zcl.DataType.enum8}}, boschManufacturer);
                return {state: {system_mode: value}};
            }

        },
        convertGet: async (entity, key, meta) => {
            switch (key) {
            case 'window_open':
                await entity.read('hvacThermostat', [0x4042], boschManufacturer);
                break;
            case 'boost':
                await entity.read('hvacThermostat', [0x4043], boschManufacturer);
                break;
            case 'system_mode':
                await entity.read('hvacThermostat', [0x4007], boschManufacturer);
[... 115 lines cut ...]
		.withDescription('Display orientation'),
	exposes.numeric('display_ontime', ea.ALL).withValueMin(5).withValueMax(30)
                .withDescription('Specifies the diplay On-time'),
	exposes.numeric('display_brightness', ea.ALL).withValueMin(0).withValueMax(10)
                .withDescription('Specifies the brightness value of the display'),
        e.battery(),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'hvacThermostat','hvacUserInterfaceCfg']);
	await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatTemperature(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);

	await endpoint.read('hvacThermostat',['localTemperatureCalibration']);
        await endpoint.read('hvacThermostat', [0x4007, 0x4042, 0x4043], boschManufacturer);

        await endpoint.read('hvacUserInterfaceCfg', ['keypadLockout']);
	await endpoint.read('hvacUserInterfaceCfg', [0x400b, 0x403a, 0x403b], boschManufacturer);
     }
};

module.exports = definition;

```

mmattel, 2022-11-25:
@BBJake you may need to create a PR at https://github.com/Koenkk/zigbee-herdsman-converters/ to get that published (new release at the beginning of Dcember...), see the [initial PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4983) above.

BBJake, 2022-11-28:
Thanks for your integration of my code.
I'm not sure how I can test the upgrade, because I use the addon in "Home Automation Supervisor".
As this runs in a docker container, I think I cannot directly use your branch...

BBJake, 2022-11-28:
Another question.
What firmware version do you have on your Thermostat II?
For me it shows: 3.02.05 (Date: 20220627).
According to [Update 2022-10-27](https://www.bosch-smarthome.com/at/de/service/aktuelle-meldungen/release-infos/system-solutions/2022-10-27.html) and [Update 2022-11-14](https://www.bosch-smarthome.com/at/de/service/aktuelle-meldungen/release-infos/system-solutions/2022-11-14.html) there are some interesting updates...
I do not own a Bosch Smarthome Controler. So the question is: Is there a way to get the update and use it with Z2M OTA?

itkama, 2022-11-28:
> Another question. What firmware version do you have on your Thermostat II? For me it shows: 3.02.05 (Date: 20220627). According to [Update [...]

Same version on my thermostat (that is working wonderfully with your code, thank you for that). Regarding getting our hands on the ota-File - I don't think Bosch is willing to provide that. I guess we would need someone with the Bosch SHC and a firewall that can log every request to get the download links for the thermostat.
I did how ever a[dded a feature request post on the bosch forums](https://community.bosch-smarthome.com/t5/feedback-und-w%C3%BCnsche/frei-verf%C3%BCgbare-firmware-update-files/m-p/20695#M2534) to maybe convince them to provide the files. I don't have high hopes - but can't hurt to ask.

Nezz, 2022-12-12:
Would it be possible to expose how much the valve is open as a sensor? The Bosch app shows it and when connecting the thermostats to HA via the Bosch Smart Controller it's exposed too.

mmattel, 2022-12-12:
> Would it be possible to expose how much the valve is open as a sensor?

@BBJake any ideas?

PaulLeichtfried, 2022-12-12:
I would like to take up the update topic again, because the radiator thermostats with the delivered FW version work anything but precise for many users.

Is there any indication of encryption/authentication in the official update process? Otherwise, one could try to obtain a FW using Wireshark or similar. Are there any efforts in this direction or does anyone here own the Bosch Smart Home Controller and could try their luck?

Maybe we should also create a discussion, as I think many would be happy about updates.

itkama, 2022-12-12:
@PaulLeichtfried You can vote for it on the Bosch forums, see link above. I don't think Bosch will provide them. Still got a little bit of hope left though 😅.
Regarding the wireshark part: As long as there is no authentification of the communication between the SHC and the Bosch servers one could get the link. Maybe a new issue to discuss things regardings Bosch firmware updates is in order, since this issue is already closed anyway.

niklasarnitz, 2022-12-12:
@itkama @PaulLeichtfried I just had a look into a decompiled version of the Bosch SmartHome app.
There are no signs of the firmware being downloaded in the app and then being installed via the gateway.
So the gateway itself would have to download the firmware.
So imo the best way to get access to the firmware images is to get a SHC and then dump it's firmware and try to have a look at the update fetching process that way.

They go for 40-50 Euros on eBay-Kleinanzeigen where I live, so if one or two more would give 10 Euros, I could buy one and extract the firmware and try to find the images.
I also have the Thermostat II so I could also test it.

danieledwardgeorgehitchcock, 2022-12-12:
I already looked into dumping the controller firmware or gaining root access - It's not for the faint hearted! there are 3 microcontrollers all with SRAM that _could_ contain firmware - there are also some unlabelled diagnostic ports but I could not work out the communication between it all.. I also decompiled the phone app too and can confirm that it uses an API to talk to the hub - the hub is where the main grunt work is done

niklasarnitz, 2022-12-12:
@danieledwardgeorgehitchcock Could you shoot me an email? Maybe we could find something together.
I have a fair amount of hardware hacking experience.

PaulLeichtfried, 2022-12-12:
@itkama I will upvote your contribution, I don't really think Bosch will publish anything either, but it's worth a try.

@niklasarnitz I will gladly contribute 10€ if it brings us closer to the goal.

@danieledwardgeorgehitchcock But wouldn't it be easier to first monitor the network traffic from the SHC than to somehow get the firmware from the SHC? Have you already tried something in this direction, is there anything obvious that prevents it?

danieledwardgeorgehitchcock, 2022-12-12:
The traffic from the controller is HTTPS encrypted and MITM proxying is useless without being able to install a root certificate on the controller itself

niklasarnitz, 2022-12-12:
@danieledwardgeorgehitchcock Do you have pictures of the PCB? I have just ordered a SHC and would like to research a bit until it arrives.

danieledwardgeorgehitchcock, 2022-12-12:
I can deffo take some tomorrow

niklasarnitz, 2022-12-12:
That would be awesome.

BBJake, 2022-12-12:
> > Would it be possible to expose how much the valve is open as a sensor?
> 
> @BBJake any ideas?

I have not seen the thermostat exposing that value. But it should be possible to get it on demand.
I will try something tomorrow, but I cannot verify it as don't have the SHC to see if the value is correct...

BBJake, 2022-12-13:
I guess it works... but the value is not published automatically by the device...
So you have to manually update:
[image]

danieledwardgeorgehitchcock, 2022-12-13:
Here's the photos of the hub:

[Bottom](https://drive.google.com/file/d/1HPRX5XK0qdRo14vgHfuk7cyoazhA43eY/view?usp=share_link)
[Top](https://drive.google.com/file/d/1HYNK4ki5JxeLWrXc3cCUPSkRY8ktfm1o/view?usp=share_link)
[Possible UART / Debug](https://drive.google.com/file/d/1Hd92lVlqeCq00d5MMkPdvDy_kCTQd9Fd/view?usp=share_link)
[Side Ports](https://drive.google.com/file/d/1HhkSW7rhNO2EWmR_rh_lLaqIloJyJxt8/view?usp=share_link)

niklasarnitz, 2022-12-13:
After having a look at your images I came up with the following components:
- Telegesis ETRX3587: Zigbee SOC
- TPS65217D: Power Management IC (Usually meant for battery powered devices?)
- SN506526OBZOZ: TI AM6526 (ARM Dual Cortex-A53 CPU)
- HM-MOC-UART-AW-SH-2: Homematic Wireless Module (Propably for the older Bosch Smarthome devices still running Homematic?)
- 6XA18 JW999: Micron JW999 NAND Flash Module (Size sadly unknown, I found no datasheet explaining the 6XA18 string)
- 6VP77 D9SHD: DDR3 RAM (1G?)

I will propably just connect up my logic analyzer to the pins around the CPU and then have a look at the signals.
Maybe I will just straight up find a UART. But that will have to wait until my SHC arrives.

@danieledwardgeorgehitchcock did you probe some of the pins already?

danieledwardgeorgehitchcock, 2022-12-13:
I probed as many pins as I felt confident doing (mostly the ones in the 3rd picture) I have a feeling that the UART may be disabled but you may be able to do a power glitch if you're lucky? I gave up after that point as I didn't want root access that badly!

niklasarnitz, 2022-12-13:
That would be frustrating if the uart was disabled.

But maybe one would be able to read the NAND flash if there were enough debug pins routed out as pads.

 

Let’s see when my unit arrives.

Reply to: Koenkk/zigbee2mqtt ***@***.***>
Date: Tuesday, 13. December 2022 at 10:58
To: Koenkk/zigbee2mqtt ***@***.***>
Cc: Niklas Arnitz ***@***.***>, Mention ***@***.***>

Nezz, 2022-12-13:
> I guess it works... but the value is not published automatically by the device...

Awesome, thanks. Is it a separate endpoint? I guess we could read that endpoint whenever we receive some data in `onEvent`, such as the temperature changing.

BBJake, 2022-12-13:
I also managed to get the device to report changes... I just had to deal with "configureReporting" and understand the principle... the documentation isn't really helpful there.

BBJake, 2022-12-14:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5150

Nezz, 2022-12-14:
My thermostat seems to be stuck in an infinite loop between 18 and 24 degrees. Any idea what might cause this?
[image]

BBJake, 2022-12-15:
Funnily I had the same problem yesterday.
For me the thermostat toggled between 20 and 21 degrees. I have no idea what caused this. It also happened without a connection to z2m and the only way to stop it was to completely reset the thermostat.
Removing the batteries or changing settings had not effect.

Nezz, 2022-12-15:
What a coincidence :D
By completely reset do you mean holding the button when reinsterting the batteries until "RES" appears and pressing the button again to confirm?

mmattel, 2022-12-15:
> hopefully there will be way to use OTA over z2m in the future...

desperately needed...

Nezz, 2022-12-15:
Too bad Linus has no radiators, so he won't be calling out Bosch anytime soon.

BBJake, 2022-12-15:
Does anybody see this effect with the Bosch thermostat? I reported that here: #13478

_I have one thermostat which is directly connected to the nearby coordinator (Sonoff Dongle-P).
After some times this device does not accept any commands sent to it. It still reports the actual temperature or changes that I do directly on the thermostat.
Other Bosch thermostats which are connected over a router are not affected. Also other thermostats from a different brand (Saswell) which are connected to the coordinator directly are not affected.
Most of the time I do not get error messages, so I guess the message is stuck somewhere in z2m._

Nezz, 2022-12-15:
Yes, I've had plenty of issues with the thermostat being unresponsive. I set a temperature on the dashboard, then it just jumps back to the previous temperature. What I learned from that is that I should just be patient and wait for the command to propagate.

itkama, 2022-12-28:
@niklasarnitz any luck on hacking the SHC yet? Don't really want to get my hopes up - but a late christmas miracle would be cool 🤣

niklasarnitz, 2022-12-28:
I haven't received mine yet.
But when I do, I'll do my best.

Nezz, 2023-01-03:
I had another case where the thermostat went crazy overnight, switching between `"occupiedHeatingSetpoint":500` and `"occupiedHeatingSetpoint":1700`. I have another thermostat that I use as a reference with the Bosch Smart Home Controller and this has not happened there so far.

It started with failing to set the heating to 17:
```
debug 2023-01-03 00:00:00: Publishing 'set' 'occupied_heating_setpoint' to 'Bedroom Thermostat'
debug 2023-01-03 00:26:36: Received Zigbee message from 'Bedroom Thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2320}' from endpoint 1 with groupID 0
error 2023-01-03 00:27:06: Publish 'set' 'occupied_heating_setpoint' to 'Bedroom Thermostat' failed: 'Error: Write 0x18fc26000006aefd/1 hvacThermostat({"occupiedHeatingSetpoint":1700}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
```

Then a couple hours later setting the temperature twice:
```
debug 2023-01-03 02:20:17: Publishing 'set' 'occupied_heating_setpoint' to 'Bedroom Thermostat'
debug 2023-01-03 02:21:04: Received MQTT message on 'zigbee2mqtt/Bedroom Thermostat/set' with data '{"occupied_heating_setpoint":17}'
debug 2023-01-03 02:21:04: Publishing 'set' 'occupied_heating_setpoint' to 'Bedroom Thermostat'
```

Logs with MQTT publish removed: https://pastebin.com/NDPsA3ny

compujuckel, 2023-01-06:
I've extracted the OTA file from a Zigbee packet capture and successfully used it to update one of my thermostats via z2m:
https://github.com/Koenkk/zigbee-OTA/pull/231
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5283

niklasarnitz, 2023-01-06:
Are you able to capture the network traffic of checking for update?Maybe this way we could get access to future updates.

Koenkk/zigbee-OTA#231
Koenkk/zigbee-herdsman-converters#5283

compujuckel, 2023-01-06:
I haven't done any captures from SHC <-> Internet since everything is probably encrypted anyways.
Sniffing the Zigbee traffic is easy, it just requires a bit of patience :D

Also since I only borrowed the SHC from a relative I don't want to play around with it too much...

[22 more replies]