Koenkk/zigbee2mqtt #14931: Reporting with grouping not working properly

BDV77, 2022-11-11:
### What happened?

I have several different configurations of devices in groups which are bound to remotes, some of them are updating their state in Home Assistant after being manipulated by the remote and some don't.

Combinations working:
1: 
1x RWL021 Hue dimmer switch
3x LCT001 Hue white and color ambiance E26/E27/E14
2: 
2x TRADFRI remote control E1524/E1810
7x LED1537R6/LED1739R5 TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum

Combinations NOT working:
1: 
2x [E1743] TRADFRI ON/OFF switch
2x [RB 185 C] E27 bulb RGBW INNR
2:
1x 
1x [E2001/E2002] STYRBAR remote control
1x [HG07834C] Livarno Lux E27 bulb RGB

### What did you expect to happen?

For all groups/combinations that the updates are visible in Home Assistant.

### How to reproduce it (minimal and precise)

Pair lamps
Add lamps to group
Add group to Home Assistant UI
Pair remote
Bind remote to group

Switch on/off lamps with remote
Check status in Home Assistant

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

DutchDynamics, 2022-11-25:
Experiencing the same problem with the status of groups.

My config:
- Create group with lights from Ikea (LED1903C5/LED1835C6) and Ikea LED1836G9
- Added Ikea Dimmer SYMFONISK sound controller (E1744)
- Created a binding between the Dimmer and the Group, only bind LevelCtrl

After this you can control the lights with the dimmer, what works quick and smooth.

No messages are created for the status update of the Group
So controlling the lights directly will no update the status in Home Assistant

BDV77, 2022-12-12:
Hi, this is still an issue. Do you have any solutions/work arounds?

Do you require more details, or logging?

Please let me know.

Thanks in advance

closed 2023-01-20 by github-actions[bot]