Koenkk/zigbee2mqtt #14932: Ubisys C4 "configure_device_setup" cannot be read or set

K-r-i-s-t-i-a-n, 2022-11-11:
### What happened?

I have several Ubisys devices (S1, S2, J1, C4), they normally work great.

Today I realized, that I cannot read or set the "configure_device_setup" cluster for two of them (both C4):

I publish to "zigbee2mqtt/FRIENDLY_NAME/get/configure_device_setup" and get the error:
info  2022-11-11 17:36:59: MQTT publish: topic 'zigbee2mqtt/EG/WOHNZIMMER/Terassentuer_Schalter_Ubisys_C4', payload '{"last_seen":"2022-11-11T17:36:59+01:00","linkquality":80,"update":{"state":"idle"},"update_available":false}'
error 2022-11-11 17:36:59: Publish 'get' 'configure_device_setup' to 'EG/WOHNZIMMER/Terassentuer_Schalter_Ubisys_C4' failed: 'Error: Read 0x001fee0000008373/232 manuSpecificUbisysDeviceSetup(["inputConfigurations","inputActions"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'FAILURE')'

The debug message reads:
"debug 2022-11-11 17:29:59: Received MQTT message on 'zigbee2mqtt/EG/WOHNZIMMER/Terassentuer_Schalter_Ubisys_C4/get/configure_device_setup' with data ''
debug 2022-11-11 17:29:59: Publishing get 'get' 'configure_device_setup' to 'EG/WOHNZIMMER/Terassentuer_Schalter_Ubisys_C4'
debug 2022-11-11 17:29:59: Received Zigbee message from 'EG/WOHNZIMMER/Terassentuer_Schalter_Ubisys_C4', type 'readResponse', cluster 'manuSpecificUbisysDeviceSetup', data '{"inputConfigurations":[0,0,0,0]}' from endpoint 232 with groupID 0
error 2022-11-11 17:29:59: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'map')}
debug 2022-11-11 17:29:59: TypeError: Cannot read properties of undefined (reading 'map')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/devices/ubisys.js:87:72)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at ZStackAdapter.emit (node:events:513:28)
info  2022-11-11 17:29:59: MQTT publish: topic 'zigbee2mqtt/EG/WOHNZIMMER/Terassentuer_Schalter_Ubisys_C4', payload '{"last_seen":"2022-11-11T17:29:59+01:00","linkquality":153,"update":{"state":"idle"},"update_available":false}'
error 2022-11-11 17:29:59: Publish 'get' 'configure_device_setup' to 'EG/WOHNZIMMER/Terassentuer_Schalter_Ubisys_C4' failed: 'Error: Read 0x001fee0000008373/232 manuSpecificUbisysDeviceSetup(["inputConfigurations","inputActions"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'FAILURE')'
debug 2022-11-11 17:29:59: Error: Read 0x001fee0000008373/232 manuSpecificUbisysDeviceSetup(["inputConfigurations","inputActions"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'FAILURE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
    at Object.convertGet (/app/node_modules/zigbee-herdsman-converters/devices/ubisys.js:512:17)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:273:21)
"

### What did you expect to happen?

It works on another C4 device:
info  2022-11-11 17:34:08: MQTT publish: topic 'zigbee2mqtt/EG/WOHNZIMMER/Schalter_Ubisys', payload '{"configure_device_setup":{"input_actions":[[0,7,1,6,0,2],[0,130,1,5,0,5,6,0,1],[0,194,1,5,0,5,6,0,2],[1,7,2,6,0,2],[1,2,2,6,0,0],[2,7,3,6,0,2],[3,13,4,6,0,2]]},"elapsed":751374,"last_seen":"2022-11-11T17:34:08+01:00","linkquality":29,"update":{"state":"idle"},"update_available":false}'

Debug message:
debug 2022-11-11 17:34:25: Received MQTT message on 'zigbee2mqtt/EG/WOHNZIMMER/Schalter_Ubisys/get/configure_device_setup' with data ''
debug 2022-11-11 17:34:25: Publishing get 'get' 'configure_device_setup' to 'EG/WOHNZIMMER/Schalter_Ubisys'
debug 2022-11-11 17:34:25: Received Zigbee message from 'EG/WOHNZIMMER/Schalter_Ubisys', type 'readResponse', cluster 'manuSpecificUbisysDeviceSetup', data '{"inputActions":[{"data":[0,7,1,6,0,2],"type":"Buffer"},{"data":[0,130,1,5,0,5,6,0,1],"type":"Buffer"},{"data":[0,194,1,5,0,5,6,0,2],"type":"Buffer"},{"data":[1,7,2,6,0,2],"type":"Buffer"},{"data":[1,2,2,6,0,0],"type":"Buffer"},{"data":[2,7,3,6,0,2],"type":"Buffer"},{"data":[3,13,4,6,0,2],"type":"Buffer"}],"inputConfigurations":[0,0,0,0]}' from endpoint 232 with groupID 0
info  2022-11-11 17:34:25: MQTT publish: topic 'zigbee2mqtt/EG/WOHNZIMMER/Schalter_Ubisys', payload '{"configure_device_setup":{"input_actions":[[0,7,1,6,0,2],[0,130,1,5,0,5,6,0,1],[0,194,1,5,0,5,6,0,2],[1,7,2,6,0,2],[1,2,2,6,0,0],[2,7,3,6,0,2],[3,13,4,6,0,2]]},"elapsed":17492,"last_seen":"2022-11-11T17:34:25+01:00","linkquality":29,"update":{"state":"idle"},"update_available":false}'
debug 2022-11-11 17:34:27: Received Zigbee message from 'DG/BAD/Relais_Iluminize', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":47,"imageType":2077,"manufacturerCode":4107}' from endpoint 1 with groupID 0

### Zigbee2MQTT version

1.28.2 commit: 360a777

### Adapter firmware version

20221102

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

sjorge, 2022-11-12:
@Koenkk looks like the readReponse does not include inputActions on the failed read.
But I'm confused as to why it's generating an exception on the map() as it should not get there unless there is an inputAction property in msg.data. As we check for this specifically:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/1f1244cd9d43b78c28be854169f0c1b4c6b8ae1a/devices/ubisys.js#L86-L90

I also can't replicate it with my C4 and my S1/S2's. They always return both attributes.

Koenkk, 2022-11-13:
@sjorge stringify which is used to about the payload does seem to skip `undefined` values: (`bla` is missing)

[image]

I think something goes wrong in the packet parsing, @K-r-i-s-t-i-a-n can you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

@sjorge I think there is also an error here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/c54ebf0dae823545b3e228a081582193b0ac677d/devices/ubisys.js#L83 , `if (msg.data.hasOwnProperty('input_configurations')) {` should be `if (msg.data.hasOwnProperty('inputConfigurations')) {`

sjorge, 2022-11-13:
> @sjorge I think there is also an error here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/c54ebf0dae823545b3e228a081582193b0ac677d/devices/ubisys.js#L83 , `if (msg.data.hasOwnProperty('input_configurations')) {` should be `if (msg.data.hasOwnProperty('inputConfigurations')) {`

Indeed, that should be inputConfigurations. I'll do a PR for this.

K-r-i-s-t-i-a-n, 2022-11-13:
No sure if I got all the relevant lines of the log, please let me know if you need more:
2022-11-13T16:04:12.197Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,11,69,200,85,93,141,174,98,254,255,114,2,92,2,224]
2022-11-13T16:04:12.197Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,11,69,200,85,93,141,174,98,254,255,114,2,92,2,224]
2022-11-13T16:04:12.197Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 11 - 2 - 5 - 200 - [85,93,141,174,98,254,255,114,2,92,2] - 224
2022-11-13T16:04:12.197Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - concentratorIndCb - {"srcaddr":23893,"extaddr":"0x5c0272fffe62ae8d","pktCost":2}
2022-11-13T16:04:12.197Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-11-13T16:04:13.138Z zigbee-herdsman:controller:endpoint Read 0x001fee0000008373/232 manuSpecificUbisysDeviceSetup(["inputConfigurations","inputActions"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-11-13T16:04:13.138Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001fee0000008373:42606/232 (0,0,1)
2022-11-13T16:04:13.139Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":42606,"destendpoint":232,"srcendpoint":1,"clusterid":64512,"transid":80,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[16,65,0,0,0,1,0]}}
2022-11-13T16:04:13.139Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,110,166,232,1,0,252,80,0,30,7,16,65,0,0,0,1,0,240]
2022-11-13T16:04:13.151Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-11-13T16:04:13.151Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-11-13T16:04:13.151Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-11-13T16:04:13.151Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-11-13T16:04:13.151Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-11-13T16:04:13.155Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,80,150]
2022-11-13T16:04:13.155Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,80,150]
2022-11-13T16:04:13.155Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,80] - 150
2022-11-13T16:04:13.155Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":80}
2022-11-13T16:04:13.155Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-11-13T16:04:13.203Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,37,68,129,0,0,0,252,110,166,232,1,0,36,0,144,135,162,0,0,17,24,65,1,0,0,0,72,8,4,0,0,0,0,0,1,0,1,188,182,28,183]
2022-11-13T16:04:13.203Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,37,68,129,0,0,0,252,110,166,232,1,0,36,0,144,135,162,0,0,17,24,65,1,0,0,0,72,8,4,0,0,0,0,0,1,0,1,188,182,28,183]
2022-11-13T16:04:13.203Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 37 - 2 - 4 - 129 - [0,0,0,252,110,166,232,1,0,36,0,144,135,162,0,0,17,24,65,1,0,0,0,72,8,4,0,0,0,0,0,1,0,1,188,182,28] - 183
2022-11-13T16:04:13.203Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64512,"srcaddr":42606,"srcendpoint":232,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":10651536,"transseqnumber":0,"len":17,"data":{"type":"Buffer","data":[24,65,1,0,0,0,72,8,4,0,0,0,0,0,1,0,1]}}
2022-11-13T16:04:13.205Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":65,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":0,"status":0,"dataType":72,"attrData":[0,0,0,0]},{"attrId":1,"status":1}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":42606,"endpoint":232,"linkquality":36,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:error 2022-11-13 17:04:13: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'map')}
Zigbee2MQTT:info  2022-11-13 17:04:13: MQTT publish: topic 'zigbee2mqtt/EG/WOHNZIMMER/Terassentuer_Schalter_Ubisys_C4', payload '{"last_seen":"2022-11-13T17:04:13+01:00","linkquality":36,"update":{"state":"idle"},"update_available":false}'
2022-11-13T16:04:13.210Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-11-13T16:04:13.211Z zigbee-herdsman:controller:endpoint Read 0x001fee0000008373/232 manuSpecificUbisysDeviceSetup(["inputConfigurations","inputActions"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'FAILURE')
Zigbee2MQTT:error 2022-11-13 17:04:13: Publish 'get' 'configure_device_setup' to 'EG/WOHNZIMMER/Terassentuer_Schalter_Ubisys_C4' failed: 'Error: Read 0x001fee0000008373/232 manuSpecificUbisysDeviceSetup(["inputConfigurations","inputActions"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'FAILURE')'
2022-11-13T16:04:13.344Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,21,252,1,42,26,92]

Koenkk, 2022-11-14:
`{"attrId":0,"status":0,"dataType":72,"attrData":[0,0,0,0]},{"attrId":1,"status":1}` 

The status of inputActions is not 0 (indicating failed) so no value will be present in the payload. I've fixed the converter.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

K-r-i-s-t-i-a-n, 2022-11-16:
Rejoining the devices solved the issue at first. Writing back my old configuration brought the issue back. Seems like the size limit on the input-configuration table was reduced in one of the last OTA updates. Reduced my configuration table, now it seems to work again (although I now have to rethink my configuration).

Thanks for the support!

closed 2022-11-16 by K-r-i-s-t-i-a-n