Koenkk/zigbee2mqtt #14933: Error while starting zigbee-herdsman - Sonoff_Zigbee_3.0_USB_Dongle_Plus

Rustymage, 2022-11-11:
### What happened?

Z2M fails to start.

https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html followed.
```

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.141:1883
  username: mqtt
  password: password
serial:
  port: /dev/ttyACM0
  adapter: deconz
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: true
advanced:
  log_level: info
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  new_api: true
```

```
[core-ssh ~]$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx    1 root     root            13 Nov 11 17:39 usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0 -> ../../ttyUSB0
lrwxrwxrwx    1 root     root            13 Nov 11 17:39 usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2459817-if00 -> ../../ttyACM0
```

### What did you expect to happen?

Z2M to start.

### How to reproduce it (minimal and precise)

Utilise Sonoff stick.

```
Home Assistant 2022.10.5 
Supervisor 2022.10.2 
Operating System 9.3 
Frontend 20221010.0 - latest 
```

Proxmox install of HAOS, USB pass through. Conbee works fine. Migrating to newer Sonoff.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

Unknown

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

```
[17:43:47] INFO: Preparing to start...
[17:43:47] INFO: Socat not enabled
[17:43:47] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2022-11-11 17:43:49: Logging to console and directory: '/config/zigbee2mqtt/log/2022-11-11.17-43-49' filename: log.txt
Zigbee2MQTT:info  2022-11-11 17:43:49: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
Zigbee2MQTT:info  2022-11-11 17:43:49: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:error 2022-11-11 17:43:59: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-11-11 17:43:59: Failed to start zigbee
Zigbee2MQTT:error 2022-11-11 17:43:59: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-11-11 17:43:59: Exiting...
Zigbee2MQTT:error 2022-11-11 17:43:59: undefined
```

Rustymage, 2022-11-11:
I have updated Sonoff_Zigbee_3.0_USB_Dongle_Plus to `CC1352P2_CC2652P_launchpad_coordinator_20220219.hex`

HAOS is now:

```
Home Assistant 2022.11.2 
Supervisor 2022.10.2 
Operating System 9.3 
Frontend 20221108.0 - latest 
```

& Z2M is 1.28.2-1.

Still not working :(

Rustymage, 2022-11-12:
Got it working. Will write up how shortly.

wilwizard1979, 2022-11-12:
Hi Rusty, I'm having exactly the same problem!
Please see my post in [https://github.com/Koenkk/zigbee2mqtt/issues/14890](url)

Rustymage, 2022-11-12:
So I found this issue https://github.com/Koenkk/zigbee2mqtt/issues/4151

I then followed part 2 & 3 of this - the key part being removing the old database. I don't recall ever seeing this as part of an adapter move anywhere else - https://github.com/Koenkk/zigbee2mqtt/issues/4151#issuecomment-684003951

To repair my hue lights, I followed this - https://github.com/Koenkk/zigbee2mqtt/issues/454

Took about 1.5 to 2 hours to do 50 devices.

closed 2022-11-12 by Rustymage