Koenkk/zigbee2mqtt #14934: Tuya TS011F_plug_3 no value for energy

jarrah31, 2022-11-11:
### What happened?

I recently received four new Zigbee Smart Energy Plugs to complement my existing 25 of the same model and type, but these new ones appears to be slightly different (in terms of Zigbee Manufacturer) to my other TS011F_plug_3 plugs.

The problem is the value for energy is reporting as Null kWh. 

[image]

When I try to monitor this in Home Assistant using utility_meter, it constantly reports the following error:
```
WARNING (MainThread) [homeassistant.components.utility_meter.sensor] Invalid state (None > None): [<class 'decimal.ConversionSyntax'>]
```
HA yaml for this is:
```
utility_meter:
  plug_zigbee26_energy_um_daily:
    source: sensor.plug_zigbee26_energy
    cycle: daily
```
Plug details are:

[image]

### What did you expect to happen?

Provide an energy reading like this:

[image]

Plug details for working one is:
[image]

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20210120

### Adapter

CC1352P-2

### Debug log

```
Nov 11 19:25:46 zigbeepi3 npm[27399]: Zigbee2MQTT:debug 2022-11-11 19:25:46: Received Zigbee message from 'plug_zigbee26', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":268,"rmsCurrent":1202,"rmsVoltage":242}' from endpoint 1 with groupID 0
Nov 11 19:25:46 zigbeepi3 npm[27399]: Zigbee2MQTT:info  2022-11-11 19:25:46: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":1.2,"energy":null,"indicator_mode":null,"last_seen":"2022-11-11T19:25:46+00:00","linkquality":42,"power":268,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
Nov 11 19:26:46 zigbeepi3 npm[27399]: Zigbee2MQTT:debug 2022-11-11 19:26:46: Received Zigbee message from 'plug_zigbee26', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":35,"rmsCurrent":161,"rmsVoltage":242}' from endpoint 1 with groupID 0
Nov 11 19:26:46 zigbeepi3 npm[27399]: Zigbee2MQTT:info  2022-11-11 19:26:46: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":0.16,"energy":null,"indicator_mode":null,"last_seen":"2022-11-11T19:26:46+00:00","linkquality":51,"power":35,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
Nov 11 19:27:38 zigbeepi3 npm[27399]: Zigbee2MQTT:debug 2022-11-11 19:27:38: Received Zigbee message from 'plug_zigbee26', type 'attributeReport', cluster 'genBasic', data '{"appVersion":160}' from endpoint 1 with groupID 0
Nov 11 19:27:38 zigbeepi3 npm[27399]: Zigbee2MQTT:info  2022-11-11 19:27:38: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":0.16,"energy":null,"indicator_mode":null,"last_seen":"2022-11-11T19:27:38+00:00","linkquality":63,"power":35,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
Nov 11 19:27:46 zigbeepi3 npm[27399]: Zigbee2MQTT:debug 2022-11-11 19:27:46: Received Zigbee message from 'plug_zigbee26', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":35,"rmsCurrent":155,"rmsVoltage":243}' from endpoint 1 with groupID 0
Nov 11 19:27:46 zigbeepi3 npm[27399]: Zigbee2MQTT:info  2022-11-11 19:27:46: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":0.16,"energy":null,"indicator_mode":null,"last_seen":"2022-11-11T19:27:46+00:00","linkquality":63,"power":35,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":243}'
```

Koenkk, 2022-11-12:
Can you provide the data/database.db entry of this device?

jarrah31, 2022-11-12:
Thanks for looking at this - here you go:

```json
{
    "id": 48,
    "type": "Router",
    "ieeeAddr": "0x70b3d52b6000f25c",
    "nwkAddr": 37431,
    "manufId": 4660,
    "manufName": "_TZ3000_okaz9tjs",
    "powerSource": "Mains (single phase)",
    "modelId": "TS011F",
    "epList": [1],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 81,
            "inClusterList": [0, 6, 3, 4, 5, 57345, 2820, 1794],
            "outClusterList": [],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "TS011F",
                        "manufacturerName": "_TZ3000_okaz9tjs",
                        "powerSource": 1,
                        "zclVersion": 2,
                        "appVersion": 160,
                        "stackVersion": 1,
                        "hwVersion": 3,
                        "dateCode": "20210625",
                        "swBuildId": "20B+TZSKT11BS105"
                    }
                },
                "haElectricalMeasurement": {
                    "attributes": {
                        "acCurrentDivisor": 1000,
                        "acCurrentMultiplier": 1,
                        "rmsVoltage": 241,
                        "rmsCurrent": 1215,
                        "activePower": 271
                    }
                },
                "seMetering": {
                    "attributes": {
                        "divisor": 100,
                        "multiplier": 1
                    }
                },
                "genOnOff": {
                    "attributes": {
                        "32768": 0,
                        "onOff": 1
                    }
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 160,
    "stackVersion": 1,
    "hwVersion": 3,
    "dateCode": "20210625",
    "swBuildId": "20B+TZSKT11BS105",
    "zclVersion": 2,
    "interviewCompleted": true,
    "meta": {
        "configured": -522437996
    },
    "lastSeen": 1668259884953,
    "defaultSendRequestWhen": "immediate"
}
```

Koenkk, 2022-11-13:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/3e84ac39d694364e78e7c04a439ad9aa

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

saspol, 2022-11-13:
Hi, i have same problem with same _TZ3000_okaz9tjs device.
Tried to add this converter and situation not changed, collected energy still not reporting.

jarrah31, 2022-11-13:
For me the external converter resulted in Z2M failing to load, giving this error on repeat:

```
Nov 13 18:20:03 zigbeepi3 systemd[1]: Started zigbee2mqtt.
Nov 13 18:20:06 zigbeepi3 npm[14288]: > zigbee2mqtt@1.28.2 start
Nov 13 18:20:06 zigbeepi3 npm[14288]: > node index.js
Nov 13 18:20:12 zigbeepi3 npm[14288]: /opt/zigbee2mqtt/dist/util/externally-loaded.js:21
Nov 13 18:20:12 zigbeepi3 npm[14288]:     fromZigbee: [fz.on_off, fzLocal.electrical_measurement_skip_duplicate, fzLocal.metering_skip_duplicate, fz.ignore_basic_report,
Nov 13 18:20:12 zigbeepi3 npm[14288]:                             ^
Nov 13 18:20:12 zigbeepi3 npm[14288]: ReferenceError: fzLocal is not defined
Nov 13 18:20:12 zigbeepi3 npm[14288]:     at /opt/zigbee2mqtt/dist/util/externally-loaded.js:21:29
Nov 13 18:20:12 zigbeepi3 npm[14288]:     at Script.runInContext (node:vm:139:12)
Nov 13 18:20:12 zigbeepi3 npm[14288]:     at Script.runInNewContext (node:vm:144:17)
Nov 13 18:20:12 zigbeepi3 npm[14288]:     at Object.runInNewContext (node:vm:298:38)
Nov 13 18:20:12 zigbeepi3 npm[14288]:     at loadModuleFromText (/opt/zigbee2mqtt/lib/util/utils.ts:148:8)
Nov 13 18:20:12 zigbeepi3 npm[14288]:     at loadModuleFromFile (/opt/zigbee2mqtt/lib/util/utils.ts:155:12)
Nov 13 18:20:12 zigbeepi3 npm[14288]:     at Object.getExternalConvertersDefinitions (/opt/zigbee2mqtt/lib/util/utils.ts:165:25)
Nov 13 18:20:12 zigbeepi3 npm[14288]:     at getExternalConvertersDefinitions.next (<anonymous>)
Nov 13 18:20:12 zigbeepi3 npm[14288]:     at new ExternalConverters (/opt/zigbee2mqtt/lib/extension/externalConverters.ts:12:20)
Nov 13 18:20:12 zigbeepi3 npm[14288]:     at new Controller (/opt/zigbee2mqtt/lib/controller.ts:84:58)
Nov 13 18:20:12 zigbeepi3 systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Nov 13 18:20:12 zigbeepi3 systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Nov 13 18:20:13 zigbeepi3 systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Nov 13 18:20:13 zigbeepi3 systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 7.
Nov 13 18:20:13 zigbeepi3 systemd[1]: Stopped zigbee2mqtt.
```

saspol, 2022-11-14:
Sorry, still not working

```
2022-11-15 01:22:32 MQTT publish: topic 'zigbee2mqtt/0x70b3d52b60011901', payload '{"child_lock":"UNLOCK","current":0.02,"last_seen":1668446552358,"linkquality":60,"power":0,"state":"ON","voltage":247}'

2022-11-15 01:22:32Failed to configure '0x70b3d52b60011901', attempt 2 (Error: Read 0x70b3d52b60011901/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22) at Object.configureMagicPacket (/app/node_modules/zigbee-herdsman-converters/lib/tuya.js:1159:5) at Object.configure (/app/dist/util/externally-loaded.js:25:9) at Configure.configure (/app/lib/extension/configure.ts:115:13))
```

jarrah31, 2022-11-14:
Thank you for the update @koenkk 

I too receive errors when using the updated external converter:

```
Nov 14 19:30:39 zigbeepi3 npm[24897]: Zigbee2MQTT:debug 2022-11-14 19:30:39: Received Zigbee message from 'plug_zigbee26', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":12,"rmsVoltage":243}' from endpoint 1 with groupID 0
Nov 14 19:30:39 zigbeepi3 npm[24897]: Zigbee2MQTT:info  2022-11-14 19:30:39: Configuring 'plug_zigbee26'
Nov 14 19:30:39 zigbeepi3 npm[24897]: Zigbee2MQTT:info  2022-11-14 19:30:39: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":0.01,"energy":null,"indicator_mode":null,"last_seen":"2022-11-14T19:30:39+00:00","linkquality":39,"power":0,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":243}'
Nov 14 19:30:39 zigbeepi3 npm[24897]: Zigbee2MQTT:debug 2022-11-14 19:30:39: Received Zigbee message from 'plug_zigbee26', type 'readResponse', cluster 'genBasic', data '{"appVersion":160,"manufacturerName":"_TZ3000_okaz9tjs","modelId":"TS011F","powerSource":1,"zclVersion":2}' from endpoint 1 with groupID 0
Nov 14 19:30:40 zigbeepi3 npm[24897]: Zigbee2MQTT:info  2022-11-14 19:30:40: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":0.01,"energy":null,"indicator_mode":null,"last_seen":"2022-11-14T19:30:39+00:00","linkquality":39,"power":0,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":243}'
Nov 14 19:30:40 zigbeepi3 npm[24897]: Zigbee2MQTT:error 2022-11-14 19:30:40: Failed to configure 'plug_zigbee26', attempt 2 (Error: Read 0x70b3d52b6000f25c/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
Nov 14 19:31:39 zigbeepi3 npm[24897]: Zigbee2MQTT:debug 2022-11-14 19:31:39: Received Zigbee message from 'plug_zigbee26', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":12,"rmsVoltage":244}' from endpoint 1 with groupID 0
Nov 14 19:31:39 zigbeepi3 npm[24897]: Zigbee2MQTT:info  2022-11-14 19:31:39: Configuring 'plug_zigbee26'
Nov 14 19:31:39 zigbeepi3 npm[24897]: Zigbee2MQTT:info  2022-11-14 19:31:39: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":0.01,"energy":null,"indicator_mode":null,"last_seen":"2022-11-14T19:31:39+00:00","linkquality":39,"power":0,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":244}'
Nov 14 19:31:40 zigbeepi3 npm[24897]: Zigbee2MQTT:debug 2022-11-14 19:31:40: Received Zigbee message from 'plug_zigbee26', type 'readResponse', cluster 'genBasic', data '{"appVersion":160,"manufacturerName":"_TZ3000_okaz9tjs","modelId":"TS011F","powerSource":1,"zclVersion":2}' from endpoint 1 with groupID 0
Nov 14 19:31:40 zigbeepi3 npm[24897]: Zigbee2MQTT:info  2022-11-14 19:31:40: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":0.01,"energy":null,"indicator_mode":null,"last_seen":"2022-11-14T19:31:40+00:00","linkquality":42,"power":0,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":244}'
Nov 14 19:31:40 zigbeepi3 npm[24897]: Zigbee2MQTT:error 2022-11-14 19:31:40: Failed to configure 'plug_zigbee26', attempt 3 (Error: Read 0x70b3d52b6000f25c/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
Nov 14 19:32:08 zigbeepi3 npm[24897]: Zigbee2MQTT:debug 2022-11-14 19:32:08: Received Zigbee message from 'plug_zigbee26', type 'attributeReport', cluster 'genBasic', data '{"appVersion":160}' from endpoint 1 with groupID 0
Nov 14 19:32:08 zigbeepi3 npm[24897]: Zigbee2MQTT:info  2022-11-14 19:32:08: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":0.01,"energy":null,"indicator_mode":null,"last_seen":"2022-11-14T19:32:08+00:00","linkquality":48,"power":0,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":244}'
```

Koenkk, 2022-11-14:
Updated https://gist.github.com/Koenkk/3e84ac39d694364e78e7c04a439ad9aa

Koenkk, 2022-11-15:
Updated https://gist.github.com/Koenkk/3e84ac39d694364e78e7c04a439ad9aa, can you provide the logging again? Not from the log file but from the STDOUT this time.

jarrah31, 2022-11-15:
Yep, happy to try and do whatever will help.
Here is the console output after running `npm start`. I've left some other zigbee plugs in there for reference and removed other devices not of interest.
plug_zigbee27 is the same type as 26.

Just noticed that energy is now being reported! 🥳 

```
Zigbee2MQTT:debug 2022-11-15 19:05:39: Received Zigbee message from 'plug_zigbee27', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":12,"rmsCurrent":96,"rmsVoltage":241}' from endpoint 1 with groupID 0
START true true
Zigbee2MQTT:info  2022-11-15 19:05:39: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee27', payload '{"child_lock":null,"current":0.1,"energy":null,"indicator_mode":null,"last_seen":"2022-11-15T19:05:39+00:00","linkquality":81,"power":12,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
READ METERING? true
READ METERING
Zigbee2MQTT:debug 2022-11-15 19:05:39: Received Zigbee message from 'plug_zigbee26', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":224,"rmsCurrent":1066,"rmsVoltage":241}' from endpoint 1 with groupID 0
START true true
Zigbee2MQTT:info  2022-11-15 19:05:39: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":1.07,"energy":null,"indicator_mode":null,"last_seen":"2022-11-15T19:05:39+00:00","linkquality":63,"power":224,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
READ METERING? true
READ METERING
Zigbee2MQTT:debug 2022-11-15 19:05:39: Received Zigbee message from 'plug_zigbee27', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[1,6]}' from endpoint 1 with groupID 0
START true true
Zigbee2MQTT:info  2022-11-15 19:05:39: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee27', payload '{"child_lock":null,"current":0.1,"energy":0.07,"indicator_mode":null,"last_seen":"2022-11-15T19:05:39+00:00","linkquality":81,"power":12,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
Zigbee2MQTT:debug 2022-11-15 19:05:40: Received Zigbee message from 'plug_zigbee26', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[1,418]}' from endpoint 1 with groupID 0
START true true
Zigbee2MQTT:info  2022-11-15 19:05:40: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":1.07,"energy":4.19,"indicator_mode":null,"last_seen":"2022-11-15T19:05:40+00:00","linkquality":45,"power":224,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
Zigbee2MQTT:debug 2022-11-15 19:05:53: Received Zigbee message from 'plug_zigbee24', type 'attributeReport', cluster 'genBasic', data '{"appVersion":160}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-11-15 19:05:53: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee24', payload '{"child_lock":"UNLOCK","current":0.05,"energy":null,"indicator_mode":null,"last_seen":"2022-11-15T19:05:53+00:00","linkquality":72,"power":3,"power_outage_memory":"off","state":"ON","update":{"state":null},"update_available":null,"voltage":240}'
Zigbee2MQTT:debug 2022-11-15 19:05:58: Received Zigbee message from 'plug_zigbee19', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":67}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-11-15 19:05:59: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee19', payload '{"child_lock":"UNLOCK","current":0.01,"energy":7.62,"indicator_mode":"off/on","last_seen":"2022-11-15T19:05:58+00:00","linkquality":42,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
Zigbee2MQTT:debug 2022-11-15 19:05:59: Received Zigbee message from 'plug_zigbee19', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":67}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-11-15 19:05:59: Received Zigbee message from 'plug_zigbee19', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":67}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-11-15 19:05:59: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee19', payload '{"child_lock":"UNLOCK","current":0.01,"energy":7.62,"indicator_mode":"off/on","last_seen":"2022-11-15T19:05:59+00:00","linkquality":63,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
Zigbee2MQTT:info  2022-11-15 19:05:59: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee19', payload '{"child_lock":"UNLOCK","current":0.01,"energy":7.62,"indicator_mode":"off/on","last_seen":"2022-11-15T19:05:59+00:00","linkquality":63,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
Zigbee2MQTT:debug 2022-11-15 19:06:00: Received Zigbee message from 'plug_zigbee17', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":67}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-11-15 19:06:00: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee17', payload '{"child_lock":null,"current":0.47,"energy":616.25,"indicator_mode":null,"last_seen":"2022-11-15T19:06:00+00:00","linkquality":42,"power":58,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
[... cut ...]
Zigbee2MQTT:debug 2022-11-15 19:06:57: Received Zigbee message from 'plug_zigbee17', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":58}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-11-15 19:06:57: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee17', payload '{"child_lock":null,"current":0.47,"energy":616.25,"indicator_mode":null,"last_seen":"2022-11-15T19:06:57+00:00","linkquality":42,"power":58,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
```

Koenkk, 2022-11-15:
cool, can you check wether the energy reporting still updates with https://gist.github.com/Koenkk/3e84ac39d694364e78e7c04a439ad9aa ?

jarrah31, 2022-11-15:
It shows energy as still being reported, but the values aren't changing and it's giving these errors too:

```
Nov 15 19:37:43 zigbeepi3 npm[2674]: Zigbee2MQTT:debug 2022-11-15 19:37:43: Received Zigbee message from 'plug_zigbee26', type 'raw', cluster 'seMetering', data '{"data":[8,201,10,0,0,0,37,176,1,0,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Nov 15 19:37:43 zigbeepi3 npm[2674]: Zigbee2MQTT:info  2022-11-15 19:37:43: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":1.18,"energy":4.32,"indicator_mode":null,"last_seen":"2022-11-15T19:37:43+00:00","linkquality":36,"power":260,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
Nov 15 19:37:43 zigbeepi3 npm[2674]: Zigbee2MQTT:info  2022-11-15 19:37:43: Configuring 'plug_zigbee26'
Nov 15 19:37:43 zigbeepi3 npm[2674]: Zigbee2MQTT:info  2022-11-15 19:37:43: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":1.18,"energy":4.32,"indicator_mode":null,"last_seen":"2022-11-15T19:37:43+00:00","linkquality":42,"power":260,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
Nov 15 19:37:43 zigbeepi3 npm[2674]: Zigbee2MQTT:error 2022-11-15 19:37:43: Failed to configure 'plug_zigbee26', attempt 2 (Error: ConfigureReporting 0x70b3d52b6000f25c/1 seMetering([{"attribute":"currentSummDelivered","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
Nov 15 19:38:15 zigbeepi3 npm[2674]: Zigbee2MQTT:debug 2022-11-15 19:38:15: Received Zigbee message from 'plug_zigbee26', type 'attributeReport', cluster 'genBasic', data '{"appVersion":160}' from endpoint 1 with groupID 0
Nov 15 19:38:15 zigbeepi3 npm[2674]: Zigbee2MQTT:info  2022-11-15 19:38:15: Configuring 'plug_zigbee26'
Nov 15 19:38:15 zigbeepi3 npm[2674]: Zigbee2MQTT:info  2022-11-15 19:38:15: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":1.18,"energy":4.32,"indicator_mode":null,"last_seen":"2022-11-15T19:38:15+00:00","linkquality":63,"power":260,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
Nov 15 19:38:15 zigbeepi3 npm[2674]: Zigbee2MQTT:info  2022-11-15 19:38:15: MQTT publish: topic 'zigbee2mqttcc1352/plug_zigbee26', payload '{"child_lock":"UNLOCK","current":1.18,"energy":4.32,"indicator_mode":null,"last_seen":"2022-11-15T19:38:15+00:00","linkquality":66,"power":260,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":242}'
Nov 15 19:38:15 zigbeepi3 npm[2674]: Zigbee2MQTT:error 2022-11-15 19:38:15: Failed to configure 'plug_zigbee26', attempt 3 (Error: ConfigureReporting 0x70b3d52b6000f25c/1 seMetering([{"attribute":"currentSummDelivered","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
```

Koenkk, 2022-11-17:
Thanks for testing, pushed the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-17 by Koenkk

saspol, 2022-11-18:
Hello, can you please tell which patches applied? No possible update to dev branch in docker container.

ppmt, 2022-11-20:
I know the case is closed but I have the same type of plug and I can't get it to work. I have applied the external converter but I keep getting the following error. The plug is recognised as supported but there is nothing configured in bind tab or the reporting tab.

```
Zigbee2MQTT:error 2022-11-20 11:23:56: Failed to configure '0x70b3d52b60011725', attempt 1 (Error: ConfigureReporting 0x70b3d52b60011725/1 seMetering([{"attribute":"currentSummDelivered","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at Object.currentSummDelivered (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:127:9)
    at Object.configure (/app/dist/util/externally-loaded.js:54:9)
    at Configure.configure (/app/lib/extension/configure.ts:115:13)
    at Configure.start (/app/lib/extension/configure.ts:68:13)
    at Controller.callExtensions (/app/lib/controller.ts:314:17)
    at Controller.start (/app/lib/controller.ts:153:9)
    at start (/app/index.js:109:5))
```

I am using the latest zigbee addons for Home Assistant (not the edge).

Anything else I can provide to help fix my problem?

Koenkk, 2022-11-20:
@ppmt if this is a `TS011F_plug_3 `, then this error won't appear anymore in the next z2m release.

ppmt, 2022-11-21:
@Koenkk Thank for the answer. I will wait for it then :)

Is it related as well to populate the Bind and Reporting tab with data for the device?

Koenkk, 2022-11-21:
> Is it related as well to populate the Bind and Reporting tab with data for the device?

Yes, this error prevents the configure from running correctly (which adds the binding/reporting)

ppmt, 2022-11-21:
Thanks a lot for all the work you do!

grunthos503, 2022-11-22:
I just got the US version of these plugs

TS011F      _TZ3000_okaz9tjs
Firmware build date      20210625
Firmware version      20B+TZSKT11BS105

https://www.aliexpress.us/item/3256803863605307.html

Energy reporting did not work in Z2M release 1.28.1, but I am happy to say that it does work correctly in Edge release.

Only outstanding issue I have with them is that checking for updates does not work: "Failed to find endpoint which support OTA cluster"

```
info 2022-11-22 15:31:03MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Kitchen Refrigerator Plug"},"error":"Failed to check if update available for 'Kitchen Refrigerator Plug' (Failed to find endpoint which support OTA cluster)","status":"error","transaction":"wdnqm-1"}'

error 2022-11-22 15:31:03Failed to check if update available for 'Kitchen Refrigerator Plug' (Failed to find endpoint which support OTA cluster)

debug 2022-11-22 15:31:03AssertionError [ERR_ASSERTION]: Failed to find endpoint which support OTA cluster at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:248:9) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:130:19) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:192:67) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1178:20) at Receiver.emit (node:events:513:28)

```

Is this a known issue?

ppmt, 2022-12-01:
I can confirm that with the latest version of Zigbee2mqtt addons in HomeAssistant (1.28.4) the plugs seems to now be working properly. I see them being reported in the Energy tab.

And the configuration error message have gone.

Thank you !!