Koenkk/zigbee2mqtt #14935: Centralite 4200-C interview fails

TraceWilson, 2022-11-11:
### What happened?

Centralite 4200C plug interview fails with message
Failed to interview '0x5c0272fffecc8295', device has not successfully been paired

Device shows paired to the network.
Device displayed in Z2M frontend as

Device type - Router
Zigbee Model - 
Zigbee Manufacturer - 
Support status - Not supported
IEEE Address - 0x5c0272fffecc8295
Network address - 0x1067
Power - 
Interview completed - False

Found [issue](https://github.com/Koenkk/zigbee2mqtt/issues/14350) and based on that, attemtped changing zigbee channel to 24 & 25, but interview continues to fail.

### What did you expect to happen?

Interview to complete successfully

### How to reproduce it (minimal and precise)

Reset plug into pairing mode.  Interview starts, then fails.  Plug flashes multi colord lights indicating it has paired with the coordinator.
Attempted with 2 different plugs.

### Zigbee2MQTT version

1.28.2 commit: 360a777

### Adapter firmware version

20210708

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_88389f3af112ec11976821c7bd930c07-if00-port0

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9993157/log.txt)

closed 2022-12-19 by github-actions[bot]

bchoor, 2023-03-26:
It seems that the 4200-C has an updated manufacturer - from "Centralite" to "Centralite Systems". I'm not sure how to make a pull request on this, but hard coding the change in centralite.js seems to work for me. now.

How did I figure this out? 
When querying the device through DevConsole, I found the following:
Read result of 'genBasic': {"manufacturerName":"Centralite Systems"}

I know this is closed; but hoping we can have this fixed at the broader level.

bchoor, 2023-03-26:
While I'm able to control the device, it's still giving me "interview failed" ... and looking at centralite.js, it's gone back to "Centralite", so not sure my fix worked, maybe it was just a fluke.

dcarpio01, 2024-03-23:
Did you ever figure out what's going on? I can't get any of my 4200-C plugs to pair.

How do I make the change in centralite.js? I can't seem to find that file anywhere.