Koenkk/zigbee2mqtt #14938: Aqara TVOC VOC entity not updating

mihaimdinca, 2022-11-12:
### What happened?

I have an Aqara TVOC sensor connected to Home Assistant via Z2M add-on, Sonoff ZigBee 3.0 Dongle (feb 22 firmware), and a Woox ZigBee Bulb as a router. The temperature and humidity update on a regular basis while the Voc entity updates about once a day. The values displayed on the screen update regularly. I had this issue before but after updating the dongle firmware and repairing the TVOC sensor the problem was resolved. I removed the sensor and readded it several times but this did not fix the problem. Sensor firmware is 2020www

### What did you expect to happen?

Aqara TVOC sensor should report voc value more often

### How to reproduce it (minimal and precise)

Just pair the sensor and monitor the voc entity

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20220219

### Adapter

Sonoff ZigBee Dongle 3.0 zStack3x0

danieledwardgeorgehitchcock, 2022-11-12:
I created the original converter for this device.

You might want to play around with your reporting intervals / thresholds.

I aired on the side of conserving battery power on the device so, the thresholds are pretty wide.

If you don't mind eating up more battery power, you can look at reducing these values - please remember to wake the device as you are changing them so that the config can be applied.

mihaimdinca, 2022-11-12:
Pardon my noobness but how exactly do I do that? I tried to do that in the reporting tab of the device but it is unclear for me what cluster and attribute should be selected. Thank you for your patience.

danieledwardgeorgehitchcock, 2022-11-13:
If the device was successfully set up, you should have some entries in there for temperature, humidity and TVOC..?

ktpx, 2022-12-18:
Same issue here, TVOC sensors are created, but i dont see any values being reported at all.

closed 2023-01-26 by github-actions[bot]