Koenkk/zigbee2mqtt #14940: Moes BRT-100-TRV does not refresh/update in Home Assistant the temperature set in the programming mode.

catalinescu-sudo, 2022-11-12:
### What happened?

I have set the programming mode, that means between some defined hours I have X temperature (for example 21), in other time intervals, different temperatures. 

When for example the 07:00 - 12:00 interval became active, the change of the temperature (from 19 to 22) is not updated in the home assistant and also is not updated in the "current_heating_setpoint".
In the thermostat the temperature is updated and the programming mode works. The only way that I could send the new temperature from the programming mode is to physically touch the thermostat on + or - to display the set temperature - in the moment thatdisplay is turned on and shows the time interval temperature, the home assistant and current_heating_setpoint is getting the correct value.

`{
    "battery": 91,
    "boost_heating": "OFF",
    "boost_heating_countdown": 0,
    "boost_heating_countdown_time_set": 300,
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 22,
    "eco_mode": "OFF",
    "eco_temperature": 19,
    "linkquality": 72,
    "local_temperature": 21,
    "local_temperature_calibration": -2,
    "max_temperature": 30,
    "min_temperature": 5,
    "position": 25,
    "preset": "programming",
    "programming_mode": "07:00/22°C  12:00/21°C  17:00/21°C  23:00/19°C  07:00/20°C  12:00/21°C  17:00/20°C  23:00/19°C  07:00/20°C  12:00/21°C  17:00/20°C  23:00/19°C",
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "valve_state": "OPEN",
    "window": "CLOSED",
    "window_detection": "ON"
}`

### What did you expect to happen?

When the new interval became active, current_heating_setpoint  and home assistant should get the new value. 

### How to reproduce it (minimal and precise)

Set programming mode in the thermostat, from the zigbee2mqtt or from the thermostat, change some temperature values on the actual/present time interval and find out that home assistant is not updateing the new temperature, even the programming works fine. 
[image]

### Zigbee2MQTT version

 1.27.0 commit: a9b8808

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus USB stick

vladyspavlov, 2022-11-12:
Try to update to the latest zigbee2mqtt version (edge if you are using hassio zigbee2mqtt addon).
Maybe https://github.com/Koenkk/zigbee-herdsman-converters/pull/4917 fixed your issue.

catalinescu-sudo, 2022-11-13:
I have updated to version [1.28.2] commit: [360a777], in the docker and the behavior is the same, even some things have changed in the Home Assistant intergace. I don't know how to implement the issue #4917 (the link you mentioned), can you give me a hint? Thank you!

vladyspavlov, 2022-11-13:
> I have updated to version [1.28.2] commit: [[360a777](https://github.com/Koenkk/zigbee2mqtt/commit/360a777a4ef47de4fc708f120cd029ff6d754b04)]

Use the Zigbee2MQTT Edge addon or wait for the next z2m release. There is no this change in the current "stable" release.

kucik, 2022-11-29:
Sorry for offtopic, but could you please show me example of json for setting up programming_mode?

My attempts with json `"{ programming_mode: "06:00/16°C  17:30/18°C" }" sent to zigbee2mqtt/FRIENDLY_NAME/set`

ends up as:
`error 2022-11-29 10:10:18: Publish 'set' 'programming_mode' to 'valve_livingr' failed: 'TypeError: Cannot read properties of undefined (reading 'split')'`
Thank you

v6ak, 2022-12-01:
@kucik 

1. I think you need to specify all 12 values (4 events for workday, 4 events for saturday and 4 events for sunday)
2. It seems you need to use two spaces between the events, not just one.

kucik, 2022-12-01:
> @kucik
> 
> 1. I think you need to specify all 12 values (4 events for workday, 4 events for saturday and 4 events for sunday)
> 2. It seems you need to use two spaces between the events, not just one.

Thank you! Didn't noticed it wants exactly 12 values. It works now.

tijsrx, 2022-12-02:
> > @kucik
> > 
> > 1. I think you need to specify all 12 values (4 events for workday, 4 events for saturday and 4 events for sunday)
> > 2. It seems you need to use two spaces between the events, not just one.
> 
> Thank you! Didn't noticed it wants exactly 12 values. It works now.

Does it actually change the temperature? I had the same problem and fixed it with giving it 12 temperature setpoints separated by 2 spaces. I'm not getting any error now but the TRV ignores the settings completely.

romanqgit, 2022-12-09:
Any solution/workaround for this? I have the exact same issue

Rasfumi, 2022-12-16:
Same here, you can see at 7:00 the thermostat starts heating (in programming mode) but the setpoint does not get updated in the chart. Just before i took the screenshot i touched the thermostat and it immediately updated the setpoint to the correct 20°C

[image]

svisso, 2023-01-22:
Is there anyone researching on the BRT-100-TRV? This device worked perfectly until V 1.28.0 and does not work any more since. Does anybody have a replacement converter that I can use?

alancoleman1972, 2023-01-27:
I accidentally posted a duplicate issue (#15963) as I was having the same problem.  I have since purchased another BRT-100-TRV for another room and although it's configured the same it behaves differently such that it DOES update the temperature set when in programming mode.  The only difference between the two devices that I can think of is that I have not yet renamed the device from when it was added:-
[image]
TRV-Living room doesn't update....
[image]
But the new unnamed one does....
[image]

Is there any information I could extract to help?

MrSergioRodrigo, 2023-01-30:
+ have the same issue.

I also see here:
https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html#moes-brt-100-trv

Unable to read (/receive) this value (in Programming_mode)

alancoleman1972, 2023-03-02:
Still an issue.

harristom, 2023-04-03:
Issue is still present

osmeest, 2023-04-17:
I also have this unfortunate behavior in all of my 4 valves.
Is there something that could be done through the Reporting page of the device settings ? (I couldn't find information about that page.)

closed 2023-05-25 by github-actions[bot]

yanncharlou, 2023-12-24:
Issue is still present.