Koenkk/zigbee2mqtt #14941: TS0601 Tuya Human presence sensor _TZE200_ztc6ggyl

Althir1, 2022-11-12:
### Link

https://de.aliexpress.com/item/1005004326898936.html?spm=a2g0o.order_list.0.0.21ef5c5fqokBzu&gatewayAdapt=glo2deu

### Database entry

{"id":38,"type":"Router","ieeeAddr":"0xe0798dfffecd2b98","nwkAddr":31039,"manufId":4098,"manufName":"_TZE200_ztc6ggyl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000i\u0000\u0000\u0000\u0000i\u0000\u0000\u0000\u0000i\u0000\u0000\u0000\u0000i\u0000\u0000\u0000\u0000i\u0000\u0000\u0000\u0000i\u0000\u0000\u0000\u0000i\u0000\u0000\u0000\u0000i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_ztc6ggyl","stackVersion":0,"dateCode":"","appVersion":65}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1668253816861,"defaultSendRequestWhen":"immediate"}

### Comments

Home Assistant Add-on: Zigbee2mqtt Edge
Log: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ztc6ggyl'

Koenkk, 2022-11-12:
`_TZE200_ztc6ggyl` is supported in the latest dev branch.  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-12 by Koenkk

dennis-bell, 2022-11-24:
@Koenkk  do you plan to release again before new year?
Rather not switch to development

Koenkk, 2022-11-25:
@dennis-bell yes, every 1st of the month