Koenkk/zigbee2mqtt #14944: Must restart z2m after a network or coordinator restart.

JackTalisker, 2022-11-12:
### What happened?

With recent releases, if for some reason I have to restart my wifi router or simply restarting the coordinator, z2m lose its devices and must be restarted to regain control. 
This did not happen until recently.

### What did you expect to happen?

a z2m normal behaviour after a network or coordinator reboot.

### How to reproduce it (minimal and precise)

Reboot the router or the coordinator (e.g. after a firmware upgrade). Z2m loses control over its devices,
Have to restart z2m.

Debug log attached with z2m manual restart at 14:15:52

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

6.7.9.0 build 405

### Adapter

Sonoff ZBBridge Tasmota flashed

### Debug log

Debug 2022-11-12 14:13:37Received MQTT message on 'zigbee2mqtt/wayang/set' with data 'ON'
Debug 2022-11-12 14:13:37Publishing 'set' 'state' to 'wayang'
Debug 2022-11-12 14:13:38Received MQTT message on 'zigbee2mqtt/wayang/set' with data 'OFF'
Debug 2022-11-12 14:13:38Publishing 'set' 'state' to 'wayang'
Error 2022-11-12 14:13:39Publish 'set' 'state' to 'wayang' failed: 'Error: Command 0x7cb03eaa0a083cab/3 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Debug 2022-11-12 14:13:39Error: Command 0x7cb03eaa0a083cab/3 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error) at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:465:19) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Info 2022-11-12 14:13:39MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'wayang' failed: 'Error: Command 0x7cb03eaa0a083cab/3 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (sendZclFrameToEndpointInternal error)'","meta":{"friendly_name":"wayang"},"type":"zigbee_publish_error"}'
Error 2022-11-12 14:13:41Publish 'set' 'state' to 'wayang' failed: 'Error: Command 0x7cb03eaa0a083cab/3 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Debug 2022-11-12 14:13:41Error: Command 0x7cb03eaa0a083cab/3 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error) at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:465:19) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Info 2022-11-12 14:13:41MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'wayang' failed: 'Error: Command 0x7cb03eaa0a083cab/3 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (sendZclFrameToEndpointInternal error)'","meta":{"friendly_name":"wayang"},"type":"zigbee_publish_error"}'
Warning 2022-11-12 14:13:48Failed to ping 'lumino' (attempt 1/2, Read 0x7cb03eaa0a083b12/3 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
Warning 2022-11-12 14:13:53Failed to ping 'lumino' (attempt 2/2, Read 0x7cb03eaa0a083b12/3 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
Debug 2022-11-12 14:13:56Active device 'lumino' was last seen '1.50' minutes ago.
Debug 2022-11-12 14:13:56Device 'lumino' reconnected
Info 2022-11-12 14:13:56MQTT publish: topic 'zigbee2mqtt/lumino/availability', payload 'online'
Debug 2022-11-12 14:13:58Retrieving state of 'lumino' after reconnect
Error 2022-11-12 14:14:00Failed to read state of 'lumino' after reconnect (Read 0x7cb03eaa0a083b12/3 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
Warning 2022-11-12 14:14:02Failed to ping 'wayang' (attempt 1/2, Read 0x7cb03eaa0a083cab/3 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
Warning 2022-11-12 14:14:07Failed to ping 'wayang' (attempt 2/2, Read 0x7cb03eaa0a083cab/3 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
Debug 2022-11-12 14:14:10Active device 'wayang' was last seen '10.25' minutes ago.
Info 2022-11-12 14:14:10MQTT publish: topic 'zigbee2mqtt/wayang/availability', payload 'offline'
Debug 2022-11-12 14:14:36Received MQTT message on 'zigbee2mqtt/sala_2/set' with data '{"state":"ON"}'
Debug 2022-11-12 14:14:36Publishing 'set' 'state' to 'sala_2'
Error 2022-11-12 14:14:38Publish 'set' 'state' to 'sala_2' failed: 'Error: Command 0xd0cf5efffef13574/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Debug 2022-11-12 14:14:38Error: Command 0xd0cf5efffef13574/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error) at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:465:19) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Info 2022-11-12 14:14:38MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'sala_2' failed: 'Error: Command 0xd0cf5efffef13574/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (sendZclFrameToEndpointInternal error)'","meta":{"friendly_name":"sala_2"},"type":"zigbee_publish_error"}'
Debug 2022-11-12 14:14:39Saving state to file /app/data/state.json
[... cut ...]
Info 2022-11-12 14:16:37MQTT publish: topic 'zigbee2mqtt/sala_2', payload '{"brightness":254,"color_mode":"color_temp","color_temp":370,"last_seen":"2022-11-12T13:16:35.335Z","linkquality":220,"power_on_behavior":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Debug 2022-11-12 14:16:38Received Zigbee message from 'cc2531', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 8 with groupID 0
Info 2022-11-12 14:16:38MQTT publish: topic 'zigbee2mqtt/cc2531', payload '{"last_seen":"2022-11-12T13:16:37.134Z","led":true,"linkquality":100}'

JackTalisker, 2022-11-14:
That is a problem since my wifi router is scheduled for reboot once a week...

Hedda, 2022-11-18:
Also FYI, keep in mind that the EZSP support is currently experimental, so you can follow and try to help the ezsp developer here:

https://github.com/Koenkk/zigbee-herdsman/issues/319

closed 2022-12-02 by JackTalisker