Koenkk/zigbee2mqtt #14945: Aqara Smart Radiator Thermostat E1 /  Xiaomi SRTS-A01

nTx-Cheerok, 2022-11-12:
### What happened?

Basically two  issues, which seem to be connected in my eyes:

1. Setting the option of sensor from "internal" to "external", throws an error. (see logs below)
Also constantly says the adapter does not recognize value 252 (i think thats the sensor value ?)

I also think that this inadvertently causes : 
2. I am using Better Thermostat integration for automation and i first recognized, that it has issues setting the correct Heating Setpoint.  Once you set it, it stays on the desired level for minutes or seconds, but every status change like opening windows or closing them and the temp sensor trying to induce a new value for the local temp, it leads to the temperature jumping around. I think the thermostat gets the value for the temperature correction and interprets it as a new setpoint. 

I also communicate with the BT guys in this thread https://github.com/KartoffelToby/better_thermostat/issues/519
So maybe synergies can help here ?

Thanks for all your work guys :)

### What did you expect to happen?

That the temperature would stay on my set point.

### How to reproduce it (minimal and precise)

Just set the value of "sensor"

### Zigbee2MQTT version

1.28.2-1 

### Adapter firmware version

20220928

### Adapter

Sonoff USB Dongle 3.0

### Debug log

{   "away_preset_temperature": "15.0",
    "battery": 100,
    "calibrated": true,
    "child_lock": "LOCK",
    "linkquality": 174,
    "local_temperature": 19.9,
    "occupied_heating_setpoint": 19,
    "preset": "auto",
    **"sensor": "internal",**
    "system_mode": "heat",
    "valve_alarm": false,
    "valve_detection": "ON",
    "window_detection": "ON",
    "window_open": false,
    "device": {
        "applicationVersion": 25,
        "dateCode": "Jun  7 2022",
        "friendlyName": "0x54ef4410005bd019",
        "hardwareVersion": 1,
        "ieeeAddr": "0x54ef4410005bd019",
        "manufacturerID": 4447,
        "manufacturerName": "LUMI",
        "model": "SRTS-A01",
        "networkAddress": 21934,
        "powerSource": "Battery",
        "softwareBuildID": "0.0.0_0025",
        "type": "EndDevice",
        "zclVersion": 3
    },
    "voltage": null
}

##################################

{"state":"idle"},"update_available":false,"voltage":233}'
Error 2022-11-12 14:15:23Publish 'set' 'sensor' to '0x54ef4410005bd019' failed: 'Error: Write 0x54ef4410005bd019/1 aqaraOpple({"65522":{"value":

Error 2022-11-12 14:16:34Publish 'set' 'sensor' to '0x54ef4410005bd019' failed: 'Error: Write 0x54ef4410005bd019/1 aqaraOpple({"65522":{"value":[170,113,49,68,18,94,2,65,46,99,111,156,155,61,4,84,239,68,16,0,91,208,25,0,21,141,0,1,157,27,152,0,1,0,85,19,10,2,0,0,100,4,206,194,182,200,0,0,0,0,0,1,61,100,101],"type":65}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 21934 - 1 - 25 - 64704 - 4 after 10000ms)'

Debug 2022-11-12 14:24:27lumi.sensor_magnet: Processed data into payload {"contact":true,"voltage":2995,"battery":97,"power_outage_count":60}
Debug 2022-11-12 14:24:39Received MQTT message on 'zigbee2mqtt/0x54ef4410005bd019/set' with data '{"sensor":"external"}'
Debug 2022-11-12 14:24:39Publishing 'set' 'sensor' to '0x54ef4410005bd019'
Debug 2022-11-12 14:24:58Received MQTT message on 'zigbee2mqtt/0x54ef4410005bd019/get' with data '{"sensor":""}'
Debug 2022-11-12 14:24:58Publishing get 'get' 'sensor' to '0x54ef4410005bd019'
Error 2022-11-12 14:25:02Publish 'set' 'sensor' to '0x54ef4410005bd019' failed: 'Error: Write 0x54ef4410005bd019/1 aqaraOpple({"65522":{"value":[170,113,49,68,18,94,2,65,46,99,111,158,150,61,4,84,239,68,16,0,91,208,25,0,21,141,0,1,157,27,152,0,1,0,85,19,10,2,0,0,100,4,206,194,182,200,0,0,0,0,0,1,61,100,101],"type":65}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 21934 - 1 - 32 - 64704 - 4 after 10000ms)'
Info 2022-11-12 14:25:02MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'sensor' to '0x54ef4410005bd019' failed: 'Error: Write 0x54ef4410005bd019/1 aqaraOpple({\"65522\":{\"value\":[170,113,49,68,18,94,2,65,46,99,111,158,150,61,4,84,239,68,16,0,91,208,25,0,21,141,0,1,157,27,152,0,1,0,85,19,10,2,0,0,100,4,206,194,182,200,0,0,0,0,0,1,61,100,101],\"type\":65}}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 21934 - 1 - 32 - 64704 - 4 after 10000ms)'","meta":{"friendly_name":"0x54ef4410005bd019"},"type":"zigbee_publish_error"}'
Error 2022-11-12 14:25:25Publish 'get' 'sensor' to '0x54ef4410005bd019' failed: 'Error: Read 0x54ef4410005bd019/1 aqaraOpple([638], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 21934 - 1 - 34 - 64704 - 1 after 10000ms)'
Info 2022-11-12 14:25:25MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'sensor' to '0x54ef4410005bd019' failed: 'Error: Read 0x54ef4410005bd019/1 aqaraOpple([638], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 21934 - 1 - 34 - 64704 - 1 after 10000ms)'","meta":{"friendly_name":"0x54ef4410005bd019"},"type":"zigbee_publish_error"}'

nTx-Cheerok, 2022-11-12:
Okay i have now

nTx-Cheerok, 2022-11-12:
Issue returned :(

nTx-Cheerok, 2022-11-12:
It also seems that it starts with the interviewing/configuration. It gets recognized and goes through everything. You can control it for some minutes, but then starts to get uncontrollable, almost like it leaves the network. Its very weird.

nTx-Cheerok, 2022-11-12:
Sorry for spaming this thread :D Currently i have used on of my spare Aqara E1. It seems to work now. Maybe the other trv is faulty ;/ Ill report any changes.

Slomo17, 2022-11-13:
Don't know if my problem is related, but i experience a simmilar issue since yesterday with the occupied_heating_setpoint. Most of the time a cannot change it, right after changing its resets to the previous set temperature. Sometime i can change it, couldn'ffigure out when exactly, yet. Repairing multiple time doesn't solved the problem. I am running the lates developer version of z2m (1.28.2-dev commit:499b993). Reverting to the lates stable release (1.28.2) didn't help either.

nTx-Cheerok, 2022-11-13:
Well just to add to this, the second E1 showed the same issue i described. I would also send one to someones P/O box if anyone needs one for testing :D Or just a checklist what to gather in case of logs and so one to contribute in some form.

KimHEF, 2022-11-22:
> Well just to add to this, the second E1 showed the same issue i described. I would also send one to someones P/O box if anyone needs one for testing :D Or just a checklist what to gather in case of logs and so one to contribute in some form.

Did you find any solution?

akonfxd, 2022-11-29:
I had the same problem as described with my Aqara E1. I will try to debug a little bit more. However it might take a while since i migrated my zigbee network recently to zigbe2mqtt

nTx-Cheerok, 2022-12-01:
@KimHEF no i had to switch back to my TS601s. I got kids in this flat. I need some reliability :/ If ti was just me, i would gladly offer myself as a testing ground :D 

@akonfxd would be awesome. Have not tried with the newest BT updates though. If those change anything, it should be checked first as to not make this an issue thats already fixed :) Thanks for your help!

closed 2023-01-08 by github-actions[bot]

Re4zOon, 2023-01-16:
I don't know why it's closed, mine acts the same way. Basically I get all updates from the trv (if I change anything manually, or it's internal temp sensor), but cannot push anything, it just times out.

nTx-Cheerok, 2023-02-06:
> I don't know why it's closed, mine acts the same way. Basically I get all updates from the trv (if I change anything manually, or it's internal temp sensor), but cannot push anything, it just times out.

If you still got the issue: My issue was fixed with updating the coordinator firmware. I used a Sonoff Zigbee Dongle 3.0. 
Switched to Home Assistant Yellow now. Not an issue there.