Koenkk/zigbee2mqtt #14947: TV02-Zigbee incorrectly shows the temperature in automatic mode.

iksrd, 2022-11-12:
### What happened?

TV02-Zigbee incorrectly shows the temperature in automatic mode. The following schedule is set 04:00/22 09:00/21.5 18:00/22 21:30/21 24:00/21. In reality, the temperature does not match the schedule. It feels like time is jumping.
[image]
[image]

### What did you expect to happen?

When the schedule is running automatically, it corresponds to the set time.

### How to reproduce it (minimal and precise)

Set up a schedule with different temperatures for different times and observe

### Zigbee2MQTT version

 1.28.2

### Adapter firmware version

 20211217

### Adapter

zStack3x0

vladyspavlov, 2022-11-12:
Could you please provide the manufacturer ID of your TRV?

iksrd, 2022-11-13:
TS0601
_TZE200_e9ba97v

vladyspavlov, 2022-11-13:
Does the time set on TRV match your real time? I can't reproduce it on my TRV with the same manuf id.

iksrd, 2022-11-14:
Yes, the time corresponds. I recently reset the thermal heads to the factory state.

iksrd, 2022-11-14:
I'll check, in my opinion I just misunderstood the principle of the schedule.

closed 2022-11-15 by iksrd