Koenkk/zigbee2mqtt #14950: Eurotronic Spirit SPBZ0001 Thermostat sends manual Changes on the temperature only as current_heating_setpoint

DevFlorian, 2022-11-12:
### What happened?

My Eurotronic Spirit SPBZ0001 Thermostat sends manual changes on the temperature only as current_heating_setpoint. So Homeassistant and Zigbee2Mqtt don't Show the actual setpoint.
It uses firmware 20190408. Firmware 20191014 also sends occupied_heating_setpoint.

### What did you expect to happen?

Maybe it's possible to fix this by reading current_heating_setpoint as occupied_heating_setpoint, If the value ist not the same.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

6.7.8.0 build 373

### Adapter

EZSP v8

lux73, 2022-11-13:
not possible to fix this with wrote your own Rule in HASS?

`if (current_heating_setpoint != occupied_heating_setpoint) current_heating_setpoint.postUpdate(occupied_heating_setpoint)`

would be a possible Solution for OpenHAB based HomeAutomation

or had i misunderstood the Problem?

DevFlorian, 2022-11-19:
The Value Template in HA is also set by Zigbee2Mqtt via Auto discovery. I could do a workaround in HA, but to me it should still be fixed in Zigbee2Mqtt.

rolfcoder, 2022-11-29:
Exactly same problem (SPBZ0001 firmware 20190408 & 20191014) and and with Zigbee2Mqtt v1.28.2.
I'm using zStack3x0 with 20220219.
This used to work before I updated to 1.28.2.

closed 2023-01-06 by github-actions[bot]