Koenkk/zigbee2mqtt #14959: Gledopto GL-C-007 not working although supported

unf0rg0tt3n, 2022-11-13:
### Link

https://www.gledopto.eu/rgbw-controller-controller-zigbee-compatible_1

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0x00124b00213956a2","nwkAddr":652,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-007","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":49246,"epId":13,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.1","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1668345341397,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,

I have been running gledopto GL-C-007 for a few years now in combination with deconz and conbee2. Recently I got the sonoff plus-ZBdongle-E and switched to Z2M. love it, great work. but the gledopto GL-C-007 is the only device which won't play along.

I also have a GL-C-006 (ww/cw) and that is working great.

When added the following log entries appear:
`Info 2022-11-13 14:15:41Device '0x00124b00213956a2' joined
Info 2022-11-13 14:15:41MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00213956a2","ieee_address":"0x00124b00213956a2"},"type":"device_joined"}'
Info 2022-11-13 14:15:41Starting interview of '0x00124b00213956a2'
Info 2022-11-13 14:15:41MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00213956a2","ieee_address":"0x00124b00213956a2","status":"started"},"type":"device_interview"}'
Info 2022-11-13 14:16:02Successfully interviewed '0x00124b00213956a2', device has successfully been paired
Warning 2022-11-13 14:16:02Device '0x00124b00213956a2' with Zigbee model 'GL-C-007' and manufacturer name 'GLEDOPTO' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2022-11-13 14:16:02MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x00124b00213956a2","ieee_address":"0x00124b00213956a2","status":"successful","supported":false},"type":"device_interview"}`

I also tried to create an external converter based on another github issue but that doesn't work either. 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const gledoptoExtend = {
    light_onoff_brightness: (options={}) => ({
        ...extend.light_onoff_brightness(options),
        toZigbee: utils.replaceInArray(
            extend.light_onoff_brightness(options).toZigbee,
            [tz.light_onoff_brightness],
            [tz.gledopto_light_onoff_brightness],
        ),
    }),
    light_onoff_brightness_colortemp: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        toZigbee: utils.replaceInArray(
            extend.light_onoff_brightness_colortemp(options).toZigbee,
            [tz.light_onoff_brightness, tz.light_colortemp],
            [tz.gledopto_light_onoff_brightness, tz.gledopto_light_colortemp],
        ),
    }),
    light_onoff_brightness_color: (options={}) => ({
        ...extend.light_onoff_brightness_color({...options, supportsHS: true}),
        toZigbee: utils.replaceInArray(
            extend.light_onoff_brightness_color(options).toZigbee,
            [tz.light_onoff_brightness, tz.light_color],
            [tz.gledopto_light_onoff_brightness, tz.gledopto_light_color],
        ),
    }),
    light_onoff_brightness_colortemp_color: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp_color({...options, supportsHS: true}),
        toZigbee: utils.replaceInArray(
            extend.light_onoff_brightness_colortemp_color(options).toZigbee,
[... 108 lines cut ...]
        meta: {disableDefaultResponse: (entity) => !!entity.getDevice().getEndpoint(12)},
        endpoint: (device) => {
            return {rgb: 11, cct: 15};
        },
    },
    {
        fingerprint: [
            {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GLEDOPTO', endpoints: [
                {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
                {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
            ]},
        ],
        zigbeeModel: ['GL-C-008'],
        model: 'GL-C-008-1ID', // 1 ID controls color temperature and color separate
        vendor: 'Gledopto',
        description: 'Zigbee LED Controller RGB+CCT (1 ID)',
        extend: gledoptoExtend.light_onoff_brightness_colortemp_color(),
        meta: {disableDefaultResponse: true},
    },
];
```

### Supported color modes

RGBWW

liamlows, 2022-11-15:
Recently I purchased a Gledopto RGBCCT 2ID controller and ran into the same issue. On join the log read: 
```
info  2022-11-14 23:34:57: Zigbee: allowing new devices to join.
info  2022-11-14 23:34:57: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"ej0r0-1"}'
info  2022-11-14 23:35:05: Device '0x00124b0021efbe1a' joined
info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021efbe1a","ieee_address":"0x00124b0021efbe1a"},"type":"device_joined"}'
info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b0021efbe1a"},"type":"device_connected"}'
info  2022-11-14 23:35:05: Starting interview of '0x00124b0021efbe1a'
info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021efbe1a","ieee_address":"0x00124b0021efbe1a","status":"started"},"type":"device_interview"}'
info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b0021efbe1a"},"type":"pairing"}'
info  2022-11-14 23:35:26: Successfully interviewed '0x00124b0021efbe1a', device has successfully been paired
warn  2022-11-14 23:35:26: Device '0x00124b0021efbe1a' with Zigbee model 'GL-C-007' and manufacturer name 'GLEDOPTO' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-11-14 23:35:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x00124b0021efbe1a","ieee_address":"0x00124b0021efbe1a","status":"successful","supported":false},"type":"device_interview"}'
info  2022-11-14 23:35:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x00124b0021efbe1a","supported":false},"type":"pairing"}'
info  2022-11-14 23:36:41: Zigbee: disabling joining new devices.
info  2022-11-14 23:36:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"ej0r0-2"}'
info  2022-11-14 23:43:04: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
```
After some googling and researching I found the follow issue that referenced altering the database.db file. @Koenkk suggested changing the `modelID` from whatever it was (for me it was `GL-C-007`) to `GL-C-007`.

https://github.com/Koenkk/zigbee2mqtt/issues/2872#issuecomment-581780555

After stopping the addon, making the change, and then starting the addon I was able to see the device and control only the RGB colors. if i plugged in WW or CW on the controller it turned them on at full power. Without them RGB worked normally.

I did try to reflash the firmware on the V2 dongle but couldn't get it to work using secureCRT and Xmodem(Z).

unf0rg0tt3n, 2022-11-15:
> Recently I purchased a Gledopto RGBCCT 2ID controller and ran into the same issue. On join the log read:
> 
> ```
> info  2022-11-14 23:34:57: Zigbee: allowing new devices to join.
> info  2022-11-14 23:34:57: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"ej0r0-1"}'
> info  2022-11-14 23:35:05: Device '0x00124b0021efbe1a' joined
> info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021efbe1a","ieee_address":"0x00124b0021efbe1a"},"type":"device_joined"}'
> info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b0021efbe1a"},"type":"device_connected"}'
> info  2022-11-14 23:35:05: Starting interview of '0x00124b0021efbe1a'
> info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021efbe1a","ieee_address":"0x00124b0021efbe1a","status":"started"},"type":"device_interview"}'
> info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b0021efbe1a"},"type":"pairing"}'
> info  2022-11-14 23:35:26: Successfully interviewed '0x00124b0021efbe1a', device has successfully been paired
> warn  2022-11-14 23:35:26: Device '0x00124b0021efbe1a' with Zigbee model 'GL-C-007' and manufacturer name 'GLEDOPTO' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
> info  2022-11-14 23:35:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x00124b0021efbe1a","ieee_address":"0x00124b0021efbe1a","status":"successful","supported":false},"type":"device_interview"}'
> info  2022-11-14 23:35:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x00124b0021efbe1a","supported":false},"type":"pairing"}'
> info  2022-11-14 23:36:41: Zigbee: disabling joining new devices.
> info  2022-11-14 23:36:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"ej0r0-2"}'
> info  2022-11-14 23:43:04: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
> ```
> 
> which is weird because the device was seen as `GL-C-007` but the device had a model number printed on it that was `GL-C-008`.
> 
> After some googling and researching I found the follow issue that referenced altering the database.db file. @Koenkk suggested changing the `modelID` from whatever it was (for me it was `GL-C-007`) to `GL-C-008`.
> 
> [#2872 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/2872#issuecomment-581780555)
> 
> After stopping the addon, making the change, and then starting the addon I was able to see the device and control only the RGB colors. if i plugged in WW or CW on the controller it turned them on at full power. Without them RGB worked normally so this wasn't a fix but at least something that partially worked.
> 
> I did try to reflash the firmware on the Sonoff V2 dongle but couldn't get it to work using secureCRT and Xmodem(Z). I am getting the older Ti version here soon but hope I can continue using the V2 edition.
> 
> For reference, the link to the device stating its support can be found here: https://www.zigbee2mqtt.io/devices/GL-C-008-2ID.html

Life saver! you fixed my issue!

closed 2022-11-15 by unf0rg0tt3n

liamlows, 2022-11-15:
@unf0rg0tt3n that great to hear! Did you have a 2ID version though?

I found that I am still having issues since with the 2ID version where i should be able to control color and color temperature independently. The manufacturer states:

"Note: The RGBCCT controller is 2ID, Will show two SSIDs (Two light: Color and Color temperature) in the APP when you use. Easy to manage the lights. Color light （RGB) can work with white light together or separately, but maybe some gateways only supports one SSID. Important note: Works with the zigbee bridge & app"

Does anyone know if zigbee2mqtt supports the 2ID feature?

unf0rg0tt3n, 2022-11-15:
@liamlows I have the 1ID version. 2ID says unsupported. I found out that although gledopto is supported, it's really flaky

liamlows, 2022-11-16:
> @liamlows I have the 1ID version. 2ID says unsupported. I found out that although gledopto is supported, it's really flaky

Hmm, but doesn't it mean its supported if its in the devices on their docs? See the link below...

https://www.zigbee2mqtt.io/devices/GL-C-008-2ID.html