Koenkk/zigbee2mqtt #14962: Moes 1 gang switch discovered with wrong model

f0gster, 2022-11-13:
### What happened?

I have Moes 1 gang switch with neutral wire (model ZS-EUB).

The switch was detected as [ZTS-EU_1gang](https://www.zigbee2mqtt.io/devices/ZTS-EU_1gang.html#moes-zts-eu_1gang), (which has no neutral wire).

Switching works fine, the problem is that it is recognized as EndDevice and not a router... To me, it looks exactly like closed issue #11665 . I've commented there, but it is closed while the problem is still here.

### What did you expect to happen?

Recognized as itself [ZS-EUB_1gang](https://www.zigbee2mqtt.io/devices/ZS-EUB_1gang.html)

### How to reproduce it (minimal and precise)

Deleting and pairing again makes no difference - it is consistently recognized as ZTS-EU_1gang.
I've seen the same across several recent zigbee2mqtt versions. I've tried on 1.28, then 1.28.1 and now on 1.28.2 - behaviour is the same all the time.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

Here is an entry from the database.db:
{"id":2,"type":"EndDevice","ieeeAddr":"0x540f57fffe37caea","nwkAddr":48409,"manufId":4098,"manufName":"_TZE200_amp6tsvy","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u001ai\u0002+\u0011","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_amp6tsvy","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bfd765","endpointID":242}],"config
uredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1482240604},"lastSeen":1668360426927,"defaultSendRequestWhen":"immediate"}

Here is a log of discovery and join:

info  2022-11-10 21:45:28: Device '0x540f57fffe37caea' joined
info  2022-11-10 21:45:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x540f57fffe37caea","ieee_address":"0x540f57fffe37caea"},"type":"device_joined"}'
info  2022-11-10 21:45:28: Starting interview of '0x540f57fffe37caea'
info  2022-11-10 21:45:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x540f57fffe37caea","ieee_address":"0x540f57fffe37caea","status":"started"},"type":"device_interview"}'
info  2022-11-10 21:45:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x540f57fffe37caea","ieee_address":"0x540f57fffe37caea"},"type":"device_announce"}'
info  2022-11-10 21:45:30: Successfully interviewed '0x540f57fffe37caea', device has successfully been paired
info  2022-11-10 21:45:30: Device '0x540f57fffe37caea' is supported, identified as: Moes Wall touch light switch (1 gang) (ZTS-EU_1gang)
info  2022-11-10 21:45:30: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Wall touch light switch (1 gang)","exposes":[{"features":[{"access":3,"description":"On/off sta
te of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":3,"description":"Indicator light status","name":"indicate_ligh
t","property":"indicate_light","type":"enum","values":["off","switch","position","freeze"]},{"access":3,"description":"Controls the behavior when the device is powered on","name":"power_on_behavior","property":"power_
on_behavior","type":"enum","values":["off","on","previous"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"valu
e_min":0}],"model":"ZTS-EU_1gang","options":[],"supports_ota":false,"vendor":"Moes"},"friendly_name":"0x540f57fffe37caea","ieee_address":"0x540f57fffe37caea","status":"successful","supported":true},"type":"device_inte
rview"}'
info  2022-11-10 21:45:30: Configuring '0x540f57fffe37caea'
info  2022-11-10 21:45:30: MQTT publish: topic 'homeassistant/switch/0x540f57fffe37caea/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"comman
d_topic":"zigbee2mqtt/0x540f57fffe37caea/set","device":{"identifiers":["zigbee2mqtt_0x540f57fffe37caea"],"manufacturer":"Moes","model":"Wall touch light switch (1 gang) (ZTS-EU_1gang)","name":"0x540f57fffe37caea"},"js
on_attributes_topic":"zigbee2mqtt/0x540f57fffe37caea","name":"0x540f57fffe37caea","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x540f57fffe37caea","unique_id":"0x540f57fffe37caea_switch_zigbee2mqtt
","value_template":"{{ value_json.state }}"}'
info  2022-11-10 21:45:30: MQTT publish: topic 'homeassistant/sensor/0x540f57fffe37caea/indicate_light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}]
,"device":{"identifiers":["zigbee2mqtt_0x540f57fffe37caea"],"manufacturer":"Moes","model":"Wall touch light switch (1 gang) (ZTS-EU_1gang)","name":"0x540f57fffe37caea"},"enabled_by_default":false,"json_attributes_topi
c":"zigbee2mqtt/0x540f57fffe37caea","name":"0x540f57fffe37caea indicate light","state_topic":"zigbee2mqtt/0x540f57fffe37caea","unique_id":"0x540f57fffe37caea_indicate_light_zigbee2mqtt","value_template":"{{ value_json
.indicate_light }}"}'
info  2022-11-10 21:45:30: MQTT publish: topic 'homeassistant/select/0x540f57fffe37caea/indicate_light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}]
,"command_topic":"zigbee2mqtt/0x540f57fffe37caea/set/indicate_light","device":{"identifiers":["zigbee2mqtt_0x540f57fffe37caea"],"manufacturer":"Moes","model":"Wall touch light switch (1 gang) (ZTS-EU_1gang)","name":"0
x540f57fffe37caea"},"json_attributes_topic":"zigbee2mqtt/0x540f57fffe37caea","name":"0x540f57fffe37caea indicate light","options":["off","switch","position","freeze"],"state_topic":"zigbee2mqtt/0x540f57fffe37caea","un
ique_id":"0x540f57fffe37caea_indicate_light_zigbee2mqtt","value_template":"{{ value_json.indicate_light }}"}'
info  2022-11-10 21:45:30: MQTT publish: topic 'homeassistant/sensor/0x540f57fffe37caea/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}
"}],"device":{"identifiers":["zigbee2mqtt_0x540f57fffe37caea"],"manufacturer":"Moes","model":"Wall touch light switch (1 gang) (ZTS-EU_1gang)","name":"0x540f57fffe37caea"},"enabled_by_default":false,"entity_category":
"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/0x540f57fffe37caea","name":"0x540f57fffe37caea power on behavior","state_topic":"zigbee2mqtt/0x540f57fffe37caea","unique_id":"0x540f57fffe37cae
a_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
info  2022-11-10 21:45:30: MQTT publish: topic 'homeassistant/select/0x540f57fffe37caea/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}
"}],"command_topic":"zigbee2mqtt/0x540f57fffe37caea/set/power_on_behavior","device":{"identifiers":["zigbee2mqtt_0x540f57fffe37caea"],"manufacturer":"Moes","model":"Wall touch light switch (1 gang) (ZTS-EU_1gang)","na
me":"0x540f57fffe37caea"},"entity_category":"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/0x540f57fffe37caea","name":"0x540f57fffe37caea power on behavior","options":["off","on","previous"]
,"state_topic":"zigbee2mqtt/0x540f57fffe37caea","unique_id":"0x540f57fffe37caea_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
info  2022-11-10 21:45:30: MQTT publish: topic 'homeassistant/sensor/0x540f57fffe37caea/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"d
evice":{"identifiers":["zigbee2mqtt_0x540f57fffe37caea"],"manufacturer":"Moes","model":"Wall touch light switch (1 gang) (ZTS-EU_1gang)","name":"0x540f57fffe37caea"},"enabled_by_default":false,"entity_category":"diagn
ostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x540f57fffe37caea","name":"0x540f57fffe37caea linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x540f57fffe37caea","unique_id":"0x540
[... cut ...]
info  2022-11-10 21:45:35: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe37caea', payload '{"indicate_light":"position","linkquality":123,"power_on_behavior":"previous","state":"ON"}'
info  2022-11-10 21:45:35: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe37caea', payload '{"indicate_light":"position","linkquality":123,"power_on_behavior":"previous","state":"ON"}'
info  2022-11-10 21:45:35: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe37caea',payload'{"indicate_light":"position","linkquality":123,"power_on_behavior":"previous","state":"ON"}'
the

f0gster, 2022-11-15:
looks also to be related to this one: #14333

f0gster, 2022-11-15:
I can see that ZTS-EU_1gang support came from here: #12739
I've tried forcing the model through the external converter and using the most recent code for ZTS-EU_1gang from moes.js

It does not work!

Koenkk, 2022-11-15:
> Switching works fine, the problem is that it is recognized as EndDevice and not a router...

How is is recognized has no influence over the functionality. The type is announced by the device, it is not something that comes from z2m.

closed 2022-12-23 by github-actions[bot]