Koenkk/zigbee2mqtt #14965: Moes Thermostat can only set whole degrees

PulsarFX, 2022-11-13:
### What happened?

Temperature target value for the [BHT-002-GCLZB](https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#moes-bht-002-gclzb)  is reported in whole degrees and can only be set as whole degrees, whereas the device reports and changes in .5 degree steps. 

### What did you expect to happen?

I'd like to see and set current_heating_setpoint in .5 degree steps via mqtt like it is possible at the device itself. 

### How to reproduce it (minimal and precise)

Try to set a temperature for current_heating_setpoint with .5 degrees. 

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P

PulsarFX, 2022-11-19:
Is there a way to test it with a different definition?
In z2m gui I can also set only whole degrees. Does it ise the same definition?

cambro1995, 2022-11-29:
Same problem with MOES [BHT-002-GCLZB]

TTR6214, 2022-12-10:
I also confirm that we cannot step by 0.5 degree for MOES BRT-100-TRV.

PulsarFX, 2022-12-12:
So I started digging around and found this:

my model starts here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/moes.js#L98
and states:
`fromZigbee: [fz.moes_thermostat],`

which is:
`const fz = require('../converters/fromZigbee');`

this is where the settings for moes_thermostat starts:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L3896
relevant snippet for local temp:
```
  case tuya.dataPoints.moesLocalTemp:
      temperature = value & 1<<15 ? value - (1<<16) + 1 : value;
      if (!['_TZE200_ztvwu4nk', '_TZE200_ye5jkfsb'].includes(meta.device.manufacturerName)) {
          // https://github.com/Koenkk/zigbee2mqtt/issues/11980
          temperature = temperature / 10;
      }

      return {local_temperature: parseFloat(temperature.toFixed(1))};
```

relevant snippet for heating setpoint:
```
case tuya.dataPoints.moesHeatingSetpoint:
    return {current_heating_setpoint: value};
```

my question is: 
- is fromZigbee.js right to convert heatingSetpoint as non-float? 
- is there a way to see the raw zigbee message from the thermostat to verify incoming data?

PulsarFX, 2022-12-14:
has anyone tested this thermostat with the smartlife/tuya/moes app if .5°C is possible there? MOES support wants me to show them with their app, that I can only set 1C steps...

Ivano62, 2022-12-14:
On MOES BRT-100-TRV With physical input on the device the step is 0.5 °C, I don't use app

PulsarFX, 2022-12-14:
Yes, that's why I contacted support, as the display accepts .5C steps whereas zigbee can only set 1C steps. The question is: is it a zigbee2mqtt issue or a general issue with the device?

PulsarFX, 2022-12-20:
Moes support states, that the thermostat can be set in .5C steps via Smartlife or their app.

osnwt, 2022-12-24:
Anyone has Tuya app to verify? I have no Tuya hub yet, so can't try myself.
Perhaps, this TRV has firmware update, but it isn't shown in Z2M. Maybe it will work with original Tuya app, and after update we would have better position reporting and proper setpoint option?..

PulsarFX, 2022-12-28:
I bought the moes wireless zigbee hub and connected it with one of my thermostats.
Guess what: smartlife app only allows to set whole degrees, too.  >:-(
The thermostat reports these versions:
zigbee module 1.0.8
mcu model: 2.0.1

osnwt, 2022-12-28:
Sad...

PulsarFX, 2022-12-31:
I've sent a Video to moes support and got this response:

"I  checked with TUYA team,zigbee thermostat can only set the temperature according to 1 C at present.
Wifi thermostat can set 0.5C,hope you understand.
I will apply for TUYA engineer to improve this function in the future."

Let's hope they get this fixed.
For now there is nothing z2m can do to improve the implementation.

closed 2022-12-31 by PulsarFX

guny74, 2023-01-04:
Very strage, because via program e.g.  "weekdays_pX_temperature" it is possible to set .5 Degrees values...why not for setpoint ?!

PulsarFX, 2023-12-02:
A year later, I asked again (also with the hint, that the program schedule can set .5 degrees) and got this reply from support :
"This is indeed a fixed specification parameter of graffiti technology at present.
We have also submitted an application, and hope that they can revise this problem as soon as possible after reviewing it. Thank you for your feedback again."

Honestly, I don't understand what they are trying to say :-D

PulsarFX, 2023-12-06:
@ all subscribers: please see #19991 and check your device code