Koenkk/zigbee2mqtt #14968: State update not reported in mqtt (and logs), although the device sends a new state

ingar195, 2022-11-13:
### What happened?

z2m is not sending any message if the state is the same as you are trying to change it to. no trace of it in the logs or in MQTT messages 

I know this device is sending this message because I used it in zha for automation before migrating to z2m

I'm using this device:
[WDE002386 ](https://www.zigbee2mqtt.io/devices/WDE002386.html#schneider%2520electric-wde002386)

### What did you expect to happen?

Expected all Zigbee events to be passed to MQTT, even if the state is on the same as the button press 

### How to reproduce it (minimal and precise)

have a light that's on then press the physical on button
then nothing will appear in the logs or in MQTT

Also tested on the latest dev docker image

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

0x26780700

### Adapter

Conbee II

closed 2022-12-21 by github-actions[bot]

ingar195, 2023-01-18:
Can this be reopned

ingar195, 2023-01-18:
@Koenkk sorry for tagging, I know this is an issue since the switches sent this message with ZHA. have verified with  DEBUG=zigbee-herdsman* environment var enabled nothing in logs.

Cyberfly100, 2023-02-23:
I think I have the same problem, too.