Koenkk/zigbee2mqtt #14969: [New device support]: TRÅDFRI LED bulb PAR38 E26 900 lumen

apadua, 2022-11-14:
### Link

https://www.ikea.com/us/en/p/tradfri-led-bulb-par38-e26-900-lumen-smart-wireless-dimmable-white-spectrum-20518742/

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x50325ffffebf3d77","nwkAddr":1523,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRIbulbPAR38WS900lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRIbulbPAR38WS900lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20201111","swBuildId":"1.0.012"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20201111","swBuildId":"1.0.012","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1668310685181,"defaultSendRequestWhen":"immediate"}

### Comments

I tried the basic recommendations but it was also unsupported I tried the dev branch - no game. I iopend ikea.js and found other bulbs. I tried creating my own converter stealing (refactoring) some code from those files. The external conerter is below:

You'll quickly see i didn't know what i was doilng - just adding stolen const untill all missing references were gone.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const tradfriExtend = {
    light_onoff_brightness: (options = {}) => ({
        ...extend.light_onoff_brightness(options),
        ota: ota.tradfri,
        onEvent: bulbOnEvent,
    }),
    light_onoff_brightness_colortemp: (options = {colorTempRange: [250, 454]}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        ota: ota.tradfri,
        onEvent: bulbOnEvent,
    }),
    light_onoff_brightness_colortemp_color: (options = {disableColorTempStartup: true, colorTempRange: [250, 454]}) => ({
        ...extend.light_onoff_brightness_colortemp_color(options),
        ota: ota.tradfri,
        onEvent: bulbOnEvent,
    }),
    light_onoff_brightness_color: (options = {}) => ({
        ...extend.light_onoff_brightness_color(options),
        ota: ota.tradfri,
        onEvent: bulbOnEvent,
    }),
};
const bulbOnEvent = async (type, data, device, options, state) => {
    /**
     * IKEA bulbs lose their configured reportings when losing power.
     * A deviceAnnounce indicates they are powered on again.
     * Reconfigure the configured reoprting here.
     *
     * Additionally some other information is lost like
     *   color_options.execute_if_off. We also restore these.
     *
     * NOTE: binds are not lost so rebinding is not needed!
     */
    if (type === 'deviceAnnounce') {
        for (const endpoint of device.endpoints) {
            for (const c of endpoint.configuredReportings) {
                await endpoint.configureReporting(c.cluster.name, [{
                    attribute: c.attribute.name, minimumReportInterval: c.minimumReportInterval,
                    maximumReportInterval: c.maximumReportInterval, reportableChange: c.reportableChange,
                }]);
            }
        }

        // NOTE: execute_if_off default is false
        //       we only restore if true, to save unneeded network writes
        if (state !== undefined && state.color_options !== undefined && state.color_options.execute_if_off === true) {
            device.endpoints[0].write('lightingColorCtrl', {'options': 1});
        }
        if (state !== undefined && state.level_config !== undefined && state.level_config.execute_if_off === true) {
            device.endpoints[0].write('genLevelCtrl', {'options': 1});
        }
        if (state !== undefined && state.level_config !== undefined && state.level_config.on_level !== undefined) {
            let onLevel = state.level_config.on_level;
            if (typeof onLevel === 'string' && onLevel.toLowerCase() == 'previous') {
                onLevel = 255;
            } else {
                onLevel = Number(onLevel);
            }
            if (onLevel > 255) onLevel = 254;
            if (onLevel < 1) onLevel = 1;

            device.endpoints[0].write('genLevelCtrl', {onLevel});
        }
    }
};

const definition = {
    zigbeeModel: ['TTRADFRIbulbPAR38WS900lm'],
    model: 'TRADFRIbulbPAR38WS900lm',
    vendor: 'IKEA',
    description: 'Tradfri E26 PAR38 900 lumen, dimmable, white spectrum, downlight',
    extend: tradfriExtend.light_onoff_brightness_colortemp(),

};

module.exports = definition;
```

### Supported color modes

color temperature

### Color temperature range

Error 2022-11-13 22:49:07Publish 'set' 'read' to '0x50325ffffebf3d77' failed: 'Error: Read 0x50325ffffebf3d77/1 lightingColorCtrl(["colorTempPhysicalMin"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

apadua, 2022-11-14:
Willing to test beta code. All i really want is on/off as a basic function.

danieledwardgeorgehitchcock, 2022-11-14:
Submitted on your behalf.. It should be available to test in the Dev branch a few hours after it has been merged

apadua, 2022-11-14:
I don't have the box anymore. The bulb is installed and was working in ZHA for a while. I took it down and photographed the bulb but there's nothing similar to what we document on the other Ikea bulbs (LEDXXXXX)
[image]

Koenkk, 2022-11-14:
Thanks, @danieledwardgeorgehitchcock added it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-14 by Koenkk

apadua, 2022-11-14:
@Koenkk I believe there's a typo in the model # - there's an extra T. I'm not 100% sure since i'm looking for the original logs and can't find it. i'll test the dev branch as soon as it's out and see if it's the T or not.

A.

apadua, 2022-11-14:
@Koenkk Confirmed typo. Must remove the extra T for it to work:
```
- zigbeeModel: ['TTRADFRIbulbPAR38WS900lm'],
+ zigbeeModel: ['TRADFRIbulbPAR38WS900lm'],
```