Koenkk/zigbee2mqtt #14974: Symfonisk Controller (Ikea E1744) not properly reporting up/down

habitoti, 2022-11-14:
### What happened?

When rotating the Symfonisk, it mostly reports "brightness_move_up". Even when rotating left, it maybe starts with "brightness_move_down" (not always), but then quickly changes to "brightness_move_up" even when constantly continuing to rotate left. Sometimes an erratic move_down might intermittently appear, but mostly it sends move_up.

See debug log below. This is ONLY during constant left rotation, but mostly comes up with brightness_move_up events and only a few move_down events.

### What did you expect to happen?

On left rotation, only brightness_move_down should be reported.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

0x26720700

### Adapter

Conbee II

### Debug log

Debug 2022-11-14 10:44:28Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":195}' from endpoint 1 with groupID null
Debug 2022-11-14 10:44:28Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":1,"rate":195}' from endpoint 1 with groupID null
Info 2022-11-14 10:44:28MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"brightness_move_up","action_rate":195,"battery":100,"brightness":75,"last_seen":"2022-11-14T09:44:28.831Z","linkquality":248,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:28MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"","battery":100,"brightness":75,"last_seen":"2022-11-14T09:44:28.831Z","linkquality":248,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:28MQTT publish: topic 'zigbee2mqtt/Symfonisk/action', payload 'brightness_move_up'
Debug 2022-11-14 10:44:28Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":195}' from endpoint 1 with groupID null
Debug 2022-11-14 10:44:29Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":1,"rate":195}' from endpoint 1 with groupID null
Info 2022-11-14 10:44:29MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"brightness_move_up","action_rate":195,"battery":100,"brightness":90,"last_seen":"2022-11-14T09:44:29.165Z","linkquality":244,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:29MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"","battery":100,"brightness":90,"last_seen":"2022-11-14T09:44:29.165Z","linkquality":244,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:29MQTT publish: topic 'zigbee2mqtt/Symfonisk/action', payload 'brightness_move_up'
Debug 2022-11-14 10:44:29Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":195}' from endpoint 1 with groupID null
Debug 2022-11-14 10:44:29Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":1,"rate":195}' from endpoint 1 with groupID null
Info 2022-11-14 10:44:29MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"brightness_move_up","action_rate":195,"battery":100,"brightness":75,"last_seen":"2022-11-14T09:44:29.537Z","linkquality":255,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:29MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"","battery":100,"brightness":75,"last_seen":"2022-11-14T09:44:29.537Z","linkquality":255,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:29MQTT publish: topic 'zigbee2mqtt/Symfonisk/action', payload 'brightness_move_up'
Info 2022-11-14 10:44:29MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":null,"action_rate":195,"battery":100,"brightness":60,"last_seen":"2022-11-14T09:44:29.537Z","linkquality":255,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:30MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":null,"battery":100,"brightness":45,"last_seen":"2022-11-14T09:44:29.537Z","linkquality":255,"update":{"state":null},"update_available":null}'
Debug 2022-11-14 10:44:30Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":195}' from endpoint 1 with groupID null
Info 2022-11-14 10:44:30MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"brightness_move_up","action_rate":195,"battery":100,"brightness":30,"last_seen":"2022-11-14T09:44:30.259Z","linkquality":248,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:30MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"","battery":100,"brightness":30,"last_seen":"2022-11-14T09:44:30.259Z","linkquality":248,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:30MQTT publish: topic 'zigbee2mqtt/Symfonisk/action', payload 'brightness_move_up'
Debug 2022-11-14 10:44:30Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":1,"rate":195}' from endpoint 1 with groupID null
Info 2022-11-14 10:44:30MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"brightness_move_down","action_rate":195,"battery":100,"brightness":45,"last_seen":"2022-11-14T09:44:30.679Z","linkquality":236,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:30MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"","battery":100,"brightness":45,"last_seen":"2022-11-14T09:44:30.679Z","linkquality":236,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:30MQTT publish: topic 'zigbee2mqtt/Symfonisk/action', payload 'brightness_move_down'
Debug 2022-11-14 10:44:30Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":195}' from endpoint 1 with groupID null
Debug 2022-11-14 10:44:30Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":1,"rate":195}' from endpoint 1 with groupID null
Info 2022-11-14 10:44:30MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"brightness_move_up","action_rate":195,"battery":100,"brightness":30,"last_seen":"2022-11-14T09:44:31.031Z","linkquality":255,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:30MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"","battery":100,"brightness":30,"last_seen":"2022-11-14T09:44:31.031Z","linkquality":255,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:30MQTT publish: topic 'zigbee2mqtt/Symfonisk/action', payload 'brightness_move_up'
Info 2022-11-14 10:44:31MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":null,"action_rate":195,"battery":100,"brightness":15,"last_seen":"2022-11-14T09:44:31.031Z","linkquality":255,"update":{"state":null},"update_available":null}'
Debug 2022-11-14 10:44:32Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":195}' from endpoint 1 with groupID null
Info 2022-11-14 10:44:32MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"brightness_move_up","action_rate":195,"battery":100,"brightness":0,"last_seen":"2022-11-14T09:44:32.225Z","linkquality":236,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:32MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"","battery":100,"brightness":0,"last_seen":"2022-11-14T09:44:32.225Z","linkquality":236,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:32MQTT publish: topic 'zigbee2mqtt/Symfonisk/action', payload 'brightness_move_up'
Debug 2022-11-14 10:44:32Received Zigbee message from 'Symfonisk', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":1,"rate":195}' from endpoint 1 with groupID null
Info 2022-11-14 10:44:32MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"brightness_move_down","action_rate":195,"battery":100,"brightness":15,"last_seen":"2022-11-14T09:44:32.445Z","linkquality":244,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:32MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"","battery":100,"brightness":15,"last_seen":"2022-11-14T09:44:32.445Z","linkquality":244,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:32MQTT publish: topic 'zigbee2mqtt/Symfonisk/action', payload 'brightness_move_down'
[... cut ...]
Info 2022-11-14 10:44:33MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"brightness_stop","battery":100,"brightness":0,"last_seen":"2022-11-14T09:44:33.560Z","linkquality":252,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:33MQTT publish: topic 'zigbee2mqtt/Symfonisk', payload '{"action":"","battery":100,"brightness":0,"last_seen":"2022-11-14T09:44:33.560Z","linkquality":252,"update":{"state":null},"update_available":null}'
Info 2022-11-14 10:44:33MQTT publish: topic 'zigbee2mqtt/Symfonisk/action', payload 'brightness_stop'

jakemauer, 2023-01-15:
I know this is stale but just checking in to say that I have this exact same issue. 

Z2M Version: 1.29.2
IKEA Symfonisk E1744 firmware: 2.3.080
Coordinator: Sonoff P USB

KaHeMu, 2023-01-27:
I have exactly the same problem with this device using Z2M Version 1.29. It used to work fine with an earlier version (1.28 or older). 

Z2M Version: 1.29
IKEA Symfonisk E1744 firmware: 2.3.080
Coordinator: zStack3x0, FW revision 20220928  (CC2652RB Stick)

Shall I open a new issue?

KaHeMu, 2023-05-12:
Unfortunately I could not find a solution. I decided to replace the device with a different type - even though I find the Symfonisk E1744 a very nice and well designed gadget. But no use if it doesn't work properly.

jpleite, 2023-05-12:
Thanks @KaHeMu . BTW which device are you using? I use the E1744 to control Sonos

KaHeMu, 2023-05-12:
I used to use the E1744 to control my Onkyo Transceiver via LAN and Node-Red with an ISCP extension.

vaibhav2912, 2023-12-06:
> Unfortunately I could not find a solution. **I decided to replace the device with a different type** - even though I find the Symfonisk E1744 a very nice and well designed gadget. But no use if it doesn't work properly.

I have the same exact problem. Its very annoying.
@KaHeMu, what device you are using instead of E1744 ?

KaHeMu, 2023-12-10:
I'm using the Ikea E2002 'STYRBAR' instead. It is less intuitive because it has no turning wheel but it is heavy and solid and works reliably.

drsound, 2025-05-19:
Same issue here. I'm on latest firmware:
- Firmware build date: 20221019
- Firmware version: 24.4.5

Koenkk, 2025-11-30:
Can someone provide the debug log when triggering the up/down actions on z2m 2.6.3?

wutr, 2025-11-30:
> Can someone provide the debug log when triggering the up/down actions on z2m 2.6.3?

This is the log of me doing the following with the Rotary Symfonisk E1744 on firmware `24.4.5 20221019`.

- Turn slowly clockwise through two full rotations
- Pause for about 5 seconds
- Turn slowly counter clockwise through two full rotations

I _think_ I have an unopened, unused E1744 somewhere that's still on the original firmware, let me know if I need to dig that out.

```
[11/30/2025, 6:27:54 PM] frontend:api:bridge: Sending {"topic":"bridge/request/options","payload":{"options":{"advanced":{"log_level":"debug"}},"transaction":"d90l8-1"}}

[11/30/2025, 6:27:54 PM] z2m: Successfully changed options

[11/30/2025, 6:27:54 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"d90l8-1"}'

[11/30/2025, 6:28:01 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action":"brightness_move_up","action_rate":195,"battery":50,"brightness":255,"identify":null,"linkquality":208,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:01 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer/action', payload 'brightness_move_up'

[11/30/2025, 6:28:01 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action_brightness_delta":20,"battery":50,"brightness":255,"identify":null,"linkquality":208,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:01 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action":"brightness_move_down","action_rate":195,"battery":50,"brightness":255,"identify":null,"linkquality":208,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:01 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer/action', payload 'brightness_move_down'

[11/30/2025, 6:28:01 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action":"brightness_move_up","action_rate":195,"battery":50,"brightness":255,"identify":null,"linkquality":204,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:01 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer/action', payload 'brightness_move_up'

[11/30/2025, 6:28:01 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action_brightness_delta":20,"battery":50,"brightness":255,"identify":null,"linkquality":204,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:02 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action":"brightness_move_down","action_rate":195,"battery":50,"brightness":255,"identify":null,"linkquality":204,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:02 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer/action', payload 'brightness_move_down'

[11/30/2025, 6:28:02 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action":"brightness_move_up","action_rate":195,"battery":50,"brightness":255,"identify":null,"linkquality":208,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:02 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer/action', payload 'brightness_move_up'

[11/30/2025, 6:28:02 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action_brightness_delta":20,"battery":50,"brightness":255,"identify":null,"linkquality":208,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:03 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action":"brightness_move_down","action_rate":195,"battery":50,"brightness":255,"identify":null,"linkquality":208,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:03 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer/action', payload 'brightness_move_down'

[11/30/2025, 6:28:03 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action":"brightness_move_up","action_rate":195,"battery":50,"brightness":255,"identify":null,"linkquality":208,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:03 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer/action', payload 'brightness_move_up'

[11/30/2025, 6:28:03 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action_brightness_delta":20,"battery":50,"brightness":255,"identify":null,"linkquality":208,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[... cut ...] n":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:15 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer', payload '{"action":"brightness_stop","battery":50,"brightness":255,"identify":null,"linkquality":208,"rate":195,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'

[11/30/2025, 6:28:15 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button.bedroom_dimmer/action', payload 'brightness_stop'
```

andrei-lazarov, 2025-11-30:
Hi! I bought a SYMFONISK dial off marketplace (seems new - unused) a few days ago, so I can also test if needed. 

I updated it to `24.4.5 20221019`, interviewed, reconfigured, and bound it to an IKEA bulb. 
I'm on Z2M 2.6.3, SONOFF Dongle-E 8.0.2.

**I couldn't replicate the issue so far.** I don't see the opposite action at all. I will observe the next few days.
It controls the light as expected and reports the correct actions to Z2M:
- spin cw = brightness_move_up + brightness_stop
- spin ccw = brightness_move_down + brightness_stop

(once if I go fast, multiple times if I go slow)

@wutr **Did you attempt any troubleshooting steps?** 
(interview, reconfigure, reset/remove, re-pair, fresh battery)
Also how are you using the device? Can you try a bind and see if the problem persists there?

---

Sidenote: I want to update its [doc](https://www.zigbee2mqtt.io/devices/E1744.html#ikea-e1744)
- Can't bind to groups on latest fw (as far as I can tell)
- Cluster issue is fixed on latest fw: you need to bind levelCtrl for dimming and onOff for toggling
- Legacy info may be outdated?? How do I check if 'legacy is enabled by default'?

closed 2026-02-07 by github-actions[bot]