Koenkk/zigbee2mqtt #14981: SEA802: Calibration of the local temperature only allows integer values

ewolf29, 2022-11-14:
### What happened?

It's not possible to change the local_temperature_calibration using float/decimal values. It only takes integer/whole values. Hower, a step size of 1 makes no sense for temperature calibration.

### What did you expect to happen?

It should be possible to set local_temperature_calibration in 0.1 steps.

### How to reproduce it (minimal and precise)

Set local_temperature_calibration to 0.1, and it won't change; set it to 1, and it will change.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

ewolf29, 2022-11-14:
It seems like the step size was changed in commit [54d8241f491c5ed3e28eb36b91803b71005ebbfb](https://github.com/Koenkk/zigbee-herdsman-converters/commit/54d8241f491c5ed3e28eb36b91803b71005ebbfb) according to #11777 - but the discussion does not mention a reason for that. It actually reads more like a misunderstanding.

Can you please change it back? I can test it on the dev branch.

jjansen85, 2022-11-20:
You can add an external converter with the following content:

```js
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_c88teujp'}],
        model: 'SEA801-Zigbee/SEA802-Zigbee',
        vendor: 'Saswell',
        description: 'Thermostatic radiator valve',
        whiteLabel: [{vendor: 'HiHome', model: 'WZB-TRVL'}, {vendor: 'Hama', model: '00176592'},
            {vendor: 'RTX', model: 'ZB-RT1'}],
        fromZigbee: [fz.saswell_thermostat, fz.ignore_tuya_set_time, fz.ignore_basic_report, fz.legacy.tuya_thermostat_weekly_schedule],
        toZigbee: [tz.saswell_thermostat_current_heating_setpoint, tz.saswell_thermostat_mode, tz.saswell_thermostat_away,
            tz.saswell_thermostat_child_lock, tz.saswell_thermostat_window_detection, tz.saswell_thermostat_frost_detection,
            tz.saswell_thermostat_calibration, tz.saswell_thermostat_anti_scaling, tz.tuya_thermostat_weekly_schedule],
        onEvent: tuya.onEventSetTime,
        meta: {
            thermostat: {
                weeklyScheduleMaxTransitions: 4,
                weeklyScheduleSupportedModes: [1], // bits: 0-heat present, 1-cool present (dec: 1-heat,2-cool,3-heat+cool)
                weeklyScheduleConversion: 'saswell',
            },
        },
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
        },
        exposes: [e.battery_low(), e.window_detection(), e.child_lock(), e.away_mode(),
            exposes.binary('heating', ea.STATE, 'ON', 'OFF').withDescription('Device valve is open or closed (heating or not)'),
            exposes.climate()
                .withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET).withLocalTemperature(ea.STATE)
                .withSystemMode(['off', 'heat', 'auto'], ea.STATE_SET)
                // Range is -6 - 6 and step 1: https://github.com/Koenkk/zigbee2mqtt/issues/11777
                .withLocalTemperatureCalibration(-6, 6, 0.1, ea.STATE_SET)],
    },
];
```

I tried using 0.1 or 0.5 as step change. Does not work for me. Devices give only integers back... unfortuntatelly

ewolf29, 2022-11-20:
Thanks for the information. That's actually nice to know. Where do I have to add the file?

Indeed, the device seems hardware-wise not to be the best. The current temp also has a bad resolution, it usually changes only in 0.6 steps. It's probably best to add another external room temp sensor and use the TRV only to control the heater valve...

jjansen85, 2022-11-20:
I add it in the same folder as configuration.yaml, then add it via the settings, submit + restart.

[image]

Correct, I don't like the device... It also has a too high setpoint vs actual temp deviation. I suspect it to be about 2 degC

ewolf29, 2022-11-20:
Yep - I also did some tests and approximated it to be between 1.8 and 1.9 degrees. In fact, when you set the setpoint to 22 degrees, it will never reach 22 degrees but closes the valve much earlier. When you know about that fact, it's possible to cope with it by adding offsets in the automation. But still absolutely not nice. It would be cool to find a way to just control the valve value.

andercarreno, 2022-12-17:
Hi! I am also interested on this. I have a bunch of SNZB-02 Sonoff temperature sensors and I would like to fine tune the temperature that these are reporting. They are 0.6ºC off. I have tried to put a decimal value in configuration.yaml but as the OP has already reported, it changes to an integer value of 1.

jhanninen, 2023-01-24:
I'm not sure how device specific problem this is, but for Philips Hue Motion Sensors I managed to use decimal values by modifying html code. I checked how the frontend constructs the form, and there should be support for the decimals (https://github.com/nurikk/zigbee2mqtt-frontend/blob/dev/src/components/range-editor/range-editor.tsx#L43) `valueStep` property is probably missing, but I'm not quite sure where and how it should be implemented. Probably somehow @jeroen85 wrote in their converter example.

Anyway, here are the steps I made to modify my calibrations:
1. Open the device page where the calibration setting form exists.
2. Open browser's developer panel, activate the inspector, and click the calibration field.
3. html code tree will be opened where you can see the html element of the field. It looks something like:
```html
<input type="number" class="form-control" value="">
```
4. Modify the element and give step-parameter
```html
<input type="number" class="form-control" value="" step="0.1">
```
5. Now the browser accepts decimals to the field and they will be updated to the device configuration.

Note that changes to html code will be removed after you refresh the page or move away from it.

[image]

closed 2023-03-03 by github-actions[bot]

spraot, 2024-01-22:
@jhanninen 's changes should be applied to the project, the input is difficult to use as is