Koenkk/zigbee2mqtt #15000: Ketotek KTF0177 (SASWELL)

KimHEF, 2022-11-15:
### Link

https://www.amazon.de/KETOTEK-Heizk%C3%B6rperthermostat-Elektrische-Heizungsthermostat-Programmierbar/dp/B0B5GY1288/ref=sr_1_22?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=2OF4Q747I2N1R&keywords=zigbee+trv&qid=1668524836&sprefix=zigbee+tr%2Caps%2C93&sr=8-22

### Database entry

{"id":10,"type":"EndDevice","ieeeAddr":"0x003c84fffef73309","nwkAddr":9403,"manufId":4098,"manufName":"_TZE200_bvu2wnxz","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�E\u0006+f�E\u0006+\u0012�E\u0006+\u0012","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_bvu2wnxz","stackVersion":0,"appVersion":65,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d91fe5","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1668524582594,"defaultSendRequestWhen":"immediate"}

### Comments

I want to add this new TRV to z2m. I have never done this before, so I copied the saswell.js and combined it with the instruction code for TuYa devices from the z2m manual. After I removed the "legacy.tuya_thermostat_weekly_schedule" it showed up on the z2m page with name. But when I do any actions on the device, I always get messages like this:
`zigbee-herdsman-converters:SaswellThermostat: Unrecognized DP #4 with data {"dp":4, "datatype":2, "data":{"type": "Buffer", "data":[0,0,0,90]}}`
What could I do next to solve this problem?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

module.exports = [
    {
        fingerprint: [{modelID: 'GbxAXL2\u0000', manufacturerName: '_TYST11_KGbxAXL2'},
            {modelID: 'uhszj9s\u0000', manufacturerName: '_TYST11_zuhszj9s'},
            {modelID: '88teujp\u0000', manufacturerName: '_TYST11_c88teujp'},
            {modelID: 'w7cahqs\u0000', manufacturerName: '_TYST11_yw7cahqs'},
            {modelID: 'w7cahqs', manufacturerName: '_TYST11_yw7cahqs'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_c88teujp'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_yw7cahqs'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_azqp6ssj'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_zuhszj9s'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_9gvruqf5'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_zr9c0day'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_0dvm9mva'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_h4cgnbzg'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_bvu2wnxz'},      
        ],
        model: 'SEA801-Zigbee/SEA802-Zigbee',
        vendor: 'Saswell',
        description: 'Thermostatic radiator valve',
        whiteLabel: [{vendor: 'HiHome', model: 'WZB-TRVL'}, {vendor: 'Hama', model: '00176592'},
            {vendor: 'RTX', model: 'ZB-RT1'}],
        fromZigbee: [fz.saswell_thermostat, fz.ignore_tuya_set_time, fz.ignore_basic_report],
        toZigbee: [tz.saswell_thermostat_current_heating_setpoint, tz.saswell_thermostat_mode, tz.saswell_thermostat_away,
            tz.saswell_thermostat_child_lock, tz.saswell_thermostat_window_detection, tz.saswell_thermostat_frost_detection,
            tz.saswell_thermostat_calibration, tz.saswell_thermostat_anti_scaling],
        onEvent: tuya.onEventSetTime,
        meta: {
            thermostat: {
                weeklyScheduleMaxTransitions: 4,
                weeklyScheduleSupportedModes: [1], // bits: 0-heat present, 1-cool present (dec: 1-heat,2-cool,3-heat+cool)
                weeklyScheduleConversion: 'saswell',
            },
        },
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
        },
        exposes: [e.battery_low(), e.window_detection(), e.child_lock(), e.away_mode(),
            exposes.binary('heating', ea.STATE, 'ON', 'OFF').withDescription('Device valve is open or closed (heating or not)'),
            exposes.climate()
                .withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET).withLocalTemperature(ea.STATE)
                .withSystemMode(['off', 'heat', 'auto'], ea.STATE_SET)
                // Range is -6 - 6 and step 1: https://github.com/Koenkk/zigbee2mqtt/issues/11777
                .withLocalTemperatureCalibration(-6, 6, 1, ea.STATE_SET)],
    },
];
```

kombasltd, 2022-11-18:
Bought 4 of them today and would love to make them running as well. At the moment found as model "TS0601" and "online" but "unsupported" in Z2M.

Althir1, 2022-12-01:
Same here!
Home Assistant, Zigbee2Mqtt Edge latest version

2022-12-01 16:25:28Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_bvu2wnxz'

Database entry:
{"id":46,"type":"EndDevice","ieeeAddr":"0x003c84fffe267e2c","nwkAddr":16019,"manufId":4098,"manufName":"_TZE200_bvu2wnxz","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000f\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE200_bvu2wnxz","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1669908328472,"defaultSendRequestWhen":"immediate"}

adlerweb, 2022-12-13:
Duplicate of #14556 - support for bvu2wnxz should now be available in dev: https://github.com/Koenkk/zigbee-herdsman-converters/commit/9d8cbc88f2328893d18d30d6996978d2effa015f

Koenkk, 2022-12-13:
`_TZE200_bvu2wnxz` is supported in the latest dev branch now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-13 by Koenkk