Koenkk/zigbee2mqtt #15006: Gledopto RGBCCT 2ID controller not supported despite docs saying it is?

liamlows, 2022-11-15:
### What happened?

So i recently purchase a Gledopto 2ID RGBCCT controller and have had issues pairing it with zigbee2mqtt. On initial adoption of the device, it immediately thought that the device had the model number `GL-C-007` and said it was unsupported. I checked the docs for `GL-C-007` and there was no model number for that device there so I wasn't surprised it wasn't supported but variants of it did exist.

I then changed the model number of the device in the database.db to `GL-C-008-2ID` as that was the model number listed in the docs for the device I had purchased. Despite the docs stating compatibility, after making the change, zigbee2mqtt stated it does not support that device.

As a last test, I changed the model of the device in the database.db to `GL-C-008` and it managed to work but only supported changing the RGB channels. The CW and WW channels were unchangeable and stuck at 100% power essentially making them useless.

**I will note that I am using the newer Sonoff USB v2 dongle which uses the Silicon Labs chip as opposed to the older one that used the Ti chip. This may be the cause of some problems.** I have the Ti version of the dongle arriving soon.

Below is my database.db entry for the device and some notes, anyone know what is going on and if this device is actually supported?

**RGBCCT controller link:** 
https://www.amazon.com/gp/product/B07QYQRM47/ref=ppx_yo_dt_b_asin_title_o03_s02?ie=UTF8&th=1
**Sonoff dongle link:**
https://www.amazon.com/gp/product/B0B6P22YJC/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
https://sonoff.tech/product/gateway-amd-sensors/sonoff-zigbee-3-0-usb-dongle-plus-e/
**Zigbee2MQTT docs device link:**
https://www.zigbee2mqtt.io/devices/GL-C-008-2ID.html
**Database.db entry:**
```
{
  "id": 2,
  "type": "Router",
  "ieeeAddr": "0x00124b0021efbe1a",
  "nwkAddr": 38241,
  "manufId": 0,
  "manufName": "GLEDOPTO",
  "powerSource": "Mains (single phase)",
  "modelId": "GL-C-008-2ID",
  "epList": [11, 13, 15],
  "endpoints": {
    "11": {
      "profId": 49246,
      "epId": 11,
      "inClusterList": [0, 3, 4, 5, 6, 8, 768],
      "outClusterList": [],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "GL-C-008-2ID",
            "manufacturerName": "GLEDOPTO",
            "powerSource": 1,
            "zclVersion": 1,
            "appVersion": 1,
            "stackVersion": 2,
            "hwVersion": 1,
            "dateCode": "20180920",
            "swBuildId": "2.0.4"
          }
        },
        "genLevelCtrl": { "attributes": { "currentLevel": 254 } },
        "genOnOff": { "attributes": { "onOff": 1 } },
        "lightingColorCtrl": {
          "attributes": {
            "colorMode": 2,
            "currentX": 0,
            "currentY": 0,
            "enhancedCurrentHue": 6400,
            "currentSaturation": 251,
            "colorTemperature": 500
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "13": {
      "profId": 49246,
      "epId": 13,
      "inClusterList": [4096],
      "outClusterList": [4096],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "15": {
      "profId": 49246,
      "epId": 15,
      "inClusterList": [0, 3, 4, 5, 6, 8, 768],
      "outClusterList": [],
      "clusters": { "lightingColorCtrl": { "attributes": {} } },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 1,
  "stackVersion": 2,
  "hwVersion": 1,
  "dateCode": "20180920",
  "swBuildId": "2.0.4",
  "zclVersion": 1,
  "interviewCompleted": true,
  "meta": { "configured": 2 },
  "lastSeen": 1668491951545,
  "defaultSendRequestWhen": "immediate"
}
```

### What did you expect to happen?

I expected the device, Gledopto RGBCCT 2ID GL-C-008-2ID, listed as a compatible device to properly pair with zigbee2mqtt for regular usage and control.

### How to reproduce it (minimal and precise)

1. enter pairing mode and adopt the device
2. see that is pairs as model number `GL-C-007`
3. adjust database.db entry to make the model number `GL-C-008-2ID` in two locations
4. see that device is not supported despite documentation stating it is

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Zigbee 3.0 USB Dongle Plus-E V2

### Debug log

Log of adoption that failed:
```
info  2022-11-14 23:34:57: Zigbee: allowing new devices to join.
info  2022-11-14 23:34:57: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"ej0r0-1"}'
info  2022-11-14 23:35:05: Device '0x00124b0021efbe1a' joined
info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021efbe1a","ieee_address":"0x00124b0021efbe1a"},"type":"device_joined"}'
info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b0021efbe1a"},"type":"device_connected"}'
info  2022-11-14 23:35:05: Starting interview of '0x00124b0021efbe1a'
info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021efbe1a","ieee_address":"0x00124b0021efbe1a","status":"started"},"type":"device_interview"}'
info  2022-11-14 23:35:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b0021efbe1a"},"type":"pairing"}'
info  2022-11-14 23:35:26: Successfully interviewed '0x00124b0021efbe1a', device has successfully been paired
warn  2022-11-14 23:35:26: Device '0x00124b0021efbe1a' with Zigbee model 'GL-C-007' and manufacturer name 'GLEDOPTO' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-11-14 23:35:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x00124b0021efbe1a","ieee_address":"0x00124b0021efbe1a","status":"successful","supported":false},"type":"device_interview"}'
info  2022-11-14 23:35:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x00124b0021efbe1a","supported":false},"type":"pairing"}'
info  2022-11-14 23:36:41: Zigbee: disabling joining new devices.
info  2022-11-14 23:36:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"ej0r0-2"}'
info  2022-11-14 23:43:04: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
```

liamlows, 2022-11-18:
Ok so after looking into this more I think I have identified the issue. 

Essentially it boiled down to the dongle I was using and the weird way that gledopto devices broadcast themselves.

After receiving the **Sonoff USB Dongle _Plus P v1_** which uses the **Ti chip** instead of the Silicon Labs chip I attempted to pair the device again and it worked with no issue. The device is being identified as `GL-C-007` which is understandable given the issue filed back in 2020 (https://github.com/Koenkk/zigbee2mqtt/issues/3525) however the addition of the code mentioned in the issue allows zigbee2mqtt to correlate that to the proper `GL-C-008-2ID` which can be seen in the PR that was made (https://github.com/Koenkk/zigbee-herdsman-converters/commit/ca0d5e5f457d6e097a0bfa765a60f766c5fca01e)

I'm not sure why this occurs, I can confidently say I am very new to the smart home world and zigbee/zigbee2mqtt but it seems there is an issue with the way the dongle handles the connection with the gledopto controller. I have seen numerous posts stating issues with using the new V2 Silicon Labs chip dongle supporting devices and I assume this is the same thing those posts were referring to.

@Koenkk let me know if you would like any additional information as you were the last I can find that handled this issue in the past. I'm happy to help in any way but I have limited knowledge in this field however I am eager to learn. 😄 

Link to the Ti Sonoff Zigbee Dongle: https://sonoff.tech/product/gateway-amd-sensors/sonoff-zigbee-3-0-usb-dongle-plus-p/

Note: I also ordered a second Silicon Labs based dongle (Plus E v2) and attempted to pair with it again after clearing out all the zigbee2mqtt data and found the same error occurred

liamlows, 2022-11-18:
Update: the plot continues to thicken and I have found a workaround!

So after comparing the database.db entry that was generated for the Gledopto RGBCCT 2ID controller when using the **Plus P v1** dongle I noticed it differed from what the **Plus E v2** had generated. So I wiped all the settings in the zigbee2mqtt configuration again and **swapped back to the Plus E v2 dongle**. I started zigbee2mqtt and allowed the coordinator to set up and re-initialize the configuration and then shutdown zigbee2mqtt. 

I then **manually added the database.db entry from when the Plus P v1 paired with the device** and restarted zigbee2mqtt. **This time, when using the Plus E v2 dongle the addon started and recognized the device with the correct configuration**.

Again not sure what is going on here or what to do moving forward. For now the device works great with the Plus E v2 dongle and the modified database.db but I'm worried I might see issues in the future if i continue to use the Plus E v2 dongle with other zigbee devices. Any thoughts on this?

@Koenkk is the support for the v2 dongle still in development for some devices?

Here is some config information:

Sonoff USB Dongle Plus P v1 database.db:
```
{
    "id": 1,
    "type": "Coordinator",
    "ieeeAddr": "0x00124b00257cb193",
    "nwkAddr": 0,
    "manufId": 0,
    "epList": [
        242,
        47,
        13,
        12,
        110,
        11,
        10,
        8,
        6,
        5,
        4,
        3,
        2,
        1
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 5,
            "inClusterList": [],
            "outClusterList": [],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
        "2": {
            "profId": 257,
            "epId": 2,
            "devId": 5,
            "inClusterList": [],
            "outClusterList": [],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
        "3": {
            "profId": 260,
            "epId": 3,
            "devId": 5,
            "inClusterList": [],
            "outClusterList": [],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
[... 383 lines cut ...]
                4,
                5,
                6,
                8,
                768
            ],
            "outClusterList": [],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 1,
    "stackVersion": 2,
    "hwVersion": 1,
    "dateCode": "20180920",
    "swBuildId": "2.0.4",
    "zclVersion": 1,
    "interviewCompleted": true,
    "meta": {},
    "lastSeen": 1668736160327,
    "defaultSendRequestWhen": "immediate"
}
```

Koenkk, 2022-11-20:
I see that for the EZSP adapter (v2) the `deviceID` is missing.  @kirovilya is this a known issue?

kirovilya, 2022-11-20:
> I see that for the EZSP adapter (v2) the `deviceID` is missing. @kirovilya is this a known issue?

bug. found!

Koenkk, 2022-11-20:
Great, @liamlows please check if this is solved now by:

- updating to the latest dev
- force remove the device through z2m frontend
- restart z2m
- pair again.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

liamlows, 2022-11-21:
@kirovilya thanks for finding that!

@Koenkk I am away for the holidays but I will test that dev branch as soon as im back. I assume ill have to use the edge build to try this? Can I just shutdown z2m, add the edge addon, remove devices and clean wipe, and then start the edge version and give it a go? Or is there a different process to obtaining latest dev?

Koenkk, 2022-11-21:
@liamlows next release is on 1 December so depends on when you get back.

liamlows, 2022-11-21:
> @liamlows next release is on 1 December so depends on when you get back.

I should be able to test and provide confirmation on 11/26 by EOD 😄

liamlows, 2022-11-21:
@Koenkk was able to test it out quickly before leaving today. Looks good and works as expected on dev. 

I cleared out the zigbee2mqtt config folder, installed edge, ran it, shut it down and configured with the dongle/mqtt broker, then turned back on and attempted to pair the device. Worked like a charm!

let me know if you want anything else, here is log of join and database file after join:
```
info  2022-11-21 14:18:06: Device '0x00124b0021f1c180' joined
info  2022-11-21 14:18:06: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021f1c180","ieee_address":"0x00124b0021f1c180"},"type":"device_joined"}'
info  2022-11-21 14:18:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b0021f1c180"},"type":"device_connected"}'
info  2022-11-21 14:18:06: Starting interview of '0x00124b0021f1c180'
info  2022-11-21 14:18:06: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021f1c180","ieee_address":"0x00124b0021f1c180","status":"started"},"type":"device_interview"}'
info  2022-11-21 14:18:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b0021f1c180"},"type":"pairing"}'
info  2022-11-21 14:18:12: Zigbee: allowing new devices to join.
info  2022-11-21 14:18:12: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"4uqb0-1"}'
info  2022-11-21 14:18:29: Successfully interviewed '0x00124b0021f1c180', device has successfully been paired
info  2022-11-21 14:18:29: Device '0x00124b0021f1c180' is supported, identified as: Gledopto Zigbee LED Controller RGB+CCT (2 ID) (GL-C-008-2ID)
info  2022-11-21 14:18:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee LED Controller RGB+CCT (2 ID)","exposes":[{"endpoint":"rgb","features":[{"access":7,"description":"On/off state of this light","endpoint":"rgb","name":"state","property":"state_rgb","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","endpoint":"rgb","name":"brightness","property":"brightness_rgb","type":"numeric","value_max":254,"value_min":0},{"description":"Color of this light in the CIE 1931 color space (x/y)","endpoint":"rgb","features":[{"access":7,"name":"x","property":"x","type":"numeric"},{"access":7,"name":"y","property":"y","type":"numeric"}],"name":"color_xy","property":"color_rgb","type":"composite"}],"type":"light"},{"endpoint":"cct","features":[{"access":7,"description":"On/off state of this light","endpoint":"cct","name":"state","property":"state_cct","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","endpoint":"cct","name":"brightness","property":"brightness_cct","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","endpoint":"cct","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp_cct","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"access":7,"description":"Color temperature after cold power on of this light","endpoint":"cct","name":"color_temp_startup","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500},{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup_cct","type":"numeric","unit":"mired","value_max":500,"value_min":150}],"type":"light"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"GL-C-008-2ID","options":[{"access":2,"description":"Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to `0` (no transition).","name":"transition","property":"transition","type":"numeric","value_min":0},{"access":2,"description":"When enabled colors will be synced, e.g. if the light supports both color x/y and color temperature a conversion from color x/y to color temperature will be done when setting the x/y color (default true).","name":"color_sync","property":"color_sync","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Gledopto"},"friendly_name":"0x00124b0021f1c180","ieee_address":"0x00124b0021f1c180","status":"successful","supported":true},"type":"device_interview"}'
info  2022-11-21 14:18:29: Configuring '0x00124b0021f1c180'
[... cut ...]
        64514
[... 106 lines cut ...]
      "outClusterList": [],
      "clusters": {
        "lightingColorCtrl": {
          "attributes": {}
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 1,
  "stackVersion": 2,
  "hwVersion": 1,
  "dateCode": "20180920",
  "swBuildId": "2.0.4",
  "zclVersion": 1,
  "interviewCompleted": true,
  "meta": {
    "configured": 2
  },
  "lastSeen": 1669061909429,
  "defaultSendRequestWhen": "immediate"
}
```

Koenkk, 2022-11-22:
Great, then this can be closed!

closed 2022-11-22 by Koenkk