Koenkk/zigbee2mqtt #15008: Regression in the dev branch – Failure send version

jirutka, 2022-11-15:
### What happened?

I’ve been testing c2e95bdb6a84ac4c13e136fbeedac9efbc8882bb (dev), which worked for me (after I explicitly set `adapter: ezsp`). Now I tried the latest dev (16398b7810d2eb2c405ed41f9d41ae2134de7f01), and it fails to start. The error message is either:

```
Zigbee2MQTT:info  2022-11-15 21:41:41: Starting Zigbee2MQTT version 1.28.2-dev (commit #null)
Zigbee2MQTT:info  2022-11-15 21:41:41: Starting zigbee-herdsman (0.14.73)

/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/uart.js:278
            throw new TypeError("reset can only be called on a new connection");
                  ^
TypeError: reset can only be called on a new connection
    at SerialDriver.reset (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/uart.js:278:19)
    at SerialDriver.onParsed (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/uart.js:186:26)
    at Parser.emit (node:events:513:28)
    at Parser.parseNext (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/parser.js:60:26)
    at Parser.parseNext (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/parser.js:66:18)
    at Parser.parseNext (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/parser.js:66:18)
    at Parser.parseNext (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/parser.js:66:18)
    at Parser._transform (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/parser.js:51:14)
    at Parser.Transform._write (node:internal/streams/transform:205:23)
    at writeOrBuffer (node:internal/streams/writable:391:12)
```

or

```
Zigbee2MQTT:info  2022-11-15 21:49:47: Starting Zigbee2MQTT version 1.28.2-dev (commit #null)
Zigbee2MQTT:info  2022-11-15 21:49:47: Starting zigbee-herdsman (0.14.73)
Zigbee2MQTT:error 2022-11-15 21:49:59: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-11-15 21:49:59: Failed to start zigbee
Zigbee2MQTT:error 2022-11-15 21:49:59: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-11-15 21:49:59: Exiting...
Zigbee2MQTT:error 2022-11-15 21:49:59: Error: Failure send version:{"type":"Buffer","data":[1,0,1,0,0,8]}
    at /usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/ezsp.js:444:23
    at Queue.executeNext (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/queue.js:19:32)
```

I’ll provide more info later.

### Zigbee2MQTT version

dev (16398b7810d2eb2c405ed41f9d41ae2134de7f01)

### Adapter firmware version

(I’ll provide later)

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus E

kirovilya, 2022-11-16:
@jirutka Can you record herdsman log? https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging

senin24, 2022-11-23:
> @jirutka Can you record herdsman log? https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging

@kirovilya I have some problem ("Error: Failure send version:{"type":"Buffer","data":[1,0,1,0,0,8]}")

Thi is debug log:

[11:21:20] INFO: Preparing to start...
[11:21:21] INFO: Socat not enabled
[11:21:21] INFO: Zigbee Herdsman debug logging enabled
[11:21:22] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2022-11-23 11:21:28: Can't load state from file /config/zigbee2mqtt/state.json (doesn't exist)
Zigbee2MQTT:info  2022-11-23 11:21:28: Logging to console and directory: '/config/zigbee2mqtt/log/2022-11-23.11-21-28' filename: log.txt
Zigbee2MQTT:debug 2022-11-23 11:21:28: Removing old log directory '/config/zigbee2mqtt/log/2022-11-22.23-27-10'
Zigbee2MQTT:info  2022-11-23 11:21:28: Starting Zigbee2MQTT version 1.28.2 (commit #unknown)
Zigbee2MQTT:info  2022-11-23 11:21:28: Starting zigbee-herdsman (0.14.68)
Zigbee2MQTT:debug 2022-11-23 11:21:28: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220810195553-if00"}}'
2022-11-23T08:21:29.446Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2022-11-23T08:21:29.454Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220810195553-if00","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-11-23T08:21:29.458Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220810195553-if00 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2022-11-23T08:21:29.474Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2022-11-23T08:21:29.475Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-11-23T08:21:29.475Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-11-23T08:21:29.478Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2022-11-23T08:21:34.387Z zigbee-herdsman:adapter:ezsp:uart <-- [0134a1a90c2a257d313d7e]
2022-11-23T08:21:34.389Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,1,0): 0134a1a90c2a25113d7e00
2022-11-23T08:21:34.390Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2022-11-23T08:21:34.391Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2022-11-23T08:21:34.392Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 1 | 0
2022-11-23T08:21:34.394Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 768001580030a3
2022-11-23T08:21:34.396Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x1: {"_id_":1,"_cls_":"getLibraryStatus","_isRequest_":false,"status":88}
2022-11-23T08:22:41.043Z zigbee-herdsman:adapter:ezsp:uart <-- [0142a1a85c2825d8e12b7e]
2022-11-23T08:22:41.044Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,1,0): 0142a1a85c2825d8e12b7e
2022-11-23T08:22:41.044Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2022-11-23T08:22:41.045Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2022-11-23T08:22:41.045Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 1 | 0
2022-11-23T08:22:41.046Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 0080000802306a
2022-11-23T08:22:41.047Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x0: {"_id_":0,"_cls_":"version","_isRequest_":false,"protocolVersion":8,"stackType":2,"stackVersion":27184}
2022-11-23T08:22:52.684Z zigbee-herdsman:adapter:ezsp:uart <-- [c1020b0a527e]
2022-11-23T08:22:52.685Z zigbee-herdsman:adapter:ezsp:uart <-- RSTACK: c1020b0a527e
2022-11-23T08:22:52.686Z zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e
2022-11-23T08:22:52.688Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":8}
2022-11-23T08:22:52.690Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":8}
2022-11-23T08:22:52.694Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 000001000008
2022-11-23T08:22:52.697Z zigbee-herdsman:adapter:ezsp:uart --> [004221a9542a1d03c57e]
2022-11-23T08:22:52.697Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
2022-11-23T08:22:52.703Z zigbee-herdsman:adapter:ezsp:uart <-- [0142a1a9542a1db069fe63a97e]
2022-11-23T08:22:52.704Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,1,0): 0142a1a9542a1db069fe63a97e
2022-11-23T08:22:52.704Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2022-11-23T08:22:52.704Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2022-11-23T08:22:52.707Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 00800100000802306a
2022-11-23T08:22:52.708Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x0: {"_id_":0,"_cls_":"version","_isRequest_":false,"protocolVersion":8,"stackType":2,"stackVersion":27184}
[... cut ...]
Zigbee2MQTT:error 2022-11-23 11:22:54: Failed to start zigbee
Zigbee2MQTT:error 2022-11-23 11:22:54: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-11-23 11:22:54: Exiting...
Zigbee2MQTT:error 2022-11-23 11:22:54: Error: Failure send setConfigurationValue:{"type":"Buffer","data":[1,0,1,83,0,13,5,0]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:512:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

kirovilya, 2022-11-23:
@senin24 try switching to dev/edge version of z2m

senin24, 2022-11-23:
> @senin24 try switching to dev/edge version of z2m

some problem:

[12:46:39] INFO: Preparing to start...
[12:46:39] INFO: Socat not enabled
[12:46:40] INFO: Zigbee Herdsman debug logging enabled
[12:46:41] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2022-11-23 12:46:47: Can't load state from file /config/zigbee2mqtt/state.json (doesn't exist)
Zigbee2MQTT:info  2022-11-23 12:46:47: Logging to console and directory: '/config/zigbee2mqtt/log/2022-11-23.12-46-47' filename: log.txt
Zigbee2MQTT:debug 2022-11-23 12:46:47: Removing old log directory '/config/zigbee2mqtt/log/2022-11-22.23-38-22'
Zigbee2MQTT:info  2022-11-23 12:46:47: Starting Zigbee2MQTT version 1.28.2-dev (commit #ca4badc)
Zigbee2MQTT:info  2022-11-23 12:46:47: Starting zigbee-herdsman (0.14.74)
Zigbee2MQTT:debug 2022-11-23 12:46:47: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220810195553-if00"}}'
2022-11-23T09:46:48.235Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2022-11-23T09:46:48.240Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220810195553-if00","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-11-23T09:46:48.245Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220810195553-if00 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2022-11-23T09:46:48.263Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2022-11-23T09:46:48.264Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-11-23T09:46:48.265Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-11-23T09:46:48.269Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2022-11-23T09:47:37.109Z zigbee-herdsman:adapter:ezsp:uart <-- [452aa1a90c2a2597307e]
2022-11-23T09:47:37.113Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,5,0): 452aa1a90c2a2597307e
2022-11-23T09:47:37.113Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (5)
2022-11-23T09:47:37.114Z zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e]
2022-11-23T09:47:37.116Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 5 | 0
2022-11-23T09:47:37.118Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 688001580030
2022-11-23T09:47:37.120Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x1: {"_id_":1,"_cls_":"getLibraryStatus","_isRequest_":false,"status":88}
2022-11-23T09:47:43.734Z zigbee-herdsman:adapter:ezsp:uart <-- [0142a1a85c2825d8e12b7e]
2022-11-23T09:47:43.735Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,1,0): 0142a1a85c2825d8e12b7e
2022-11-23T09:47:43.736Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2022-11-23T09:47:43.739Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2022-11-23T09:47:43.740Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 1 | 0
2022-11-23T09:47:43.741Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 0080000802306a
2022-11-23T09:47:43.743Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x0: {"_id_":0,"_cls_":"version","_isRequest_":false,"protocolVersion":8,"stackType":2,"stackVersion":27184}
2022-11-23T09:47:55.380Z zigbee-herdsman:adapter:ezsp:uart <-- [c1020b0a527e]
2022-11-23T09:47:55.381Z zigbee-herdsman:adapter:ezsp:uart <-- RSTACK: c1020b0a527e
2022-11-23T09:47:55.381Z zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e
2022-11-23T09:47:55.384Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":8}
2022-11-23T09:47:55.386Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":8}
2022-11-23T09:47:55.390Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 000001000008
2022-11-23T09:47:55.392Z zigbee-herdsman:adapter:ezsp:uart --> [004221a9542a1d03c57e]
2022-11-23T09:47:55.393Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
2022-11-23T09:47:57.393Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":1} after 2000ms
2022-11-23T09:47:57.394Z zigbee-herdsman:adapter:ezsp:uart -!- break waiting (1)
2022-11-23T09:47:57.394Z zigbee-herdsman:adapter:ezsp:uart Can't send DATA frame (0,0,0): 000001000008
2022-11-23T09:47:57.394Z zigbee-herdsman:adapter:ezsp:uart ->> DATA (0,0,1): 000001000008
2022-11-23T09:47:57.396Z zigbee-herdsman:adapter:ezsp:uart --> [084221a9542a1d90687e]
2022-11-23T09:47:57.396Z zigbee-herdsman:adapter:ezsp:uart -?- rewaiting (1)
2022-11-23T09:47:59.394Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":1} after 2000ms
2022-11-23T09:47:59.394Z zigbee-herdsman:adapter:ezsp:uart -!- break rewaiting (1)
[... cut ...]
Zigbee2MQTT:error 2022-11-23 12:47:59: Failed to start zigbee
Zigbee2MQTT:error 2022-11-23 12:47:59: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-11-23 12:47:59: Exiting...
Zigbee2MQTT:error 2022-11-23 12:47:59: Error: Failure send version:{"type":"Buffer","data":[0,0,1,0,0,8]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:516:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

Sorin-1, 2022-11-23:
Same issue here, non-edge Zigbee2MQTT version.
Zigbee2MQTT:error 2022-11-23 21:18:51: Error: Failure send version:{"type":"Buffer","data":[0,0,1,0,0,8]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:512:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

Zigbee2MQTT:error 2022-11-23 21:23:25: Exiting...
Zigbee2MQTT:error 2022-11-23 21:23:25: Error: Failure send setPolicy:{"type":"Buffer","data":[12,0,1,85,0,6,97,0]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:512:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

port: >-
  /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220816102251-if00
adapter: ezsp

senin24, 2022-11-23:
> strange. for some reason, communication with the port is interrupted. maybe some other software works in parallel with this port? ZHA?

thanks

found the reason: was activated ZHA (Zigbee Home Automation), after deactivated ZHA all works fine
[image]

jirutka, 2022-11-23:
In my case, I'm sure no other app was accessing the port. I'll try it again and log later.

hooverds54, 2022-12-18:
I am having exactly same issue as described above.  I hope someone can shed some light becuase I too have tried both the release and dev version and I am getting the exact same message.  I have tried unstailling and reinstalling and I have worked through the scenarios on the "Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions"  page with no success.  Here is a copy of my logging output.

[11:57:36] INFO: Preparing to start...
[11:57:36] INFO: Socat enabled
[11:57:37] INFO: Starting socat
Socat startup parameters:
Options:     -d -d
Master:      pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
Slave:       tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
[11:57:37] INFO: Starting socat process ...
2022/12/18 11:57:37 socat[53] N PTY is /dev/pts/0
2022/12/18 11:57:37 socat[53] N listening on AF=2 0.0.0.0:8485
[11:57:37] INFO: Zigbee Herdsman debug logging enabled
[11:57:39] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2022-12-18 11:57:43: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-12-18 11:57:43: Logging to console and directory: '/config/zigbee2mqtt/log/2022-12-18.11-57-43' filename: log.txt
Zigbee2MQTT:debug 2022-12-18 11:57:43: Removing old log directory '/config/zigbee2mqtt/log/2022-12-16.15-28-00'
Zigbee2MQTT:info  2022-12-18 11:57:43: Starting Zigbee2MQTT version 1.28.4-dev (commit #da44667)
Zigbee2MQTT:info  2022-12-18 11:57:43: Starting zigbee-herdsman (0.14.81)
Zigbee2MQTT:debug 2022-12-18 11:57:43: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812161518-if00"}}'
2022-12-18T16:57:43.563Z zigbee-herdsman:adapter:zStack:znp:error Failed to determine if path is valid: 'Error: ENOENT: no such file or directory, lstat '/dev/serial''
2022-12-18T16:57:43.584Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2022-12-18T16:57:43.586Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812161518-if00"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-12-18T16:57:43.587Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812161518-if00","baudRate":115200,"rtscts":false,"autoOpen":false}
Zigbee2MQTT:error 2022-12-18 11:57:43: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-12-18 11:57:43: Failed to start zigbee
Zigbee2MQTT:error 2022-12-18 11:57:43: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-12-18 11:57:43: Exiting...
Zigbee2MQTT:error 2022-12-18 11:57:43: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812161518-if00'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:76:22)
    at /app/node_modules/@serialport/stream/dist/index.js:112:18

Sorin-1, 2022-12-19:
......"serialPort":{"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812161518-if00"}}' 2022-12-18T16:57:43.563Z zigbee-herdsman:adapter:zStack:znp:error Failed to determine if path is valid: 'Error: ENOENT: no such file or directory, lstat '/dev/serial'' 
Definitely it is a configuration problem. Also, logging level is set too high )
Try to look for Your adapter port: Select Settings-System-Euipment _(lowest line)_- then press three dots in upper right corner, and selsct "All equipment" or smilar, frst line. In the list look for similar:
**ttyUSB0
/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_f8c0ee4b221aec11837f77e5f01c6278-if00-port0**
This "ttyUSB0" will be the port.
Then Your configuration.yaml in zigbee2mqtt dir has to be like;

external_converters: []
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.96:1883
  user: mqtt
  password: yourmqttpassword
serial:
  port: /dev/ttyUSB0
advanced:
  log_level: info
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  last_seen: ISO_8601
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
availability:
  active:
    timeout: 10
  passive:
    timeout: 750

Try just copy-paste my config.

hooverds54, 2022-12-19:
Thanks for the reply.  

 

Sadly I can say that I got farther and he only significant thing I did was add a line under serial which is 

 

adapter: ezsp

 

after that I realized I had the wrong user name fro the Mosquitto MQTT user but that is now fixed and Zigbee2MQTT seems to come up fine.  I only have one device and one coordinator stick as I am just starting out and while I can pair the device a “aqara Motion Sensor”  

 

I can not see that it Is connected to my coordinator in the network map. And it is not clear there is any communicator between the motion sensor and the Home assistant instance because the entites are fixed and have not changed states regardless of what I do.  And for instance the battery state is not reporting any %.

 

I had the same basic experience  using the ZHA integration because he device was recognized but no apparent upset of states over time. 

 

I guess it is possible it is a bad motion sensor but I don’t know how I would test that.

Sorin-1, 2022-12-20:
"And for instance the battery state is not reporting any %." - Battery % reading was always broken - even in stable conditions it works only from time to time.. Have You tried "Map" feature? Does it show connection and link quality?

hooverds54, 2022-12-20:
Yes first off let me thank you for at least replying.  It can be very frustrating figuring things out even if you have some working knowledge of the way things should work.  In this case I am new to zigbee protocols so it is especially frustrating. 

 

To answer your question when I bring up the MAP here is what I see.  I am guessing there is not a connection however here is the device screen also and if I am reading right the link quality is 255?

Sorin-1, 2023-01-20:
> Y here is what I see. I am guessing there is not a connection however here is the device screen also and if I am reading right the link quality is 255? 

Sorry, I can not see pictures in Your reply. Usially link quality rarely rech 255 (i have only 2 devices with "255" that are located directly near the coordinator antenna).

jirutka, 2023-02-20:
This bug probably isn’t fixed yet.

closed 2023-03-31 by github-actions[bot]

10der, 2024-04-12:
```
info  2024-04-12 14:31:36: Logging to console and directory: '/config/zigbee2mqtt/log/2024-04-12.14-31-36' filename: log.txt
info  2024-04-12 14:31:36: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
info  2024-04-12 14:31:36: Starting zigbee-herdsman (0.40.3)
error 2024-04-12 14:31:48: Error while starting zigbee-herdsman
error 2024-04-12 14:31:48: Failed to start zigbee
error 2024-04-12 14:31:48: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-04-12 14:31:48: Exiting...
error 2024-04-12 14:31:48: Error: Failure send getNetworkParameters:{"type":"Buffer","data":[24,0,1,40,0]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:664:23
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Driver.needsToBeInitialised (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:288:27)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:224:13)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
```

`  adapter: ezsp`

10der, 2024-04-12:
```
error 2024-04-12 14:32:03: Error: Failure send version:{"type":"Buffer","data":[0,0,0,4]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:664:23
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:469:24)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:180:9)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
```

sleepypig2003, 2025-02-01:
Unsure if this is the same issue, my z2m failed to start.  It looks different from the original problem, but the most recent post shows the same call trace.  Probably a new issue?

```
[2025-02-01 17:00:45] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-01 17:00:45] info: 	z2m: Loaded external converter 'TS011F.js'
[2025-02-01 17:00:45] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-02-01 17:00:45] info: 	z2m: Starting Zigbee2MQTT version 1.41.0 (commit #unknown)
[2025-02-01 17:00:45] debug: 	z2m: sd-notify loaded
[2025-02-01 17:00:45] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2025-02-01 17:00:45] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":31307,\"extendedPanID\":[229,241,187,249,52,99,223,176],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyAMA1\",\"adapter\":\"ezsp\"},\"adapter\":{\"disableLED\":false}}"'
[2025-02-01 17:00:45] debug: 	zh:ezsp: Adapter concurrent: 8
[2025-02-01 17:00:45] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":31307,"extendedPanID":[229,241,187,249,52,99,223,176],"channelList":[11]},"serialPort":{"path":"/dev/ttyAMA1","adapter":"ezsp"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-02-01 17:00:45] warning: 	zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: RTS/CTS config is off, enabling software flow control.
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: Opening SerialPort with {"path":"/dev/ttyAMA1","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: Serialport opened
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: Uart reseting
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: --> Write reset
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: --> [1ac038bc7e]
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: -?- waiting reset
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: <-- [75ceb1a97d312a15b65894a524ab5593499c817b7d38788e9874f0cf518bfc0a39a5ebc9de6f8ed9c538a27e05ceb1a97d312a15b65894a524ab5593499c9e7b7d38788e9874f0cf518bfc0a39a5ebc9de6f8ed9c52ee17e]
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: <-- DATA (7,5,0): 75ceb1a9112a15b65894a524ab5593499c817b18788e9874f0cf518bfc0a39a5ebc9de6f8ed9c538a27e
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: Unexpected DATA packet sequence 7 | 0: reject condition
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: --> [a0547d3a7e]
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: <-- DATA (0,5,0): 05ceb1a9112a15b65894a524ab5593499c9e7b18788e9874f0cf518bfc0a39a5ebc9de6f8ed9c52ee17e
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: --> ACK  (1)
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: --> [8160597e]
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: <-- ACK (5): 05ceb1a9112a15b65894a524ab5593499c9e7b18788e9874f0cf518bfc0a39a5ebc9de6f8ed9c52ee17e
[2025-02-01 17:00:45] debug: 	zh:ezsp:uart: Unexpected packet sequence 5 | 0
[2025-02-01 17:00:45] debug: 	zh:ezsp:ezsp: <== Frame: 8c9001450000040100ef010100010000d05cb39540ffff0d0932020074060200040000012602
[2025-02-01 17:00:45] debug: 	zh:ezsp:ezsp: <== 0x1: {"_cls_":"getLibraryStatus","_id_":1,"_isRequest_":false,"status":69}
[2025-02-01 17:00:46] debug: 	zh:ezsp:uart: <-- [c1020b0a527e]
[2025-02-01 17:00:46] debug: 	zh:ezsp:uart: <-- RSTACK c1020b0a527e
[2025-02-01 17:00:46] debug: 	zh:ezsp:uart: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e
[2025-02-01 17:00:46] debug: 	zh:ezsp:uart: -+- waiting reset success
[2025-02-01 17:00:48] debug: 	zh:ezsp:ezsp: ==> version: {"desiredProtocolVersion":4}
[2025-02-01 17:00:48] debug: 	zh:ezsp:ezsp: ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":4}
[2025-02-01 17:00:48] debug: 	zh:ezsp:uart: --> DATA (0,1,0): 00000004
[2025-02-01 17:00:48] debug: 	zh:ezsp:uart: --> [014221a850477d5d7e]
[2025-02-01 17:00:48] debug: 	zh:ezsp:uart: -?- waiting (1)
[2025-02-01 17:00:48] debug: 	zh:ezsp:uart: <-- [a0547d3a7e]
[2025-02-01 17:00:48] debug: 	zh:ezsp:uart: <-- NAK (0): a0541a7e
[2025-02-01 17:00:48] debug: 	zh:ezsp:uart: NAK Unexpected packet sequence 0
[2025-02-01 17:00:52] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":1} after 4000ms
[2025-02-01 17:00:52] error: 	zh:ezsp:uart: -!- break waiting (1)
[2025-02-01 17:00:52] error: 	zh:ezsp:uart: Can't send DATA frame (0,1,0): 00000004
[2025-02-01 17:00:53] debug: 	zh:ezsp:uart: ->> DATA (0,1,1): 00000004
[2025-02-01 17:00:53] debug: 	zh:ezsp:uart: --> [094221a85045507e]
[2025-02-01 17:00:53] debug: 	zh:ezsp:uart: -?- rewaiting (1)
[2025-02-01 17:00:57] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":1} after 4000ms
[2025-02-01 17:00:57] error: 	zh:ezsp:uart: -!- break rewaiting (1)
[2025-02-01 17:00:57] error: 	zh:ezsp:uart: Can't resend DATA frame (0,1,1): 00000004
[2025-02-01 17:00:57] debug: 	zh:ezsp:ezsp: onSerialReset()
[2025-02-01 17:00:57] debug: 	zh:ezsp:driv: onEzspReset()
[... cut ...]
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:457:24)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:195:9)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:152:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)
```

Sorin-1, 2025-02-03:
)
> [2025-02-01 17:00:52] error: 	zh:ezsp:uart: Can't send DATA frame (0,1,0): 00000004
> [2025-02-01 17:00:53] debug: 	zh:ezsp:uart: ->> DATA (0,1,1): 00000004
> [2025-02-01 17:00:53] debug: 	zh:ezsp:uart: [...]

> ```

Looks like connection problem. As I think, this is the usual cause of zigbee adapter faults. If it was working fine before, consider rechecking power adapter (replace to test) and USB extension cord for zigbee adapter. Changing/reattaching of both may help in most cases, as I think.