Koenkk/zigbee2mqtt #15009: [New device support]:

mikeg-de, 2022-11-15:
### Link

https://de.aliexpress.com/item/1005004696707285.html?spm=a2g0o.order_list.0.0.6aaf5c5flG1pw9&gatewayAdapt=glo2deu

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x84ba20fffe9a617a","nwkAddr":53443,"manufId":4098,"manufName":"_TZE200_bvu2wnxz","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"L�\u0006+iV�\u0006+ia�\u0006+ik�\u0006+iu�\u0006+i��\u0006+i��\u0006+iS�\u0006+i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_bvu2wnxz","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1668545364476,"defaultSendRequestWhen":"immediate"}

### Comments

Pairing did work but getting an unsupported manufacturer error message. Trying to create an external converted but that is new to me and might take some time.

mikeg-de, 2022-11-17:
Might be duplicate: https://github.com/Koenkk/zigbee2mqtt/issues/12315

@Koenkk Would you agree to close this in favor of 12315?

mikeg-de, 2022-11-17:
Works, to the greater extend, with the external converted from https://github.com/twhittock/avatto_me167

Some exchange is happening in https://github.com/Koenkk/zigbee2mqtt/issues/12315 which is already closed. Hence, keeping this open.

Battery is said to have no readout but got a blinking battery icon in the device pane after inserting an 1.2 V instead of 1.5 V battery when using 'e.battery()'.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tuyaLocal = {
  dataPoints: {
    me167Mode: 2,
    me167HeatingSetpoint: 4,
    me167LocalTemp: 5,
    me167ChildLock: 7,
    me167Heating: 3,
    me167Schedule1: 28,
    me167Schedule2: 29,
    me167Schedule3: 30,
    me167Schedule4: 31,
    me167Schedule5: 32,
    me167Schedule6: 33,
    me167Schedule7: 34,
    me167ErrorCode: 35,
    me167FrostGuard: 36,
    me167AntiScaling: 39,
    me167TempCalibration: 47,
  },
};

const fzLocal = {
  me167_thermostat: {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};

        function weeklySchedule(day, value) {
          // byte 0 - Day of Week (0~7 = Wed ~ Tue) ???
          // byte 1 - hour ???
          // byte 2 - minute ???
          // byte 3 - Temp (temp = value )
          // byte 4 - Temperature (temp = value / 10)

          const weekDays=[ 'wed', 'thu', 'fri', 'sat', 'sun','mon', 'tue'];
          // we get supplied in value only a weekday schedule, so we must add it to
          // the weekly schedule from meta.state, if it exists
          const weeklySchedule= meta.state.hasOwnProperty('weekly_schedule') ? meta.state.weekly_schedule : {};
          meta.logger.info(JSON.stringify({'received day': day, 'received values': value}));
          let daySchedule = []; // result array
          for (let i=1; i<16 && value[i]; ++i) {
            const aHour=value[i];
            ++i;
            const aMinute=value[i];
            ++i;
            const aTemp2=value[i];
            ++i;
            const aTemp=value[i];
            daySchedule=[...daySchedule, {
              temperature: Math.floor((aTemp+aTemp2*256)/10),
              hour: aHour,
              minute: aMinute,
            }];
[... 219 lines cut ...]
        tzLocal.me167_thermostat_system_mode,
        tzLocal.me167_thermostat_schedule,
        tzLocal.me167_thermostat_calibration,
        tzLocal.me167_thermostat_anti_scaling,
        tzLocal.me167_thermostat_frost_guard,
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
      e.child_lock(),
      exposes.switch().withState('anti_scaling', true).withDescription('Anti Scaling feature is ON or OFF'),
      exposes.switch().withState('frost_guard', true).withDescription('Frost Protection feature is ON or OFF'),
      exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 1)
                     .withLocalTemperature()
                     .withSystemMode(['auto','heat','off'])
                     .withRunningState(['idle', 'heat'], ea.STATE)
                     .withLocalTemperatureCalibration(-10, 10, 1, ea.STATE_SET)
    ],
};

module.exports = definition;
```

mikeg-de, 2022-11-19:
The kudos should go to @raketenemo  per https://github.com/twhittock/avatto_me167

I don't get any errors and have the debug running since I installed all thermostats. However, I do get a strange behavior as if there is a manufacturer setting overwriting mine.

Commands send via Node-Red seem to take effect only briefly
[image]

Manualy overwrite via thermostat seems to be kept
[image]

But, settings are only kept for circa 15 to 20 minutes which significantly degrades comfort.
[image]

The product page states it is "programmable" so I assume there are some yet to be discovered and thus not exposed settings.

[image]

I tried to retrieve / decipher the raw data points as described in https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats by registering as described but that tutorial isn't applying anymore and I got stuck.

I will try to follow https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_3-understanding-tuya-data-points and attempt to reach out to the manufacturer as well asking for UART protocol ans DPIDs and data formats for their functions.

GrumpyMeow, 2022-11-22:
Indeed clicking on the button to retrieve the state of anti_scaling or frost_guard, will trigger an error.
I've been comparing the me167.js file with https://github.com/Koenkk/zigbee-herdsman-converters/blob/4f7734b373921fb5524df33a777a290d1bc77bd4/lib/tuya.js.
I changed some code to:
```
 exposes: [
      e.child_lock(),
      exposes.binary('anti_scaling', ea.STATE_SET, 'ON', 'OFF').withDescription('Anti Scaling feature is ON or OFF'),
      exposes.binary('frost_guard', ea.STATE_SET, 'ON', 'OFF').withDescription('Frost Protection feature is ON or OFF'),
      ...
```

This removed the get-state-buttons. Which "resolves" the error.

mikeg-de, 2022-11-23:
So far no PR was made as we are presumably still in discovery phase. The feedback form the manufacturer was not quite constructive as, apparently, he simply wants to sell his stuff (gateway).

Therefore, I dug up the logs to retrieve the date from the "interview" phase when pairing an and device and got this schema:

**Raw log**
`info  2022-11-23 11:20:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Thermostatic radiator valve","exposes":[{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{"description":"Anti Scaling feature is ON or OFF","features":[{"access":7,"name":"state","property":"anti_scaling","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"description":"Frost Protection feature is ON or OFF","features":[{"access":7,"name":"state","property":"frost_guard","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"features":[{"access":7,"description":"Temperature setpoint","name":"current_heating_setpoint","property":"current_heating_setpoint","type":"numeric","unit":"°C","value_max":35,"value_min":5,"value_step":1},{"access":5,"description":"Current temperature measured on the device","name":"local_temperature","property":"local_temperature","type":"numeric","unit":"°C"},{"access":7,"description":"Mode of this device","name":"system_mode","property":"system_mode","type":"enum","values":["auto","heat","off"]},{"access":1,"description":"The current running state","name":"running_state","property":"running_state","type":"enum","values":["idle","heat"]},{"access":3,"description":"Offset to be used in the local_temperature","name":"local_temperature_calibration","property":"local_temperature_calibration","type":"numeric","unit":"°C","value_max":10,"value_min":-10,"value_step":1}],"type":"climate"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ME167","options":[],"supports_ota":false,"vendor":"Avatto"},"friendly_name":"0x84ba20fffe9a6a5e","ieee_address":"0x84ba20fffe9a6a5e","status":"successful","supported":true},"type":"device_interview"}'
`

**Formatted JSON**
```
{
  "data": {
    "definition": {
      "description": "Thermostatic radiator valve",
      "exposes": [
        {
          "features": [
            {
              "access": 3,
              "description": "Enables/disables physical input on the device",
              "name": "state",
              "property": "child_lock",
              "type": "binary",
              "value_off": "UNLOCK",
              "value_on": "LOCK"
            }
          ],
          "type": "lock"
        },
        {
          "description": "Anti Scaling feature is ON or OFF",
          "features": [
            {
              "access": 7,
              "name": "state",
              "property": "anti_scaling",
              "type": "binary",
              "value_off": "OFF",
              "value_on": "ON",
              "value_toggle": "TOGGLE"
            }
          ],
          "type": "switch"
        },
        {
          "description": "Frost Protection feature is ON or OFF",
          "features": [
            {
              "access": 7,
              "name": "state",
              "property": "frost_guard",
              "type": "binary",
              "value_off": "OFF",
              "value_on": "ON",
              "value_toggle": "TOGGLE"
            }
          ],
          "type": "switch"
        },
        {
          "features": [
            {
              "access": 7,
              "description": "Temperature setpoint",
              "name": "current_heating_setpoint",
              "property": "current_heating_setpoint",
              "type": "numeric",
              "unit": "°C",
              "value_max": 35,
              "value_min": 5,
[... 205 lines cut ...]
Worth to note the later can be generally described as x*y (x representing the hour and y representing the respective settings). Therefore, the mapping between the two messages looks like so:

```
[
  1, // Day of week, ranging from 1 to 7
  6,0,0,120, // Hour of day, unknown or minute, desired temperature as int
  8,0,0,160,
  18,0,0,220,
  20,0,0,200
]
```

Day-wise 
```
[
  {"temperature":12,"hour":6,"minute":0},
  {"temperature":16,"hour":8,"minute":0},
  {"temperature":22,"hour":18,"minute":0},
  {"temperature":20,"hour":20,"minute":0}
]
```

What is intriguing is the extra value in the buffer message as there are four values vs. three in the day-wise data. Maybe the datatype per initial debug message which is preceding the info messages?

I am also curious if, in order for a day wise scheduling, sending both or only either of two messages is required.

Evgeka07, 2022-12-06:
@mikeg-de 
Hello, I installed according to your introduction. Works great
But don't understand where and how to enter the schedules
Can I somehow do this via the card or do I need a special card for this?
Formatted JSON - in which file does the text go?
Formatted - where does that go?
or can you upload the finished files?
Sorry for my English

woody4165, 2022-12-09:
Hi
I got it working with me167, but when I set system_mode to heat and set a temperature, after some minutes it always returns to auto and changes the temperature set before.
Any suggestion?
Thanks

mikeg-de, 2022-12-09:
@Evgeka07 I haven't figured out the scheduling by back engineering the format form my earlier post https://github.com/Koenkk/zigbee2mqtt/issues/15009#issuecomment-1324892985 either yet. The manufacturer was quite reluctant sharing the UART protocol and DPIDs data formats. had no time to test but plan to do so soon.

@woody4165 check your thermostats. I had the exact same issue and the timing / auto function was active. Please see my previous comment as well:

> Checking the manual it becomes apparent that there are different operation modes as such "manual" and "programming". The later indicated by the clock icon on the thermostat display. I figure out that, in order to switch on the thermostat to manual mode, you have to press the "rotator" button twice if it sleeps and once if enabled..
> 
> [ME167 Thermostat Manual.pdf](https://github.com/Koenkk/zigbee2mqtt/files/10046381/ME167.Thermostat.Manual.pdf)
> 
> Checking the debug log I traced this back to "system_mode": "auto" vs. "system_mode": "heat". which is an exposed device setting.

woody4165, 2022-12-11:
> @woody4165 check your thermostats. I had the exact same issue and the timing / auto function was active. Please see my previous comment as well:

Thanks, I've tried, but strangely after few seconds, sometimes minutes, it returns in programming mode, even if don't do anything physically or from zigbee2mqtt
My model should be _TZE200_bvu2wnxz

scoof, 2022-12-12:
> Thanks, I've tried, but strangely after few seconds, sometimes minutes, it returns in programming mode, even if don't do anything physically or from zigbee2mqtt
> My model should be _TZE200_bvu2wnxz

Set system mode to "heat", and it will not go back to the schedule, but will just target the setpoint.

woody4165, 2022-12-12:
I've solved installing the dev branch
It was recognized immediately and now it's working well

It's recognized as 
Brand  [TuYa]
Model [TS0601_thermostat_3]

Thanks

Evgeka07, 2022-12-12:
@woody4165 How did you do it? Can you set the instructions?

woody4165, 2022-12-12:
@Evgeka07 I've followed these steps [https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html](url)
I'm using Proxmox with Z2M in a container.
So I was sure that if anything was going to fail, I can restore the backup of the CT

closed 2023-01-21 by github-actions[bot]